jgyates/genmon #263: Exercise notification by email

grsthegreat, 2019-05-11:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below.

***NOTE:*** If you are having issues with your serial connection, please read this page before posting:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

If you are having other issues, please see the following page:

https://github.com/jgyates/genmon/wiki/3.5---General-Troubleshooting

If you need to send you logs and registers to the developer, if you email is setup and working properly you can click send your logs on the About page in the web interface.

### Expected Behavior

{Please write here}

### Actual Behavior

{Please write here}

### Steps to Reproduce (including precondition)

{Please write here}

### Screenshot or Pictures relating to the problem (if possible)

{Please write here}

### Your Environment

- Generator Model: {Please write here}
- Generator Registers: Submit via the About page
- Genmon Version: {Please write here}

grsthegreat, 2019-05-11:
Will the nit email me when it performs its weekly exercise program?

jgyates, 2019-05-11:
Yes. If you configure your email on the settings and notification page the software will send you notifications when the status of the generator changes. This includes cranking, exercising, etc.

closed 2019-05-11 by jgyates