jgyates/genmon #264: Generator time not sinced

grsthegreat, 2019-05-12:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below.

***NOTE:*** If you are having issues with your serial connection, please read this page before posting:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

If you are having other issues, please see the following page:

https://github.com/jgyates/genmon/wiki/3.5---General-Troubleshooting

If you need to send you logs and registers to the developer, if you email is setup and working properly you can click send your logs on the About page in the web interface.

### Expected Behavior

{Please write here}

### Actual Behavior

{Please write here}

### Steps to Reproduce (including precondition)

{Please write here}

### Screenshot or Pictures relating to the problem (if possible)

{Please write here}

### Your Environment

- Generator Model: {Please write here}
- Generator Registers: Submit via the About page
- Genmon Version: {Please write here}

grsthegreat, 2019-05-12:
It shows time 2 hours off actual time. As if its running central time instead of pacific time

jgyates, 2019-05-12:
I am not sure from your description, but I am assuming you have clicked "Set Generator Time" button on the Maintenance page on the web interface. This will sync the generator time to the raspberry pi time once. The "Sync Time" option on the settings page will sync the time once per hour.

If you have done either of these and your time is not correct I would guess that your raspberry pi does not have the correct time zone selected.

This article describes how to do that:

https://howchoo.com/g/njnlzjmyyjg/how-to-set-the-timezone-on-your-raspberry-pi

grsthegreat, 2019-05-12:
Thank you. The Raspberry pi zone was set wrong. All good now

closed 2019-05-12 by jgyates