jgyates/genmon #265: Last Update

diverbill, 2019-05-12:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below.

***NOTE:*** If you are having issues with your serial connection, please read this page before posting:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

If you are having other issues, please see the following page:

https://github.com/jgyates/genmon/wiki/3.5---General-Troubleshooting

If you need to send you logs and registers to the developer, if you email is setup and working properly you can click send your logs on the About page in the web interface.

### Expected Behavior

system update and restart

### Actual Behavior

system stopped responding could not ssh the pi.  rebooted the pi and genmon will not start.
currently reinstalling I do have backups.

### Steps to Reproduce (including precondition)

{Please write here}

### Screenshot or Pictures relating to the problem (if possible)

{Please write here}

### Your Environment

- Generator Model: RD20
- Generator Registers: Submit via the About page
- Genmon Version: was the latest

diverbill, 2019-05-12:
./genmonmaint.sh install does not seem to do anything to install.

jgyates, 2019-05-12:
@diverbill 

can you give more info? What does the output of genmonmaint.sh show?

diverbill, 2019-05-12:
nothing that is the problem im looking for logs someplace - this is a fresh git pull and it seems to run but nothing happens 

pi@raspberrypi:~/genmon $ ./genmonmaint.sh install
pi@raspberrypi:~/genmon $ ./genmonmaint.sh
pi@raspberrypi:~/genmon $ ./genmonmaint.sh show
pi@raspberrypi:~/genmon $

jgyates, 2019-05-12:
OK, the issue with genmonmaint.sh is that the command line options changed in this release:

https://github.com/jgyates/genmon/wiki/1----Software-Overview#genmonmaintsh

 try 

    ./genmonmaint.sh -i

diverbill, 2019-05-12:
I ran the update and saw the warning about backing up the data as things move with this update. I ran the backup and have the .tar file. once the update was done the page just kept saying loading then everything seemed to lock up on the pi. i rebooted the pi via my poe injector and it came back up but the software would not start. so i figured if files moved just do a fresh install.

The -i seemed to have worked. I will work on getting my config files placed back into place. 
I wanted to give you a heads up prior to anyone else having the same issue.
Thanks again.

jgyates, 2019-05-12:
I checked in an update that I hope will help (V1.13.09). I think in certain circumstances the file system was not keeping up with the python code (i.e. the code coped some files and expected them to be there but the file system was not finished upating yet) but I added a fix that should address this. If anyone has issues please let me know.

jgyates, 2019-05-12:
more details on this issue are in this thread:

https://github.com/jgyates/genmon/issues/266

closed 2019-05-12 by jgyates