jgyates/genmon #266: Monitor will not load after update

RoyRomeo, 2019-05-12:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below.

***NOTE:*** If you are having issues with your serial connection, please read this page before posting:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

If you are having other issues, please see the following page:

https://github.com/jgyates/genmon/wiki/3.5---General-Troubleshooting

If you need to send you logs and registers to the developer, if you email is setup and working properly you can click send your logs on the About page in the web interface.

### Expected Behavior

Generator monitor to load

### Actual Behavior

After most recent update, screen shows "Loading Generator Monitor" but does not load.

### Steps to Reproduce (including precondition)

{Please write here}

### Screenshot or Pictures relating to the problem (if possible)

{Please write here}

### Your Environment

- Generator Model: {Please write here}
- Generator Registers: Submit via the About page
- Genmon Version: {Please write here}

jgyates, 2019-05-12:
Are you using V1.13.08 or V1.13.09?  V1.13.09 was posted about an hour or two ago.

Try this to get you back to a known state. Log into the pi and get into the genmon directory:

     ./genmonmaint.sh -u
     cp ./conf/genloader.conf /etc/genmon 
     ./startgenmon.sh restart

   
Let me know how that goes.

SethL, 2019-05-12:
I'm having the same exact issue.  I followed the steps you just suggested and no go.  It still just hangs on loading in the webui.  After running the commands you suggested, my device lost ssh and vnc for a while.  then i get an email saying genmon is not communicating with the controller.  Now I have VNC and ssh back but the webui is still stuck on loading

RoyRomeo, 2019-05-12:
I'm using the one that was available yesterday.  Do you log in to the Pi via Putty?  If so, what is the command line to access the Pi?

SethL, 2019-05-12:
[image]
 
Could it be rebuilding the db or something?  the CPU is pegged at 100% and genmon.py is consuming it via multiple threads. see attached

jgyates, 2019-05-12:
Try a reboot, then see if that makes a difference. I am not sure why it is locking, but the update from today is needed as I discovered a problem with the file copy. 

If you have a backup you can unzip the files and copy them into the /etc/genmon directory, then reboot. The base issue that appears to be happening is that the config files are being corrupted during the copy. the new version of the software moves all the config files from /etc/ to /etc/genmond to better support multiple instances. but V1.13.08 had a problem that could occur if the system did not copy the files fast enough, they could be corrupted.

Also, you can alway delete the files in the /etc/genmon/ directory and the software should re-write them with a fresh copy. This is the equivalent of resetting all settings to the original.

jgyates, 2019-05-12:
one thing I am trying to determine is this: Is there a problem doing an upgrade with V1.13.09. Know there is a problem that can occur if you upgrade to V1.13.08, but I think I fixed that with V1.13.09. If anyone has tried this (besides me) and can validate that would be great.

SethL, 2019-05-12:
Im not sure what version I updated to.. I left the UI open for a day or so and then clicked update.  so it may have updated to 1.13.08.  I did create a backup before I did it so once I copied the files over and restarted it worked like a charm :-)

Thanks!

swvalenti194, 2019-05-12:
1.13.08 updated but genmom wouldn't start. Had to reboot pi and it came up. Updated to .09 with zero issues came right up. Only thing I noticed is the notification via MMS on ATT took days or hours to notify on change etc...

RoyRomeo, 2019-05-13:
I tried a reboot of the pi, and can no longer connect.

jgyates, 2019-05-13:
Did you have your pi setup for a static or dynamic IP address? If it is dynamic then the reboot may have caused your router to issue a new IP address. If this is the case then logging into your router may give you some insight as to the new IP address. If it is setup for a static IP and you can not connect then your best best is to connect a keyboard and monitor and try to determine what is not working.

RoyRomeo, 2019-05-13:
When I type in "wifisetup.pi:1880/GenPi", I get a "Welcome to the Node-RED Dashboard, Please add some UI nodes to your flow and redeploy.

SethL, 2019-05-13:
You should connect to your pi directly with kbd and monitor if you cannot
connect remotely.

liltux, 2019-05-13:
@RoyRomeo did you reset your network setup?  The wifisetuo.pi:1880/GenPi only works if your computer is connected to the genpi network when you reset your network settings and restart the AP.  Else you would type in the IP address of the pi and :1880/GenPi to access the node red UI.  You may have to use your router or a free app like FING to scan your network and identify your Pi's IP address.

jgyates, 2019-05-13:
@RoyRomeo 

I believe you have the wifi config from @liltux. He may contact you about this if he has any suggestions regarding the Node-Red config.

If you are familure with ping you should be able to open a command prompt or terminal session and type:

    ping wifisetup.pi

If this is successful then you know that your IP is still on your network.

From there you should be able to log into your pi using the ssh program. the Node-Red configuration is something that @liltux has setup on the the pre-configured pi available on e-bay.

liltux, 2019-05-13:
@RoyRomeo you can email me directly at liltux @ liltux dot com.

jcherney, 2019-05-18:
Same problem here. Loaded version 1.13.09 and it just spins waiting for the page to load. Tried using 
 ./genmonmaint.sh -u
 cp ./conf/genloader.conf /etc/genmon 
 ./startgenmon.sh restart

But after this command:    cp ./conf/genloader.conf /etc/genmon   I get a    cp:missing destination file operand after './conf/genloader.conf/etc/genmon'

Thoughts?

jcherney, 2019-05-18:
Actually, ./genmonmaint.sh -u errors out also.  no such file or directory

jcherney, 2019-05-18:
Ok, I wasnt in the genmon directory, sorry.  The Script uploaded fine, but the next command still gets an error or command not found.

jcherney, 2019-05-18:
[image]

liltux, 2019-05-18:
@jcherney what version were you updating from?  Might try:
sudo bash genmonmaint -i 
This will reinstall everything.  Also if doing this backup your current conf files so they don't get overwritten.

jcherney, 2019-05-18:
@liltux  I think it was 1.13.07   Your command suggestion gives no such file or directory.

jcherney, 2019-05-18:
I got the genmontmaint.sh -u to work- see above screenshot.  It's the next command that doesnt

liltux, 2019-05-18:
Oops, 
sudo bash /home/pi/genmon/genmonmaint.sh -i

liltux, 2019-05-18:
@jcherney the script is trying to copy or verify files and either files or directory is missing.

jcherney, 2019-05-18:
Can I back up the config files from the ssh session before I issue that command?

PS - I can do a hard restart outside. Would that eliminate any of this ?

liltux, 2019-05-18:
The resetting outside would more than likely not affect.  Look for Gen*.conf files into the /etc directory.  You can then cp them to ~/ directory (home folder shorthand).
Easy command
ls /etc | grep gen*
See what is listed, bet most are the files that you need.

jcherney, 2019-05-18:
[image]

liltux, 2019-05-18:
@jcherney

ls(space)/etc(space)|(space)grep(space)gen*

ls /etc | grep gen*

I see, I left off a colon.  It's LS not is, sorry.

jcherney, 2019-05-18:
Sorry to be a pain in the ass, but: 
[image]

liltux, 2019-05-18:
Leave off the wild card, not thinking straight.

liltux, 2019-05-18:
Also will need mymail.conf

jcherney, 2019-05-18:
[image]

jcherney, 2019-05-18:
[image]

liltux, 2019-05-18:
Now you need to issue the copy command for those files.
Example:
cp /etc/mymail.conf ~/
Explained:
cp space orig file and location space new location.  Copy my mail.conf from etc to ~/ folder which is short for current user home directory

jcherney, 2019-05-18:
Ok, done. mymail.conf doesnt' seem to exist there though.
[image]

liltux, 2019-05-18:
It does, just do the run the cp command for mymail.conf just like you did the rest

jcherney, 2019-05-18:
Done.

jcherney, 2019-05-18:
Now run this?  sudo bash /home/pi/genmon/genmonmaint.sh -i

liltux, 2019-05-18:
Yes

jcherney, 2019-05-18:
Do I want to copy the configuration files to /etc/genmon?

jcherney, 2019-05-18:
[image]

liltux, 2019-05-18:
Answer yes to everything.  Then you will copy your back up files to /etc/genmon yes.

jcherney, 2019-05-18:
Ok, script is finished. what directory do i need to be in to do the copy back?

liltux, 2019-05-18:
In pi home dir.  Type:

cd ~/

jcherney, 2019-05-18:
so  cp gengpioin.conf /etc/genmon ?

liltux, 2019-05-18:
That would be correct syntax

jcherney, 2019-05-18:
[image]

liltux, 2019-05-18:
Sorry again, you will have to perform the action as sudo, so:
sudo cp ....

jcherney, 2019-05-18:
No worries. Done!

jcherney, 2019-05-18:
now this?  ./startgenmon.sh restart

liltux, 2019-05-18:
Best just to type:
sudo reboot
This will allow the pi to restart and run the scripts automatic

jcherney, 2019-05-18:
Awesome! It's back! Once again, thank you SO much!   What do you think is causing this?

liltux, 2019-05-18:
What I have noticed is that once performing the update on the webUI, on my
images(before latest) I have also been having to ssh in or using my NodeRed
webUI and reboot.  Works every time.  But when I am unsure what you may
have done, it's always best to go through the install again.  Makes sure
all needed files are where they should be.  The latest has update on GenMon
has worked for me, again I just have to issue a reboot to the pi and
everything comes back up.

jcherney, 2019-05-18:
What is NodeRed?

liltux, 2019-05-18:
Nice way to program IOT and other things using node and js.  NodeRed.org

jgyates, 2019-05-19:
Sorry for the late response, I was helping my son move today so I was away from the internet until now. Sounds like @liltux got you back up and running.

@jcherney : One question, are you running on a pi3 or pi zero? I have a delay to let the file system catch up in the upgrade code that runs only once when upgrading to 1.13.09 (it does not need to run in future versions). I am wondering if the the pi zero may need a longer delay. another thing that could cause this is an unusually slow SD card.

jcherney, 2019-05-19:
He did. He’s a huge help

I’m running on a Pi3

jcherney, 2019-05-19:
Hope the move went well!

liltux, 2019-05-19:
@jgyates how much of a delay do you have in the update?  I am wondering if I'm getting in too much of a hurry?  All of my images are now up with the latest revision.  I did work on 2 today that had older revisions and after update, once I saw the spinning wheel, I just ran sudo reboot and once restarted everything was good.   No to say there was a problem before, I was just impatient.

gmqest, 2019-05-20:
Just upgraded pi zero from V1.13.07  one hour ago and encountering the same issue: hangs on loading the webui. 

After a reboot, the pi zero seems to be doing something - CPU usage at ~30%, load average ~0.43. Python using all resources and no other services running on the pi. Will leave it alone for a while to see if it starts.

[9 more replies]

closed 2019-06-04 by jgyates