jgyates/genmon #267: [Feature Request] Start+Transfer / Stop with single GPIO

JMVS, 2019-05-28:
Could it be possible to have an additional GPIO or a config setting that will allow changing the current state between start+transfer and stop with a single input?
I'm trying to interface GenMon with an existing installation that has a remote control receiver with a single relay (now working NC/NO) that can be configured for a single button to activate the relay (NC) for 3 seconds. With my feature request, this will allow the generator to stop  when is running and start+trasnfer when is stopped with a single GPIO and a pulse. 
Thanks.

liltux, 2019-05-28:
@JVMS have you looked at the gengpioin.py?  It allows for inputs into the gpio header to start, stop, and start/transfer (if the controller supports it).  The controller will not allow a remote stop if running during an outage.

jgyates, 2019-05-28:
@JMVS 

@liltux is correct, this is what gengpioin.py does now:

https://github.com/jgyates/genmon/wiki/1----Software-Overview#gengpioinpy-optional

You can enable it from the add on page in the UI. Let me know if you have any questions.

JMVS, 2019-05-30:
@liltux @jgyates Thanks for your responses.
@liltux I did not look at gengpioin.py. If you are referring about this addon to be able to do what I need **but on different GPIO inputs**, yes I did know that. If you are talking about changing the code to suit my needs, I don't want to do that because it will brake after every update. 
I understood you correctly? **We cannot stop the generator via the RPi when is running during an outage?** That's precisely one of the options I was expecting to have...being able to stop the generator _when I'm out_ and _when is running due to an outage_ and not really needing it.
@jgyates Is there any workaround?
I have a solution in place (involving relays, DC-AC inverters and hardware hacking) but the GenMon + RPi is a far more elegant solution.

liltux, 2019-05-30:
As a technician, stopping the generator remotely underload during an outage is not recommended.  This can cause property and equipment damage.  The problem lies in there is no way to remove the load during an outage remotely without extensive rework.  Shutting down the generator underload will cause brownout effects on personal property as well as not allowing the generator ample cool down time to protect equipment.

JMVS, 2019-05-30:
Thank you for the explanation. I understand your concerns, but I thought that was handled internally by the controller (remove the load and cool down). Right now I'm doing this by re-routing the voltage sensing cables through a DPDT relay with a DC-AC Inverter and hooked to a RF receiver (simplified explanation). This does exactly what you explain, as I'm fooling the generator to believe there's utility power (inverter) which in turn removes the load, cools down and stops. The downside is that the system will drain the generator's battery (relays, inverter & RF receiver run on 12vdc) for as long there's no real utility back. Is not an elegant solution either.
There's no way to achieve this via GenMon/Controller (Evolution in my case)? 2-wire start?
I think this is a pretty common scenario...not wanting to use the generator even if there's no utility power and the generator is not easily reachable.

liltux, 2019-05-30:
@JVMS your are correct as in cooldown is handled by the controller and transfer of the switch is also controlled by the Evo/Nexus controller by grounding wire 23.  But the problem lies in the fundamentals of transfer switches.  It always takes source voltage to transfer to source position.  Therefore without utility power the transfer switch cannot transfer itself to utility position.  There is no built in control to remove load during the outage.  But, if you perform a start with transfer command and utility is present then a stop command will transfer the switch and start cooldown since utility is present.

jgyates, 2019-06-04:
@JMVS ,

I am closing this issue for now as it appears the functionality you want is not an option considering the operation of the transfer switch. If you have any other questions feel free to post them in this thread.

closed 2019-06-04 by jgyates