jgyates/genmon #268: Anyone tried with E Panel (0A7605) on a Model: 5086360100/  Type: SD0060?

jimmypuckett, 2019-06-03:
I have searched through the open & closed issues & I am not seeing anyone talking about the SD line with the E Panel...

[image]

http://soa.generac.com/manuals/373113/0A7605 (page 64)

From looking at the manuals for this panel, it appears that there are about 18 properties that you can get via serial...

http://soa.generac.com/manuals/373113/0A9825

I see some images of the panel that actually have RS 232 on them, so I am not sure if there are different versions?

- Generator Model: 5086360100
- Generator Registers: Unkown
- Genmon Version: any

[image]

liltux, 2019-06-03:
The serial port on the back of the E Panel supports RS232 and RS485.  The 485 is for annunciator and relay panel communication.  The 232 is for used for programming and diagnostic.  I know I don't have any modbus mapping for this panel but if you could hook up a device to your Epanel generator you could perform a modbus dump and see if you can read registers.  The modbusdump.py is in the Others Programs folder.  I'm sure @jgyates would be willing to add support if you could assist.

jgyates, 2019-06-03:
@jimmypuckett 

I can guide you through using the modbusdump.py program. This is in the the repository to aid in reverse engineering the registers. As long as it supports modbus we should be able to figure it out. The first thing we will need is a system that runs python connected via to the generator via the serial port. Then we can use modbusdump in the /genmon/OtherApps folder to get a dump of the registers:

Here is the syntax for modbusdump.py:

python mobusdump.py -r <Baud Rate> -p <serial port> -a <modbus address to query> -s <start modbus register>  -e <end modbus register>

   Example: python mobusdump.py -r 9600 -p /dev/serial0 -a 9d -s 5 -e 100 

      -r  Baud rate of serial port (9600, 115300, etc)
      -p  Operating System device name of the serail port (/dev/serial0)
      -a  Modbus address to query in hexidecimal, 0 - ff. (e.g. 9d, 10, ff)
      -s  Starting modbus register to read (decimal number)
      -e  Ending modbus register to read (decimal number, must be greater than start register)
      -b  Stop bits. If omitted 1 stop bit, if present use 1.5 stop bits
      -x  Omit for no parity. -x 1 for odd parity, -x 2 for even parity

Let me know if you have any questions if you want to take this route or if you have any other ideas for figuring out the register map.

liltux, 2019-06-03:
@jgyates this controller uses a form of genlink for programming so I would try 64 as the address just like h panel. fyi

jimmypuckett, 2019-06-03:
So I have no problem running any of these commands, but I am just trying to figure out where to connect to the device.  From some of the pictures that I have seen, there is supposed to be a serial port on the back of the control box.  For the "D Panel", there is an extra wiring harness, so maybe mine needs that harness too?

Here is the back of my control box...

[image]

I think that that are screw down post in that box, that I can potentially hook into.  Do you guys know anything about how to hook into this device?

liltux, 2019-06-03:
@jimmypuckett the connection is directly on the back of the E-panel controller.  You have to get inside of the box your looking at.  4 screws in each corner and the top should come off.  There should be a db9 connection on the back of the E-panel itcontroller.  Notice the toggle switch above it, one position is for RS232 and the other is for RS485.  Sorry I don't have a picture for assistance.

jimmypuckett, 2019-06-03:
@liltux That is exactly what I needed to know.  I will get into it this afternoon.

Thanks!

jimmypuckett, 2019-06-04:
@liltux & @jgyates I took the cover off of the box to get to the control panel...

[image]

I see the label that discusses the 232/485, but I don't see the switch.  I need to get more light, but there is already a serial cable plugged in that goes down to the screw terminals (The grey wire in bottom right)...

[image]

There is this sticker on the top of the control...

[image]

This brings me to my questions...

1. Can I unplug the existing cable?
2. Do you know where the toggle switch located?
3. If we do get this figured out, will I be able to have both?

jimmypuckett, 2019-06-04:
In looking over that screw down post, it appears that that serial cable from the panel to the post is just a jumper to that distribution.  I am about 99% sure that I could just unhook that connection from the panel, switch to 232 & hook up my computer.

Does that sound right?

liltux, 2019-06-04:
@jimmypuckett you are correct.  Just unplug the DB9 connector on the back of the control panel, near it is a selector switch directly below it.  Be sure to select 232 if you are using a 232 to ttl converter.  Every time the selector is changed you will need to pull the fuse up front and cycle power to reset the communication settings and load the new selection.  I was talking with a fellow tech and he thinkgs the E-panel does not communicate modbus.   It is still serial but he is thinking it is a different protocol, I am unsure and do not have any good information at this time.  Good Luck.

jimmypuckett, 2019-06-09:
@liltux & @jgyates, I finally had time to run some test...

[image]

I did not have a serial cable at the house, so I used a USB to TTL cable that I have for console cable to raspberry pi.  I hooked alligator clips to the 485 post in the control box...

[image]

I ran the `modbusdump.py` script for all addresses from 1 to 255 with the following script...

```bash
for address in {1..255}; do hex=$(echo "obase=16; ${address}" | bc); echo "Address ${hex}"; python modbusdump.py -r 9600 -p /dev/tty.usbserial -a ${hex} -s 5 -e 100; echo ""; echo ""; done > out1
```

Then I swapped the clips to the other post to make sure that I had the correct connection & ran the script again...

```bash
for address in {1..255}; do hex=$(echo "obase=16; ${address}" | bc); echo "Address ${hex}"; python modbusdump.py -r 9600 -p /dev/tty.usbserial -a ${hex} -s 5 -e 100; echo ""; echo ""; done > out2
```

I checked both files for any output...

```bash
cat out1 out2 | grep -v "Baud Rate : 9600" | grep -v "Port is : /dev/tty.usbserial" | grep -v "Start Register : 5" | grep -v "End Register : 100" |  grep -v ^$
```

There is nothing in the files other than the address lines.

So I guess these are my options...

1. My adapter or cable is not good
2. The post I hooked to are wrong
3. The control panel does not support `Modbus`

I hope that it is not `3`.  I have another serial adapter, so I am going to get a serial cable & null modem adapter to hook directly to the control panel to eliminate `1` & `2`.

Do you guys have any other ideas for when I do my next test?

If not `Modbus`, then what what protocol would it be?

jgyates, 2019-06-10:
My guess is that it is something proprietary / non-standard, but that is just a guess. You may have better luck trying to interface with the relays if you have a relay kit. This doc from omnimetrix may offer some hints:

https://www.omnimetrix.net/wp-content/uploads/2018/05/IM-735-TrueGuard-PRO-Generac-EPanel-and-EPanel-w-Relay-Rev-E.pdf

Another option is to try to connect it to Genlink and capture the output, but that would require a genlink license.

jgyates, 2019-06-15:
I am closing this issue for now. If you have any additional data on this feel free to post it here.

closed 2019-06-15 by jgyates

mstackler, 2023-11-15:
I just arrived to my test site and discovered this panel.  Did this panel get supported? or was it to old?  

mark

jgyates, 2023-11-15:
The E panel does not appear to support modbus, or the comm parameters tested we're not correct. After looking at the above history I could be possible that the controller does not support modbus, or it could be that it uses another baud rate or parity setting. You could try it with some alternate settings if you want to exhaust all options, but to date we have not been able to get any data back.