jgyates/genmon #269: genmonmaint.sh option issue

jwtaylor310, 2019-06-04:
### Expected Behavior

Installation instructions state that the command "bash ./genmonmaint.sh install" should cause Installation of genmon

### Actual Behavior

gemmonmaint.sh exits without no action and no display

### Steps to Reproduce (including precondition)

execute command "bash ./genmonmaint.sh install"

### Your Environment

Raspberry Pi 3 Model B Rev 1.2 running Raspbian GNU/Linux 9 (stretch)

- Genmon Version: V1.13.12

I am a new user of your great software and ran into the following during the initial setup:

The installation instructions on the wiki indicate that genmonmaint.sh should be invoked with option "install" to perform an installation.  (https://github.com/jgyates/genmon/wiki/3.3--Setup-genmon-software)  However, it looks like the correct option syntax is "-i".  Apparently the script ignores any option that doesn't have a "dash" preceding it.  As a result, running the script with an invalid option without a preceding "dash"  (or running it with no option at all) will exit silently.  And, the  help display only appears if an invalid option is entered with a preceding dash. 

One other minor issue ... the wiki indicates that the "Update Software" button is on the "Monitor" page of the web interface (https://github.com/jgyates/genmon/wiki/4--Updating-to-the-Latest-Version).   On my installation of V1.13.12 it appears on the "About" page.

Thank you for a great product and for providing support for it!

jgyates, 2019-06-04:
Thanks for this input. I have updated the wiki. Let me know if you have any other issues

closed 2019-06-04 by jgyates

jwtaylor310, 2019-06-06:
Thank you!  --John

 

 

John Taylor

37 W Point Drive

Cocoa Beach, FL 32931

(321) 613-0271 (Residence)

(262) 225-3810 (Cell)