jgyates/genmon #270: Sropped working

JelloBiafra, 2019-06-05:
I had an installation of this running well until a week ago.  We had a power outage that lasted for about an hour.

When the outage ended, I could no longer access the app.  Pi was not working.  I removed from the generator and set itg up indoors.  The pi booted fine, but serial loopback test failed.  I replaced the rs232 interface and loopback worked.

I reinstalled but now there is no serial communication.  I swapped the tx and rx lines.  No change.  I pulled the unit, retested and reinstalled.  No change.

How do I troubleshoot from here?

liltux, 2019-06-05:
@JelloBiafra if everything works on the bench and loopback is good I would suspect the controller on the generator.  Try to power cycle the controller. The controllers are susceptible to surges and spikes from the utility sensing circuit.  Could have locked up the comm port of the controller.

JelloBiafra, 2019-06-05:
Thanks gor idea.  Do I power cycle by shutting down & restarting from panel?

liltux, 2019-06-05:
@JelloBiafra https://github.com/jgyates/genmon/wiki/Appendix-I-Power-Cycle-the-Evolution-or-Nexus-Controller , check out this in the wiki.  It should walk you through a power cycle.

jgyates, 2019-06-09:
@JelloBiafra, is this issue resolved? I am closing this issue for now. Feel free to post here if you still have an issue?

closed 2019-06-09 by jgyates

JelloBiafra, 2019-06-09:
Thanks for reaching out.

Sadly, it is not resolved.  I power cycled the generator.  No difference.

I pulled the pi and retested on the bench.  Works fine.

Hooked it back up.  No dice.  Switch tx/rx just to be sure.   No change.

Still receiving no data from generator.  Is there a way to test the
controller output with dmm or other equipment?

jgyates, 2019-06-09:
You may have tried this, but you could check your ground connection. Make sure you have continuity between the pi ground and the battery ground. A bad ground would cause communications to fail since the voltage reference would be different between the pi and the generator.

JelloBiafra, 2019-06-11:
Reset the ground connection just in case ... no joy.

Is there a way for me to test the controller output with a meter?

jgyates, 2019-06-13:
The transmit pin on the controller should be connected to the receive pin on the pi. This should be moving between -3V and -12V for a logic level zero and between +3v and +12v for a logic 1. If the line is idle (not transmitting or not connected) it should be between +3v and +12v. This will at least validate that the transmit transceiver is working on the controller. if the rx transceiver is blown on the controller it will be harder to determine with a multi-meter.

JelloBiafra, 2019-06-13:
> The transmit pin on the controller should be connected to the receive pin
on the pi. This should be moving between -3V and -12V for a logic level
zero and between +3v and +12v for a logic 1.
> If the line is idle (not transmitting or not connected) it should be
between +3v and +12v. This will at least validate that the transmit
transceiver is working on the controller. if the rx transceiver is blown on
the
> controller it will be harder to determine with a multi-meter.

What is best way for me to measure the voltage?  I would assume a DMM, but
your last sentence left me wondering.

jgyates, 2019-06-13:
Since the receive pin is an input it would be hard to measure a voltage that is meaningful. A DMM is the  way I would measure the transmit output.