jgyates/genmon #272: Evolution 2.0 controller update

ebovine, 2019-06-12:
I received a firmware update this evening. After the controller updated, modbus communication appears to have fallen silent. I've iterated through all addresses with modbusdump and received no register data. I don't have version information, however I am running a 16kW with Evo 2 that was up to date before this update.
Has anyone seen similar behavior with this or any prior updates?

liltux, 2019-06-12:
#198 I believe talks about the same issue.  It appears if you leave the generac wifi connected and it performs a push update the comm port for rs232 shuts off.  You should be able to power cycle the controller to restore comms on rs232 port.

ebovine, 2019-06-12:
Thanks, liltux.  That was indeed the issue. How that remains a bug in their update process since it affects any attached modbus controller, I can only guess. 
Further, I identified my mistake in missing issue #198; it was late and I missed the issue:is_open tag.  Curses!

closed 2019-06-12 by ebovine

jwtaylor310, 2019-06-12:
I encountered the same problem with my generator last night.  Resetting the controller fixed it for me.  Another thread here lists this as the procedure to reset the controller:

_Pull the F1 fuse - which is the fuse on top, and disconnect the small 2 wire white connector under the controller - it has about a 12 inch cable with a white sheath around it, this is the battery charge power supply. By removing the fuse and this cable connection the controller will be removed of all power supply. Wait a few seconds and install the fuse first and then connect the battery charge cable._

I didn't find that thread until after doing a reset using a different procedure:

1. Set controller to "Off".
2. Remove the battery cable from the negative battery post.
3. Briefly (30 seconds of so) turn off the main disconnect in the transfer panel.  This will disconnect the power for your house.  The controller display should be blank.  Turn the main disconnect back on.
4. Reconnect the battery cable.

Either of these procedures should work.  When the controller restarts it will ask you to confirm the language and time and will re-establish the WiFi connection.  Once that is done, be sure to press the "Auto" button to return to normal operation.

Good luck!