jgyates/genmon #273: Can't run gengpioin.py or genpio.py

stevehill22, 2019-06-14:
Not sure why I am seeing this error. Am I missing a file, or is something in the wrong directory?

### Expected Behavior

gengpioin.py and genpio.py should start without errors.

### Actual Behavior

When I start gengpioin.py from the command line, I get the following error:
Traceback (most recent call last):
  File "gengpioin.py", line 250, in <module>
    log.error("Error: " + str(e1))
NameError: name 'log' is not defined

Similarly, if I start genpio.py from the command line, I get the following error:
Traceback (most recent call last):
  File "gengpio.py", line 256, in <module>
    log.error("Error: " + str(e1))
NameError: name 'log' is not defined

### Steps to Reproduce (including precondition)
For gengpioin.py, from command line (genmon directory), type:
pi@raspberrypi:~/genmon $ python gengpioin.py

For gengpio.py, from command line, type:
pi@raspberrypi:~/genmon $ python gengpio.py

### Screenshot or Pictures relating to the problem (if possible)

{Please write here}

### Your Environment

- Generator Model: Evolution
- Generator Registers: Submit via the About page
- Genmon Version: V1.13.13

jgyates, 2019-06-14:
@stevehill22 

This is a bug that was introduced in V1.13.08. I checked in a new version (V1.13.14) that corrects this problem. I had inadvertently put some things out of order related to logging when making a change and  that is what is causing this error. Also, you will need to run these apps as root (sudo) since they write to the /var/log/ directory.

Sorry about the problem, let me know if you have any other issues. Thanks for reporting this.

stevehill22, 2019-06-15:
Thanks! I'm not seeing that error anymore. I'll let you know if I find any other issues.
BTW - Great program!

closed 2019-06-15 by jgyates