jgyates/genmon #274: locating controller connection help

farberm, 2019-06-15:
I have a new Evolution 2 22KW air cooled generator
I cannot identify the controller connection to hook up the serial connection
I took a few pictures and would appreciate any input.
One connection has a black/grey end connector on it (upper left part of picture). Does the grey cap need to be removed and connect into there?

- Generator Model: 7043-2 - Evolution 2 
- Generator Registers: N/A
- Genmon Version: 1.13.14

[image]
[image]
[image]

swvalenti194, 2019-06-15:
Peal off the orange tape on bottom of controller?

farberm, 2019-06-15:
Is that a question or are you sure?

jgyates, 2019-06-15:
These posts have some pictures of Evo2.0 connections:

https://github.com/jgyates/genmon/issues/234

https://github.com/jgyates/genmon/issues/198

https://github.com/jgyates/genmon/issues/149

jgyates, 2019-06-15:
No problem. FYI you can search the issues by the tag "Evolution 2.0" to find all of the posts related to Evolution 2.0.

swvalenti194, 2019-06-15:
It looks like red white blue and black next to the orange tape is where it should be. That feeding genlink?  Where does it go?

swvalenti194, 2019-06-15:
Power. Ground. TX. RX. pinout looks right

jgyates, 2019-06-15:
In the links above it shows it going to the built in wifi unit.

closed 2019-06-16 by jgyates