jgyates/genmon #275: Fuel level resetting to zero again after exercise

SethL, 2019-06-15:
I'm having the issue again where my fuel level resets to zero after an exercise.  I didn't make any changes it just decided to do it on its own.  I know how to reset but is that something you would like to track down?  I mention because if you would like the info I can submit logs otherwise I can just reset again...

jgyates, 2019-06-15:
Yes, please submit your logs and I will take a look. Thanks

SethL, 2019-06-15:
done

jgyates, 2019-06-15:
@SethL 

I think the problem is that the 22kW Evo air cooled units do not have CTs (current transformers) installed. This makes the values read from those registers unknown so I am guessing what is happening is that the software is reading random values from these registers, which makes the software think that high values read translate into high amperage and high fuel usage. This issue discusses this:

https://github.com/jgyates/genmon/issues/156

If you enable the "Disable Power / Current Display" setting, the software will ignore fuel consumption.

jgyates, 2019-06-16:
closing for now. Feel free to post here if you have any other questions on this topic.

closed 2019-06-16 by jgyates

SethL, 2019-06-16:
Hmmm   it went from 419 gal to 0 in a couple of minutes.  I really like the fuel gauge and for months it will work perfect,  I see slight usage when the gen exercises usually and then about every 3 months it goes to zero.  I guess there is no hope for me?

 

#156 <https://github.com/jgyates/genmon/issues/156>

jgyates, 2019-06-16:
From the data that I have it appears the 22kw Evo air cooled units doe not have a CT so there is not way to read the current.

SethL, 2019-06-16:
The current seems accurate (after 5kw since it shows that value as a minimum).  When I put a load on it the current by genmon seems spot on

jgyates, 2019-06-16:
If you want to post the file /etc/genmon/kwlog.txt I can look at the power readings. This could let us know if you readings from the generator are out of line and causing your fuel reading to be off. You will need to post the file from the web interface, it will get stripped off if you reply in email.

SethL, 2019-06-17:
[kwlog.txt](https://github.com/jgyates/genmon/files/3295064/kwlog.txt)
here it is.  the first of every month I do a enhanced exercise where i transfer to the gen for a few minutes.  Then the 15th of the month I do a regular exercise.

SethL, 2019-06-17:
I just looked at the file and there are about 3 months missing,, wonder if that had anything to do with it

jgyates, 2019-06-17:
it might, the kwlog.txt file is the power log. For performance reasons the file size is set to 15MB maximum. Once the maximum is reached the log is deleted  and restarted. The software should send an email when the log reaches 98 percent of the limit. You can change the limit of 15MB on the advanced settings page. If you had a lot of outages or a long one this may cause the log to go past the 15BM limit. If the log is reset it would have the opposite effect that you are seeing (your would gain fuel in the estimated reading) however if the fuel calculation ever makes a calculation that is greater than the tank size it will report zero (not a negative number) for estimated fuel in the tank. This may be what is happening. My power log is currently at 12MB so I should run into this issue in the near future. Let me think on this as there may be some issue between the power log getting reset and the fuel estimation. Off the top of my head it is at minimum an issue that needs to be documented better, and possibly handled a different way. The core design issues is balancing the relatively slow performance of a SD card based file system with a large file and keeping and accurate power log.

One thing to try in the short term is to increase the size of the power log to 25MB. In an ideal situation the power log size would not meet the limit between tank refills. Once the tank is refilled the log would need to be reset form the web interface Maintenance page.

SethL, 2019-07-08:
So I had a 6 hour outage a couple of weeks ago.  Prior to this outage I updated genmon and also increased the log file size to 25meg.  The fuel log worked fine during the outage and then for several days afterwards.  Last night I had a 20 sec power outage, this morning the the fuel is listed as 0 gal remaining.  I think it might have something to do with short outages or short run cycles maybe? My log file size is listed as .1Meg before and after the short outage so I am pretty sure that nothing was overwritten.   I uploaded my logs in case you want to take a look.  

I do have the tankutility device installed and after editing the config file genmon uses it just fine.  Can I disable the calculated fuel monitor while still keeping the kW power meter?  I like to be able to tell the load on the gen during outages so I can adjust usage so I don't overload the gen (turn the AC off when someone is taking a shower using the electric tankless water heater etc).

jgyates, 2019-07-08:
What is the speed of your SD card?

SethL, 2019-07-08:
Had to go back and look at Amazon to find it:
https://www.amazon.com/gp/product/B06XWN9Q99/ref=ppx_od_dt_b_asin_title_s02?ie=UTF8&psc=1

jgyates, 2019-07-08:
Honestly I am sort of at a loss as to why this is happening. Your logs do not show anything unusual. I will look over the code and see if I can find anything and possibly add some more checks so that if / when this happens again we can have a better understanding of the cause. Thanks for reporting this. You are the only one reporting this so I am guessing that this is an issue that does not occur often or may require some combination of events.

jgyates, 2019-07-08:
I checked in an update for 1.13.23 that, I hope, will fix the problem. I don't think the file size has anything to do with what you are seeing so you can move that back to 15MB. The code will check once an hour for the size of the log and perform some optimization. I changed the way and the occurrence of the optimization. I could see a scenario that could cause problems if the file system was responding slowly so I made some changes to avoid that situation.

SethL, 2019-07-11:
I just updated to .24.  I am wondering if the timing could be an issue because I have my exercise start at 12:00pm.  Wonder if I offset it to like 12:03 or something if that would have helped.

jgyates, 2019-07-11:
I don't know for sure, my best guess tells me that this is likely not the issue. the power code does not care about the time, only if the engine is running. There is come code that runs after an hour with no engine activity that performs some housekeeping tasks and this is what I changed and where I think that the problem could occur. Let me know if it happens again. If it does send me your logs.