jgyates/genmon #276: Can't find start and stop log entries for gengpioin.py

stevehill22, 2019-06-15:
gengpioin start and stop inputs start and stop the generator as expected, but no log entries are created. Would it be possible to also get an email notification when the input changes?

In the code, I see:
def StartCallBack(channel):
    try:
        MyClientInterface.ProcessMonitorCommand("generator: setremote=start")
        self.LogError("Sent Remote Start Command")
    except Exception as e1:
        self.LogErrorLine("Error StartCallback: " + str(e1))

MyClientInterface.ProcessMonitorCommand("generator: setremote=start") appears to be working, but 
self.LogError("Sent Remote Start Command") does not seem to do anything. 

Does that log appear somewhere else? The only log entry I see in gengpioin.log is the one created when the module starts up:
2019-06-15 15:57:26,103 : Settings:  Resitor Pull Up  Trigger Falling  bounce = 200

### Expected Behavior
"Sent Remote Start Command", "Sent Remote Stop Command", and "Sent Remote Start and Transfer Command" log entries should appear in gengpion.log

### Actual Behavior
No log entries appear for start, stop, and transfer commands.  I have not tested, but I imagine the same problem exists in gengpio.py

### Steps to Reproduce (including precondition)
After gengpioin.py is initialized, connect pin 11, 13, or 15 to ground (activate stop, start, or transfer).

### Screenshot or Pictures relating to the problem (if possible)

{Please write here}

### Your Environment

- Generator Model: Evolution
- Generator Registers: Submit via the About page
- Genmon Version: V1.13.14

jgyates, 2019-06-16:
The log prints for the non class functions were using self.LogError instead of LogError. I checked in a fix (V1.13.15). Let me know if this corrects the problem. 

I will think about the email question. My first thought is that this is sort of redundant since genmon will send emails when the generator starts, stops, etc.

stevehill22, 2019-06-16:
I noticed a couple of additional lines that use the same functions:

	Line 74:             self.LogErrorLine("Error in MyGPIOInput:init: " + str(gpio) + " : " + str(e1))
	Line 82:                     self.LogError('Edge detected on pin ' + str(self.GPIO))
	Line 89:             self.LogErrorLine("Error GPIOInputMonitor: " + str(self.GPIO)  + ": "+ str(e1))
	Line 99:             self.LogErrorLine("Error in Close: " + str(e1))

 Are these OK as is, or should they also be changed?

jgyates, 2019-06-16:
All of these are OK. they are part of the MyGPIOInput class which inherits the log functions from it's parent so the "self." is needed there.

stevehill22, 2019-06-16:
This fix worked. Thanks!

closed 2019-06-16 by stevehill22