jgyates/genmon #279: Feature Request: Support for Generac/Tank Utility fuel level monitor

jwtaylor310, 2019-06-18:
Please consider adding support in GenMon for the Generac 7005 Propane Tank Fuel Level Monitor, reference:  http://www.generac.com/GeneracCorporate/media/Library/content/all-products/parts-and-accessories/Tank%20Utility/Generac-Tank-Utility_spec-sheet.pdf 

That would provide a way of monitoring and alerting on low fuel level using a device that is independent of the specific generator model.  The device clips into a "Remote Ready Gauge" which is available on many LP tanks.  Older tanks can generally be retrofitted with a Remote Ready Gauge for under $50 and without emptying the tank.  The device reads the levels every 6 hours and transmits the information daily via WiFi to Tank Utility's web site.  (Generac is private-labelling the Tank Utility system).

Tank Utility has published a straight-forward API here:  http://apidocs.tankutility.com/  

I have written a Python module that uses the API to read the level and related information for my generator.  It could be used as a starting point for adding support for the device to GenMon.  If you would find it helpful, I would be pleased to post a copy for you.  Also, if you need a device to test against I can provide you with access to my Tank Utility account.

Thanks for providing a great product and outstanding support!

jgyates, 2019-06-18:
@jwtaylor310 

that sound like a great idea. If you can send me what you have now I will use this as a starting point. You can email it to me at generatormonitor.software@gmail.com

jwtaylor310, 2019-06-18:
Email sent with sample program.

jgyates, 2019-06-19:
I checked in an update (V1.13.19) that adds this functionality to the Add-On page. Let me know how it works for you. Once you enable it and input your settings on the about page, then save it. Once saved you will need to refresh your browser for the gauge to show up. Let me know if you have any questions.

jwtaylor310, 2019-06-19:
Wow!  That was fast – thank you!

 

Unfortunately, the new gauge shows Empty and the gentankdata.log shows “Invalid user name or password, exiting”.  I double-checked my credentials and they appear to be OK.  I’m assuming that the Username/Password doesn’t require quotes:

 

 

 

There might be a version issue – I couldn’t find V1.13.19, the latest that I found on github is V1.13.17 which downloaded automatically when I selected “Upgrade to latest version” on the “About” page.  What am I missing?

 

--John

jgyates, 2019-06-19:
Sorry, typo on my part. the current version is 1.13.17. 

I think the problem is I have a typo in the file /etc/genmon/gentankutil.conf.

The first line should be 

[gentankutil]

not 

[gentankdata]

make this change and restart genmon and try it again.

jgyates, 2019-06-19:
also once you make the change and restart you can validate that the settings made in the UI are in the /etc/genmon/gentankutil.conf file.

The error you are seeing is when no username or password is present.

jwtaylor310, 2019-06-19:
Hi;

 

No joy.

 

I changed the header in /etc/genmon/gentankutil.conf:

 

 

[gentankdata]

 

# number of minutes between polls for tank data. Note, if the API is

# called too often and error may occur in an attempt to limit the traffic

# by tankutility.com. 

poll_frequency = 60

 

# username

# The username of the online account at https://app.tankutility.com

username = john@sunbeachandocean.com

 

# password

# The password of the online account

password = chek*tank

 

# tank name

# This value is used to match the tank name with the on-line tank name

# if multiple tanks are installed. If empty then the first tank found

#  is used.

tank_name = Generator

 

 

I then restarted genmon and the web browser.  Same error in the log file.  Just to be sure, I made the same change in /home/pi/genmon/conf/gentankutil.conf.  Restarted and no improvement.  The configuration information does appear to be copied correctly from the web UI.

 

The error messages in the log are the same as previously.

 

--John

jwtaylor310, 2019-06-19:
My bad … I was using raspian’s  gui text editor to change the header in the conf file and it didn’t save the changes because the file is owned by root.  I used “sudo nano” instead and everything is now working.  

 

Looks great!

 

My apologies for the false alarm

 

--John

jgyates, 2019-06-20:
No worries. I actually did find an error in gentankutil.py, but it only occurs in a path that handles and error. I checked in V1.13.18 that corrects this.

Closing for now. Let me know if you have any other issues.

closed 2019-06-20 by jgyates

jwtaylor310, 2019-06-20:
Looking good here.  Thank you!  --John

SethL, 2019-06-26:
I just purchased the tank utility fuel monitor, waiting for it to be delivered.  Will it replace the mathematical fuel gauge that I have running now in genmon or will it show a separate one for the tankutility?

jwtaylor310, 2019-06-26:
Hello, Seth;

 

On my copy of Genmon the tank utility level shows up as a separate gauge labelled “External Tank”.  If your Genmon currently shows the “Estimated Fuel” gauge both will be displayed.

 

However, I have a 12kW air cooled generator that does not have a current transformer so the current-related items (“Power Output” gauge, “kW Output” graph and “Estimated Fuel”) aren’t usable.  To hide those items, I have turned on the “Disable Power/Current Display” switch on the “Settings” menu.  With that switch turned on, only the “External Tank” gauge showing the Tank Utility reading is displayed.

 

FYI, the configuration information for the Tank Utility monitor appears in the “Add-Ons” section of the Genmon web page.  When your Tank Utility monitor has been installed and linked to the Tank Utility web site, go to Add-Ons in Genmon, set the “External Tank Fuel Monitor” to “ON” and enter the configuration information.  Then refresh the browser and you should see the new gauge on the Genmon “Status” screen.  It may take a few seconds for the fuel level to update.

 

The gauge has been working perfectly on my system since the feature was added to Genmon.  It’s a great addition to the program!

 

--John

jgyates, 2019-06-26:
@jwtaylor310  is correct, it is a separate gauge. You can disable the estimated fuel gauge on the settings page and enable / disable the Tank Utility gauge on the add-ons page.