jgyates/genmon #280: System Initializing. Not Receiving Data Error

farberm, 2019-06-23:
### Expected Behavior

I have installed the latest software on a Pi2. I have tested the serial cable all the way to the molex connector and it passes. I have made my cable connector accordingly for an Evolution 2 generator (new) as described in other posts (cut wires for pin 7 and 8 and used them to connect to the serial cable pin 2 and 3 respectively.-post 234 wiring images). I am receiving no data to genmon...
[image]

### Actual Behavior . No Data
Monitor Page Says System Initializing.Not Receiving Data

Pi starts up fine and genmon run. I can loginto the website without issues
After connecting the serial cable, I disconnected the fuse and battery cable and plugged the fuse and battery cable back in as directed under post 198

### Your Environment
- Generator Model: Evolution 2 22 kw air colled

- Generator Registers: Submit via the About page - done
- Genmon Version: 1.13.18

[image]

liltux, 2019-06-23:
How is your pi powered?  Is there a common ground between the two?  If your pi is powered from a 110 source you will need a gnd ran too in addition to tx and rx.  Of the pi is powered from the battery using a DC step down then the pi should have the same gnd as the controller.  Also what type of rs232 to ttl are you using?

jgyates, 2019-06-23:
In addition to checking your ground you should validate the serial port is configured properly with the [serialconfig.py](https://github.com/jgyates/genmon/wiki/1----Software-Overview#otherappsserialconfigpy-optional) program. This will validate that you have disabled other linux system apps that use the serial port. I received your logs and your /var/log/myserial.log file contains many of these errors:

    Resetting SerialDevice:ReadThread Error: /dev/serial0:device reports readiness to read but returned no data (device disconnected or multiple access on port?)

When using the built in port this typically means that the tty process or bluetooth is trying to use the serial port. the serialconfig.py program should correct that if that is the issue.

farberm, 2019-06-23:
The pi is power by a regular adapter. I ran the 18AWG/2 wire into the basement so it is inside. I did not power it from the generator or the battery. So yes its a 110 source. 

I purchased these two items for the cable.
Serial Board:
https://www.amazon.com/gp/product/B01JYNHNW6/ref=oh_aui_detailpage_o06_s00?ie=UTF8&psc=1 <https://www.amazon.com/gp/product/B01JYNHNW6/ref=oh_aui_detailpage_o06_s00?ie=UTF8&psc=1>

Breakout board:
https://www.amazon.com/gp/product/B07437GS66/ref=oh_aui_detailpage_o09_s03?ie=UTF8&psc=1. <https://www.amazon.com/gp/product/B07437GS66/ref=oh_aui_detailpage_o09_s03?ie=UTF8&psc=1.>

So what if I power the Pi from a POE adapter? Do I need to run a gnd wire and if so from where to where?

If I power from the 110 adapter where do I run the gnd wire  from and to?

Thanks for all your help.

The image below is before I connected the long 18AWG/2 wire to the generator. I checked it first here and it passed and then at the molex after it was connected.

farberm, 2019-06-23:
I did run that. And everything was ok.  I had the Genmon software running initially when I did the serialtest.py so I turned that off. The cable then passed without issues.

liltux, 2019-06-23:
To get a common gnd for signal, all you have to do is run a gnd wire to the generator and to pin 5 on your rs232 adapter.  I have done this before and works.  Regardless how the pi is powered.  Ground at the generator can either be from the common ground plane or from one of the ground pins on the 8 pin mini fit jr connector.

farberm, 2019-06-23:
Excuse my ignorance…and Noob questions..

Can you be more specific about where the common ground plane is or where the 8 pin mini fit jr connector is located on the generator?

jgyates, 2019-06-23:
@farberm ,

no worries on the questions. 

The connector is the same connector that you connected TX and RX to the RS-232 cable. This page has some diagrams as well as pin orientation for the connector:

https://github.com/jgyates/genmon/wiki/3.1--Making-a-Cable

Molex is the company is the company that makes the connector, mini fit jr is their product name for this type of connector.

You need to connect the ground from the pi to the ground on the generator controller. The reason for this is that you can have RS232 compatible voltages, but the voltage is measured from the ground, so if the grounds are not common then what is 5 volts on one system will be interpreted as 5 volts on the other system.

You can tie the ground on the pi at either the RS-232 connector (pin 5) or get the pi ground from the GPIO header. This ground should be connected to the generator controller, either to the mini fit jr connector that you are using now or to the battery ground (negative) terminal.

liltux, 2019-06-23:
The 8 pin connector is the 8 pin connector your tx and rx are connected.  Off the top of my head I am not sure of the gnd pin location but I believe it is list under the the wiki.  The new Evolution 2.0 generators h attached all the ground wires to the steel ground connections instead of a bus bar.  If you remove the top black panel covering the controller you will see the ground connections.

farberm, 2019-06-23:
So if I understand correctly. I can either ...

1. purchase the battery power adapter that you mention in the GitHub wiki and in that case I would not need to connect a ground but move the pi to the generator location and use wifi to connect.

2. Run a Third 18AWG wire from the negative battery terminal (this seems the easiest) to pin 5 on my serial board or the GPIO header. If I go this route it is easier to goto pin 5 on the breakout board

farberm, 2019-06-23:
One other question - Can I just use ethernet cable to run the Rx, Tx, and Ground. I cannot find 18/3 in flat cable but I can get flat Cat 6 cable to do the same thing correct?

jgyates, 2019-06-23:
that will probably be fine,  assuming you are within the distance specification of RS-232

farberm, 2019-06-23:
It would be about 27 feet so I think that would work?

jgyates, 2019-06-25:
I am closing this issue for now. Feel free to post here if you have any other updates or questions.

closed 2019-06-25 by jgyates

farberm, 2019-11-25:
If I am using the generator battery to power my pi do I need to run a ground to the rs232 board on Pin   5 from the battery ground in addition to the rx  and tx wires . if I do not then there are only 2 wires from the co troller to the serial board correct?

I am using the motorcycle usb power cord that is mention on the boards to connect to the battery.

Thanks in advance.

Sent from my iPhone

Sent from my iPhone

jgyates, 2019-11-25:
I would recommend it. In theory if the motorcycle supply has a common ground between the 12v and 5v side then no, you don't need it. I use the ground wire on the RS232 side to the controller since I don't know the way the power supply is wired and I don't know how the controller supplies power to the connector, so this is the safest bet in my view.

farberm, 2019-11-25:
Ok thx