jgyates/genmon #281: alerts

jwtaylor310, 2019-06-26:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below.

***NOTE:*** If you are having issues with your serial connection, please read this page before posting:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

If you are having other issues, please see the following page:

https://github.com/jgyates/genmon/wiki/3.5---General-Troubleshooting

If you need to send you logs and registers to the developer, if you email is setup and working properly you can click send your logs on the About page in the web interface.

### Expected Behavior

{Please write here}

### Actual Behavior

{Please write here}

### Steps to Reproduce (including precondition)

{Please write here}

### Screenshot or Pictures relating to the problem (if possible)

{Please write here}

### Your Environment

- Generator Model: {Please write here}
- Generator Registers: Submit via the About page
- Genmon Version: {Please write here}

jwtaylor310, 2019-06-26:
Generator: 12kW air cooled, Evolution 2.0 Controller
Genmon Version: V1.13.18

I have been receiving email alerts with no apparent cause.  They occur in pairs about 10 seconds apart every 2-3 days.   Here is a pair received this morning.  The only differences that I see are the timestamp and the "No Alarms" information at the bottom of each email.  I am unaware of any possible environmental issues like momentary power outages or sags.  The statistics on the "Monitor" page show no data communication errors and I don't see any unusual entries in the logs on the Raspberry Pi that is running Genmon.  Any suggestions?  Thank you!

-----------------------------
Time: 05:58:16 AM
Date: Wed 26-Jun-2019

NOTE: This message is a notice that the state of the generator has changed. The system is not in alarm.

Status : 

    Engine : 
        Switch State : Auto
        Engine State : Off - Ready
        Battery Voltage : 13.50 V
        RPM : 0 
        Frequency : 0.00 Hz
        Output Voltage : 0 V
        Active Rotor Poles (Calculated) : 0 

    Line : 
        Utility Voltage : 246 V
        Utility Voltage Max : 250 V
        Utility Voltage Min : 239 V
        Utility Threshold Voltage : 156 V

    Last Log Entries : 

        Logs : 
            Alarm Log : 06/12/19 11:46:46 RPM Sensor (Twin Cylinder+Cranking) : Alarm Code: 1505
            Service Log : 
            Run Log : 06/19/19 12:05:57 Stopped - Auto 

    Time : 
        Monitor Time : Wednesday June 26, 2019 05:58:14
        Generator Time : Wednesday June 26, 2019 05:58

No Alarms: 0001:20200000
--------------------------------------
Time: 05:58:26 AM
Date: Wed 26-Jun-2019

NOTE: This message is a notice that the state of the generator has changed. The system is not in alarm.

Status : 

    Engine : 
        Switch State : Auto
        Engine State : Off - Ready
        Battery Voltage : 13.50 V
        RPM : 0 
        Frequency : 0.00 Hz
        Output Voltage : 0 V
        Active Rotor Poles (Calculated) : 0 

    Line : 
        Utility Voltage : 245 V
        Utility Voltage Max : 250 V
        Utility Voltage Min : 239 V
        Utility Threshold Voltage : 156 V

    Last Log Entries : 

        Logs : 
            Alarm Log : 06/12/19 11:46:46 RPM Sensor (Twin Cylinder+Cranking) : Alarm Code: 1505
            Service Log : 
            Run Log : 06/19/19 12:05:57 Stopped - Auto 

    Time : 
        Monitor Time : Wednesday June 26, 2019 05:58:25
        Generator Time : Wednesday June 26, 2019 05:58

No Alarms: 0001:00000000
--------------------------------------

jgyates, 2019-06-26:
@jwtaylor310 

This is a know issue (at least know to me, don't know about generac). The genmon software sends an email status every time modbus register 0001 changes, even if it doesn't know what the values represent. It appears that the the Evolution controller 2.0 is sending some either random or unknown data in this register. The first due to the random change of the register in the firmware and the 2nd email is the register going back to normal (all zeros denotes the engine is off and the generator is ready)

this post has some more details:

https://github.com/jgyates/genmon/issues/196

jwtaylor310, 2019-06-26:
Thanks for the update.  I missed the previous thread.

 

Would it be possible to put a 15 second delay before responding to the register change?  I would expect that any “real” change would last longer than that…..

 

--John 

 

#196 <https://github.com/jgyates/genmon/issues/196>

jgyates, 2019-06-26:
I would rather handle this in a different way, for example add an option for Evolution 2.0 that would ignore unknown values in register 0001. A 15 second delay would end up removing some other valid status. I had considered doing this (adding and option to ignore the messages for Evo 2.0) but was concerned that this may be valid info from the controller (for example when a firmware update is complete when the generac wifi adapter is installed) so I wanted to see if a pattern emerged, but I don't have enough data to say that there is something meaningful with the odd results we are seeing on Evo 2.0.

I will try to add an option to disable these messages.

jgyates, 2019-06-26:
I checked in V1.13.20 that added an option on the advanced page that allows the software to ignore unknown values from register 0001. Once you enable this let me know if you have any issues. I am closing this issue for now but feel free to post here if you have any issues. 

I also updated the Known Issues section in the wiki with info about this topic.

closed 2019-06-26 by jgyates

jwtaylor310, 2019-06-26:
I will give it a try.  Thank you for the fast response!  --John

jwtaylor310, 2019-07-07:
I no longer receive alerts with unknown values in register 0001.  However, I am still getting occasional email 'alerts' showing no alarms and status 0001:00000000.  Is it possible that the system is suppressing the email when the status goes to an unknown value but is then detecting the return to normal status and sending an email to show the change?  In other words, instead of receiving pairs of emails as before, is the system now suppressing the first email but not the "return to normal" email?

jgyates, 2019-07-07:
I checked in an update that should fix this. Update to the latest and let me know if you still have this issue.

jwtaylor310, 2019-07-07:
I have installed the update and will  let you know in a few days if the spurious alerts are gone.  Thanks for the very fast response!

jwtaylor310, 2019-07-15:
It's been a week since the update and I haven't received any new spurious alerts.  Looks like the problem is fixed.  Thank you!

jwtaylor310, 2019-07-17:
Looks like my "OK" was premature.  Today my generator ran its exercise cycle and I didn't receive any notifications.  Later, I disconnected commercial power and ran the house on generator for about 10 minutes and didn't receive notification of the start or stop events.  The Genmon web page shows the correct status and the events are correctly listed in the log.  I confirmed that I have all alerts enabled.  It appears that the patch killed valid alerts as well as bogus ones....

jgyates, 2019-07-17:
It is working for me when I test. I have a test program that can simulate a generator using serial over IP. When I load Evolution 2.0 registers and manually change the contents of register 0001 it works as expected (including ignoring the random values when this option is enabled). If you restart genmon, do you get two emails ? On that notifies you of genmon starting and a 2nd on of the initial status?

jgyates, 2019-07-17:
that should read: One that notifies you of genmon starting and a 2nd on of the initial status?

jwtaylor310, 2019-07-17:
The problem was on my end – my ISP timed out the password on the email account that I use for Genmon notifications.  I provided the ISP with a new password & updated the email PW in Genmon and I am now getting normal alerts.  My apologies for blaming Genmon for the problem!

jgyates, 2019-07-17:
No worries, I am glad it is working for you.

jwtaylor310, 2019-07-19:
Unfortunately, I just got a new “bogus” alert email followed 12 seconds later by the reset email.  The register code at the bottom of the alert is 0001:00000074.  I’m running V1.13.22.  --John

jgyates, 2019-07-19:
This may be a bogus value, however the format is different than the other bogus values were in a different format. The contents of register 0001 look like this:

XXXYXXZZ

Where Y is the  Engine State and ZZ is the switch and alarm type and X is not used. In the past the bogus values were all in the digits in the X place. The value 00000074 does not follow this pattern and is why you got an email.

The when you enable the setting to ignore unknown values it ignores any values in the X digits but not the others since the software is not aware of all alarm codes, only the ones that people report.

Please send your registers since this may be helpful in debugging this.

Also check your logs to make sure there is no event that corresponds to the time that you received this email. I was thinking about adding some code to detect when the firmware version changed on Evolution 2.0 and a guess may be that this value corresponds to something related to wifi or firmware updates.

jgyates, 2019-07-19:
I checked in an update that will send an email if the Evolution firmware version changes. After you send  your registers and logs we can determine if some other change is needed.

jwtaylor310, 2019-07-19:
I have sent a copy of my registers and logs.  Here's a copy of the email alert with the time stamp and register info.   A related anomaly is that the "NOTE:" line in the email states that the system is not in alarm but the "Engine Switch State: indicates "System in Alarm".  I haven't found anything to suggest that there was any actual problem.  

The generator is also configured for Generac's Mobile Link and it hasn't reported any changes since 7/17 when I forced a power outage for testing purposes.  That matches the "Stopped - Auto" entry which is listed as the Last Log Entry in the Run Log in the Genmon alert email.

In addition to the bogus alert, I received (as expected) a second "back to normal" email that was timestamped 02:18:26 PM with register value 0001:00000000.  

I will install the updated release in a few minutes.

Thank you for working on this!

--John

-----Original Message-----
From: Genmon
Sent: Friday, July 19, 2019 2:18 PM
To: <deleted>
Subject: Generator Notice: 37 W Point Drive

Time: 02:18:14 PM
Date: Fri 19-Jul-2019

NOTE: This message is a notice that the state of the generator has changed. The system is not in alarm.

Status : 

    Engine : 
        Switch State : System in Alarm
        Engine State : Off - Ready
        Battery Voltage : 13.20 V
        RPM : 0 
        Frequency : 0.00 Hz
        Output Voltage : 0 V
        Active Rotor Poles (Calculated) : 0 

    Line : 
        Utility Voltage : 243 V
        Utility Voltage Max : 17741 V
        Utility Voltage Min : 168 V
        Utility Threshold Voltage : 156 V

    Last Log Entries : 

        Logs : 
            Alarm Log : 06/12/19 11:46:46 RPM Sensor (Twin Cylinder+Cranking) : Alarm Code: 1505
            Service Log : 
            Run Log : 07/17/19 15:18:20 Stopped - Auto 

    Time : 
        Monitor Time : Friday July 19, 2019 14:18:12
        Generator Time : Friday July 19, 2019 14:18

No Alarms: 0001:00000074

jgyates, 2019-07-19:
I made one additional checked that should tighten it up a bit more regarding what values are allowed. I am hesitant to just disallow anything except known values as this is the primary way I add new alarm values to the software for all other modules. 

Just check out the lastest ( the version number is still 1.13.26)

The email you got with conflicting info (alarm vs no alarm) is likely due to the value 000000074 is staying in the register for and extremely short period of time.

jwtaylor310, 2019-07-20:
Hi;

 

I have installed the latest version and will let you know how it works.

 

I agree with your hesitation to do a blanked “disallow”.  The occasional false alert isn’t a serious problem … though it seems they always occur in the wee hours of the morning when I should be sleeping!

 

--John