jgyates/genmon #282: New install with unexpected email messages

mdetweiler, 2019-06-26:
Great product and really well done!  

Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below.

***NOTE:*** If you are having issues with your serial connection, please read this page before posting:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

If you are having other issues, please see the following page:

https://github.com/jgyates/genmon/wiki/3.5---General-Troubleshooting

If you need to send you logs and registers to the developer, if you email is setup and working properly you can click send your logs on the About page in the web interface.

### Expected Behavior

I expected a message that the Generator was exercising.

### Actual Behavior

I received two messages that were unexpected:

1st message: 
Time: 11:05:59 AM
Date: Wed 26-Jun-2019
Generator Monitor is not communicating with the controller at Woodland

2nd message:
Time: 11:12:54 AM
Date: Wed 26-Jun-2019

NOTE: This message is a notice that the state of the generator has changed. The system is not in alarm.

Status : 

    Engine : 
        Switch State : Auto
        Engine State : Cranking
        Battery Voltage : 14.10 V
        RPM : 0 
        Frequency : 0.00 Hz
        Output Voltage : 0 V
        Active Rotor Poles (Calculated) : 0 

    Line : 
        Utility Voltage : 234 V
        Utility Voltage Max : 244 V
        Utility Voltage Min : 231 V
        Utility Threshold Voltage : 144 V

    Last Log Entries : 

        Logs : 
            Alarm Log : 05/22/19 10:10:52 Low Fuel Pressure 
            Run Log : 06/24/19 13:33:59 Stopped - Auto 

    Time : 
        Monitor Time : Wednesday June 26, 2019 11:12:51
        Generator Time : Wednesday June 26, 2019 09:59

No Alarms: 0001:00020000

### Steps to Reproduce (including precondition)

{Please write here}

### Screenshot or Pictures relating to the problem (if possible)

{Please write here}

### Your Environment

- Generator Model: 22kw liquid cooled
- Generator Registers: Submit via the About page
- Genmon Version: {Please write here}

jgyates, 2019-06-26:
@mdetweiler 

The first message is sent when the software is not receiving responses back from the controller after a specific period of time (2 seconds for serial , 8 seconds for serial over TCP/IP). To validate this is actually happening you can check the Monitor page and see if you see an time out errors under Serial Stats. 

The second message could be being sent for one of three reasons that I can think of:

   1) You generator is attempting to crank but failing (check your outage logs and exercise schedule to see if an outage or exercise is scheduled for this time)
   2) The problem you are having with serial communications is impairing the communications  and as a result the software requested a register value, it timed out and the then the software requested another register but the controller responded with the first request. This could cause the software to think an a value returned was valid when it is not. 
  3) If you see values that are unknown to the software (other than cranking) and you have an Evolution 2.0. this could be a bug in the firmware. I have seen the Evo 2.0 controller send odd results periodically for register 0001, but never that correspond to something meaningful like Cranking.

I would investigate the serial comms first, once you get your comms reliable then let's see if the other issue goes away..

mdetweiler, 2019-06-26:
I should have mentioned that the messages were sent at almost the exact time that the generator is scheduled to exercise.   I will try to debug as suggested.

Sent from Bloomberg Professional for Android

----- Original Message -----
From: jgyates <reply@reply.github.com>
At: 26-Jun-2019 15:01:26

@mdetweiler 
The first message is sent when the software is not receiving responses back from the controller after a specific period of time (2 seconds for serial , 8 seconds for serial over TCP/IP). To validate this is actually happening you can check the Monitor page and see if you see an time out errors under Serial Stats. 
The second message could be being sent for one of three reasons that I can think of: 
                                                                                       1)You generator is attempting to crank but failing (check your outage logs and exercise schedule to see if an outage or exercise is scheduled for this time) 
                                                                                       2)The problem you are having with serial communications is impairing the communications  and as a result the software requested a register value, it timed out and the then the software requested another register but the controller responded with the first request. This could cause the software to think an a value returned was valid when it is not. 
                                                                                       3)If you see values that are unknown to the software (other than cranking) and you have an Evolution 2.0. this could be a bug in the firmware. I have seen the Evo 2.0 controller send odd results periodically for register 0001, but never that correspond to something meaningful like Cranking. 
I would investigate the serial comms first, once you get your comms reliable then let's see if the other issue goes away..

jgyates, 2019-06-26:
If these were send at the time of exercise, then you issue may be that vibrations from the generator are  causing loose connection to become intermittent. The message "No Alarms: 0001:00020000" is the status for the engine cranking so this is normal (assuming your engine cranked as it should).

I don't know how you are cabling you serial communications, but in an early iteration of my cable I was crimping a molex cable myself, which I did not do very well. the result was I got comms errors when the generator was cranking or running. I switch to this:

https://www.amazon.com/gp/product/B000M802RG

and it made my comm errors due to vibrations go away. Ended up using a few of these to make my cable and just cut off the end of one and connected that to the RS-232 converter.

mdetweiler, 2019-06-27:
The monitor page shows that it is not receiving data, but odd that it must be able to communicate at some point because the logs are there and the battery status seems to update along with some other things.  I'm using a fully configured system that I purchased on eBay.  The connectors seem solid.  Any advice on what to try next?  Is there a way to update the firmware on the controller? Attached is a screenshot of the monitor page.

Sent from Bloomberg Professional for Android

----- Original Message -----
From: jgyates <reply@reply.github.com>
At: 26-Jun-2019 16:47:29

If these were send at the time of exercise, then you issue may be that vibrations from the generator are  causing loose connection to become intermittent. The message "No Alarms: 0001:00020000" is the status for the engine cranking so this is normal (assuming your engine cranked as it should). 
I don't know how you are cabling you serial communications, but in an early iteration of my cable I was crimping a molex cable myself, which I did not do very well. the result was I got comms errors when the generator was cranking or running. I switch to this: 
https://www.amazon.com/gp/product/B000M802RG 
and it made my comm errors due to vibrations go away. Ended up using a few of these to make my cable and just cut off the end of one and connected that to the RS-232 converter.

mdetweiler, 2019-06-27:
I have a theory about the intermittency.  I made a small modification to the system by adding a USB temp and humidity sensor.  I suspect it may be starving the USB bus power.  I will disconnect it and see...

jgyates, 2019-06-27:
You can send your logs via the about page and I can see if there is anything odd going on. Also, you can look at the monitor page under platform stats. These three values should all read OK:

        Pi CPU Frequency Throttling : OK
        Pi ARM Frequency Cap : OK
        Pi Undervoltage : OK

You may want to look at the RS-232 converter to make sure thoes connections are still solid. If one of the connections is loose on that it could cause this problem.

jgyates, 2019-06-30:
I am closing this issue for now. if you have any other data or questions feel free to post to this thread.

closed 2019-06-30 by jgyates