jgyates/genmon #283: Enhanced exercise oddity

UPS99, 2019-06-27:
The Enhanced Exercise mode apparently won't work if the system is in alert status.  I had received an alert for the informational Service Schedule A, which needs acknowledgement from the front panel to clear, and had not done that yet.  Genmon showed a red highlighted Status button and also in other places - a nice feature.  The time set for Enhanced Exercise was passed with nothing happening.  I tried setting the time ahead 3 times to be sure.  Standard built-in exercise mode did work OK, though.  I reset the alert at the front panel and again scheduled Enhanced Exercise and it worked fine.  This could be in the Genmon code, or, of course, could be something in the Evolution code blocking remote start if the status is in alert. 

- Generator Model: 6244
- Generator Registers: Submit via the About page
- Genmon Version:  V1.13.18

jgyates, 2019-06-27:
It is my understanding that this is a function of the controller, i.e. it will not start if the system is in alarm. The generator will not start from the front panel either unless the alarm is acknowledged by pressing stop, then auto or manual.

UPS99, 2019-06-27:
I thought that it might be in the Evolution code.  Those non-technical reminder alerts were a very poor engineering decision by Generac.  That type of alert should have been restricted to reporting actual system problems and not scaring non-technical owners.  They are fortunately infrequent.  Maybe this should be listed as a known issue?

UPS99, 2019-06-27:
I think this can be closed.

jgyates, 2019-06-27:
I updated the wiki documentation for genexercise.py to reflect your suggestion.

closed 2019-06-27 by jgyates