jgyates/genmon #284: GenMon stopped working several weeks ago - Evol 2.0

jimland58, 2019-07-02:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below.

***NOTE:*** If you are having issues with your serial connection, please read this page before posting:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

If you are having other issues, please see the following page:

https://github.com/jgyates/genmon/wiki/3.5---General-Troubleshooting

If you need to send you logs and registers to the developer, if you email is setup and working properly you can click send your logs on the About page in the web interface.

### Expected Behavior
Work as it was several weeks ago

### Actual Behavior
Was working fine for many months and about 2 weeks ago noticed I wasn't getting my weekly text notifications from Genmon of the generator doing its weekly exercise.  Went to the Genmon Webserver and noticed the Genmon Time and Generator Time were several weeks apart.  I saw that Genmon was several versions behind and so I clicked on link to update software.  When I tried going back to the website I get "Loading Generator Monitor" and circle spinning.  Cycled power on the Raspberry Pi and started getting one message every 10 seconds or so with the SUBJ line stating "Generator Monitor Starting at ......".   When I go back to the website I still get "Loading Generator Monitor" and circle spinning.

### Steps to Reproduce (including precondition)

{Please write here}

### Screenshot or Pictures relating to the problem (if possible)

{Please write here}

### Your Environment

- Generator Model: Generac 22kw w/ Evo 2.0
- Generator Registers: Submit via the About page
- Genmon Version: Can't tell

jgyates, 2019-07-02:
What browser and operating system are you? i.e. Windows Chrome, Mac Firefox, etc

jgyates, 2019-07-02:
the reason I ask is that know issue #10 describes the latter part of what you are seeing:

https://github.com/jgyates/genmon/wiki/Appendix-D-Known-Issues

It could be that you have an issue with your serial comms an if you updated with Firefox for windows (or palemoon browser for windows) then the browser tried to issue the upgrade several times in a row.

jimland58, 2019-07-02:
Does the same thing with Internet Explorer, Firefox and Chrome

jimland58, 2019-07-02:
I did the last update with Firefox.  When I use IE, Firefox or Chrome, I get the "Loading Generator Monitor" and circle spinning.

jgyates, 2019-07-02:
Is this on Windows? 

You did not mention this so i dont know if you did or not, but Firefox must be completely closed. If the page is still up or attempting to be displayed in firefox for windows (assuming windows, you did not say in your response) and no other browser will work if Firefox is still loaded.

Try this:

1) Quit firefox completely on all system.
2) log into your pi via ssh 
3) type :

      cd genmon 
      bash ./startgenmon.sh restart 

3) Try to bring up the web page in a browser (let's try Chrome first)

Let me know how it goes.

jgyates, 2019-07-07:
@jimland58 

Are you still having an issue?

jimland58, 2019-07-08:
Thanks for checking!  Yes, still doing the same thing.  I shut down Firefox and used PuTTY to log into the Raspberry Pi and did a restart using the command bash ./startgenmon.sh restart    Immediately I started getting text messages saying SUBJ: Generator Monitor Starting at ..... and I would get a new text message about every 10 seconds saying the same thing.  I started Chrome and tried opening the web browser and I would get a spinning circle and text saying “Loading Generator Monitor”.   After about 10 of these text messages I went and issued a shudown command sudo bash startgenmon.sh stop  to stop the text messages.

From: jgyates 
Sent: Sunday, July 07, 2019 9:56 AM
To: jgyates/genmon 
Cc: jimland58 ; Mention 
Subject: Re: [jgyates/genmon] GenMon stopped working several weeks ago - Evol 2.0 (#284)

@jimland58

Are you still having an issue?

—

---
This email has been checked for viruses by AVG.
https://www.avg.com

jimland58, 2019-07-08:
FYI,  I have a Windows machine

From: jgyates 
Sent: Sunday, July 07, 2019 9:56 AM
To: jgyates/genmon 
Cc: jimland58 ; Mention 
Subject: Re: [jgyates/genmon] GenMon stopped working several weeks ago - Evol 2.0 (#284)

@jimland58

Are you still having an issue?

—

---
This email has been checked for viruses by AVG.
https://www.avg.com

jgyates, 2019-07-08:
OK. Let's try this:

1) Log into the pi via ssh 
2) change directories to the genmon directory and stop genmon 

     cd genmon
     ./startgenmon.sh stop

3) update to the latest version

     ./genmonmaint.sh -u 

     you will need to type Y to make the update proceed 

4) make a backup of your settings:

     ./genmonmaint.sh -b 

    This will create a file named genmon_backup.tar.gz that has your config files backed up.

5) copy a fresh set of config files:

     ./genmonmaint.sh -s

6) unzip you old config files:

     mkdir /home/pi/backup 
     tar -zxvf ./genmon_backup.tar.gz -C /home/pi/backup

7) copy over your mail settings and your power log

     sudo cp /home/pi/backup/genmon_backup/mymail.conf /etc/genmon
     sudo cp /home/pi/backup/genmon_backup/kwlog.txt /etc/genmon

      if you have any other add on config files you can copy them over as well. Do not copy over genloader.conf and genmon.conf. You can also just setup your add on programs again if that is easier.

8) Load genmon with the fresh copy of genloader.conf and genmon.conf that we copied over in step 5 

      ./startgenmon.sh restart 

If your serial connection is not working then you may see a delay before the web interface comes up. The delay should be no more than a couple of minutes.

Let me know if the web interface comes up and if you are having an issue with serial communications (are there serial timeout errors on the Monitor page). IF it does come up send me your logs on the About page.

If the web interface does not come up then from the genmon directory on the pi type:

     python ClientInterface.py 

This will give you a prompt that has the status of the program / generator. From the prompt type:

    generator: sendlogfiles 

Also, let me know what the prompt says. You can type :

   generator: monitor

and it will display the contents of the Monitor page. copy and past that output to this thread.

jgyates, 2019-07-08:
forgot to mention, to exit ClientInterface.py type "exit"

jimland58, 2019-07-08:
At step 4 I got the following response: 

pi@GenMon:~/genmon $ ./genmonmaint.sh -b
Backup genmon...
rm: cannot remove ‘genmon_backup’: No such file or directory
rm: cannot remove ‘genmon_backup.tar.gz’: No such file or directory
cp: cannot stat ‘/etc/genmon/maintlog.json’: No such file or directory
genmon_backup/
genmon_backup/genmqtt.conf
genmon_backup/genmon.conf
genmon_backup/gengpioin.conf
genmon_backup/mymail.conf
genmon_backup/mymodem.conf
genmon_backup/genemail2sms.conf
genmon_backup/outage.txt
genmon_backup/gensms.conf
genmon_backup/genpushover.conf
genmon_backup/genloader.conf
genmon_backup/genslack.conf
genmon_backup/genexercise.conf
genmon_backup/kwlog.txt
Done.

Does this look okay?

Then on Step 5, I got the following response:

pi@GenMon:~/genmon $ ./genmonmaint.sh -s
mkdir: cannot create directory ‘/etc/genmon/’: File exists

This didn’t look right so I stopped here to check with you.

Thanks,
Jim

From: jgyates 
Sent: Sunday, July 07, 2019 9:56 PM
To: jgyates/genmon 
Cc: jimland58 ; Mention 
Subject: Re: [jgyates/genmon] GenMon stopped working several weeks ago - Evol 2.0 (#284)

OK. Let's try this:

  1.. Log into the pi via ssh

  2.. change directories to the genmon directory and stop genmon

  cd genmon
  ./startgenmon.sh stop

  3.. update to the latest version

  ./genmonmaint.sh -u

  you will need to type Y to make the update proceed

  4.. make a backup of your settings:

  ./genmonmaint.sh -b

  This will create a file named genmon_backup.tar.gz that has your config files backed up.

  5.. copy a fresh set of config files:

  ./genmonmaint.sh -s

  6.. unzip you old config files:

  mkdir /home/pi/backup
  tar -zxvf ./genmon_backup.tar.gz -C /home/pi/backup

  7.. copy over your mail settings and your power log

  sudo cp /home/pi/backup/genmon_backup/mymail.conf /etc/genmon
  sudo cp /home/pi/backup/genmon_backup/kwlog.txt /etc/genmon

  if you have any other add on config files you can copy them over as well. Do not copy over genloader.conf and genmon.conf. You can also just setup your add on programs again if that is easier.

  8.. Load genmon with the fresh copy of genloader.conf and genmon.conf that we copied over in step 5

  ./startgenmon.sh restart

If your serial connection is not working then you may see a delay before the web interface comes up. The delay should be no more than a couple of minutes.

Let me know if the web interface comes up and if you are having an issue with serial communications (are there serial timeout errors on the Monitor page). IF it does come up send me your logs on the About page.

If the web interface does not come up then from the genmon directory on the pi type:

 python ClientInterface.py 
This will give you a prompt that has the status of the program / generator. From the prompt type:

generator: sendlogfiles 
Also, let me know what the prompt says. You can type :

generator: monitor

and it will display the contents of the Monitor page. copy and past that output to this thread.

—

---
This email has been checked for viruses by AVG.
https://www.avg.com

jgyates, 2019-07-08:
Yes, so far everything looks normal. The messages are because the /etc/genmon directory already exists and there is not existing backup file to overwrite.

jimland58, 2019-07-09:
Completed all the steps.  After step 8 I started getting text messages saying SUBJ: Generator Monitor Starting at .... about every 10 seconds.   Tried bringing up the web interface with Chrome and get spinning circle with message of “Loading Generator Monitor”.  Here is what I get running the ClientInterface.

pi@GenMon:~/genmon $ python ClientInterface.py
WARNING: System Initializing.  Not receiving data. : ,

>generator: sendlogfiles
WARNING: System Initializing.  Not receiving data. : ,
Log files submitted
>generator: monitor
WARNING: System Initializing.  Not receiving data. : ,

Monitor :

    Generator Monitor Stats :
        Monitor Health : System Initializing.  Not receiving data.
        Controller : Nexus, Air Cooled
        Run time : Generator Monitor running for 0:00:00.
        Generator Monitor Version : V1.13.23

    Serial Stats :
        Packet Count : M: 1, S: 0
        CRC Errors : 0
        CRC Percent Errors : 0.00%
        Packet Timeouts : 0
        Packet Timeouts Percent Errors : 0.00%
        Modbus Exceptions : 0
        Validation Errors : 0
        Invalid Data : 0
        Discarded Bytes : 0
        Comm Restarts : 0
        Packets Per Second : 1.71

    Platform Stats :
        CPU Temperature : 148.10 F
        Pi Model : Raspberry Pi 3 Model B Rev 1.2
        Pi CPU Frequency Throttling : OK
        Pi ARM Frequency Cap : OK
        Pi Undervoltage : OK
        CPU Utilization : 0.63%
        OS Name : Raspbian GNU/Linux
        OS Version : 8 (jessie)
        System Uptime : 6 days, 7:34:22
        Network Interface Used : wlan0
        WLAN Signal Level : -63 dBm
        WLAN Signal Quality : 47/70
        WLAN Signal Noise : -256 dBm

From: jgyates 
Sent: Sunday, July 07, 2019 9:56 PM
To: jgyates/genmon 
Cc: jimland58 ; Mention 
Subject: Re: [jgyates/genmon] GenMon stopped working several weeks ago - Evol 2.0 (#284)

OK. Let's try this:

  1.. Log into the pi via ssh

  2.. change directories to the genmon directory and stop genmon

  cd genmon
  ./startgenmon.sh stop

  3.. update to the latest version

  ./genmonmaint.sh -u

  you will need to type Y to make the update proceed

  4.. make a backup of your settings:

  ./genmonmaint.sh -b

  This will create a file named genmon_backup.tar.gz that has your config files backed up.

  5.. copy a fresh set of config files:

  ./genmonmaint.sh -s

  6.. unzip you old config files:

  mkdir /home/pi/backup
  tar -zxvf ./genmon_backup.tar.gz -C /home/pi/backup

  7.. copy over your mail settings and your power log

  sudo cp /home/pi/backup/genmon_backup/mymail.conf /etc/genmon
  sudo cp /home/pi/backup/genmon_backup/kwlog.txt /etc/genmon

  if you have any other add on config files you can copy them over as well. Do not copy over genloader.conf and genmon.conf. You can also just setup your add on programs again if that is easier.

  8.. Load genmon with the fresh copy of genloader.conf and genmon.conf that we copied over in step 5

  ./startgenmon.sh restart

If your serial connection is not working then you may see a delay before the web interface comes up. The delay should be no more than a couple of minutes.

Let me know if the web interface comes up and if you are having an issue with serial communications (are there serial timeout errors on the Monitor page). IF it does come up send me your logs on the About page.

If the web interface does not come up then from the genmon directory on the pi type:

 python ClientInterface.py 
This will give you a prompt that has the status of the program / generator. From the prompt type:

generator: sendlogfiles 
Also, let me know what the prompt says. You can type :

generator: monitor

and it will display the contents of the Monitor page. copy and past that output to this thread.

—

---
This email has been checked for viruses by AVG.
https://www.avg.com

jgyates, 2019-07-09:
@jimland58 

You have going to have to give it a little more time. Notice the line:

     Run time : Generator Monitor running for 0:00:00.

It has been running for less than a second. Only one packet has been sent. Every packet has a three second time out. The initialization is complete when an attempt is made for every unique register which for Evolution is about 100 registers plus 150 log registers. The web interface needs some base info from genmon to display properly and that information is not sent until the controller is identified or the initialization is complete. This happens quickly if you serial comms are working, slower if they are not. You can use ClientInterface to get immediate status. You can also use it from your windows machine if you have python installed.

      python ClientInterface.py -a 192.168.1.170

where 192.168.1.170 is the IP address of your pi

If you run the ClientInterface.py command "generator: monitor" after 10 seconds or more and you see this line:

     Packet Count : M: 1, S: 0

where the S: value is still zero that means that you controller is not sending anything back to the pi. Check your cabling per these steps:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

jimland58, 2019-07-09:
I think the problem is the Pi is restarting over and over hence the reason I keep getting the text messages and only showing it running for a short time.  I was having a problem awhile back where I would loose connection on the network or with the generator so I seem to remember adding something to the Pi that would restart the Genmon program when this was detected but can’t remember how I did that!  I think whatever I did is causing the program to restart over and over.  I will check the serial connection.  Do you know if there was ever a Genrac software update issued that might cause the serial communications to stop? 

From: jgyates 
Sent: Monday, July 08, 2019 9:52 PM
To: jgyates/genmon 
Cc: jimland58 ; Mention 
Subject: Re: [jgyates/genmon] GenMon stopped working several weeks ago - Evol 2.0 (#284)

@jimland58

You have going to have to give it a little more time. Notice the line:

 Run time : Generator Monitor running for 0:00:00.
It has been running for less than a second. Only one packet has been sent. Every packet has a three second time out. The initialization is complete when an attempt is made for every unique register which for Evolution is about 100 registers plus 150 log registers. The web interface needs some base info from genmon to display properly and that information is not sent until the controller is identified or the initialization is complete. This happens quickly if you serial comms are working, slower if they are not. You can use ClientInterface to get immediate status. You can also use it from your windows machine if you have python installed.

  python ClientInterface.py -a 192.168.1.170
where 192.168.1.170 is the IP address of your pi

If you run the ClientInterface.py command "generator: monitor" after 10 seconds or more and you see this line:

 Packet Count : M: 1, S: 0
where the S: value is still zero that means that you controller is not sending anything back to the pi. Check your cabling per these steps:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

—

---
This email has been checked for viruses by AVG.
https://www.avg.com

jimland58, 2019-07-09:
Found the program I was using to reboot if it dropped network connection.  /usr/local/bin/checkwifi.sh

            ping -c4 192.168.1.1 > /dev/null 

     

      if [ $? != 0 ]  

      then 

      sudo /sbin/shutdown -r now 

fi

Changed it to this to restart wireless connection vs reboot:

      ping -c4 192.168.1.1 > /dev/null 

     

      if [ $? != 0 ]  

      then 

      echo "No network connection, restarting wlan0" 

      /sbin/ifdown 'wlan0' 

      sleep 5 

      /sbin/ifup --force 'wlan0' 

fi 

Made these changes and rebooted the Pi and when I started Genmon and run the Generator: Monitor is only runs for a few seconds and then it looks like it restarts since everytime I run the Generator: Monitor command it will say 2 or 3 seconds for Generator Time.   I will do a loopback test tomorrow.  Thanks for all the help.

From: jimland58@eatel.net 
Sent: Monday, July 08, 2019 10:56 PM
To: jgyates/genmon ; jgyates/genmon 
Cc: Mention 
Subject: Re: [jgyates/genmon] GenMon stopped working several weeks ago - Evol 2.0 (#284)

I think the problem is the Pi is restarting over and over hence the reason I keep getting the text messages and only showing it running for a short time.  I was having a problem awhile back where I would loose connection on the network or with the generator so I seem to remember adding something to the Pi that would restart the Genmon program when this was detected but can’t remember how I did that!  I think whatever I did is causing the program to restart over and over.  I will check the serial connection.  Do you know if there was ever a Genrac software update issued that might cause the serial communications to stop? 

From: jgyates 
Sent: Monday, July 08, 2019 9:52 PM
To: jgyates/genmon 
Cc: jimland58 ; Mention 
Subject: Re: [jgyates/genmon] GenMon stopped working several weeks ago - Evol 2.0 (#284)

@jimland58

You have going to have to give it a little more time. Notice the line:

 Run time : Generator Monitor running for 0:00:00.
It has been running for less than a second. Only one packet has been sent. Every packet has a three second time out. The initialization is complete when an attempt is made for every unique register which for Evolution is about 100 registers plus 150 log registers. The web interface needs some base info from genmon to display properly and that information is not sent until the controller is identified or the initialization is complete. This happens quickly if you serial comms are working, slower if they are not. You can use ClientInterface to get immediate status. You can also use it from your windows machine if you have python installed.

  python ClientInterface.py -a 192.168.1.170
where 192.168.1.170 is the IP address of your pi

If you run the ClientInterface.py command "generator: monitor" after 10 seconds or more and you see this line:

 Packet Count : M: 1, S: 0
where the S: value is still zero that means that you controller is not sending anything back to the pi. Check your cabling per these steps:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

—

---
This email has been checked for viruses by AVG.
https://www.avg.com

jgyates, 2019-07-12:
I would remove the check wifi script until you can validate your serial communications is working. The wifi script or a power problem are the most likely things that would cause your pi to reboot.

jgyates, 2019-07-17:
I am closing this issue for now since it has been 9 days without a status update. If you still have issues feel free to post them to this thread.

closed 2019-07-17 by jgyates

jimland58, 2019-07-23:
Sorry for the late reply, got busy with other projects.  So, I decided to start over and reflashed the SD card and reload Genmon.  I did the serial loopback test at the far end of the cable thru the RS-232 converter and it passed.  Hooked it up to the generator and when I bring up the webserver it shows that it is not receiving data and getting packet timeouts.  Have Automated Feedback On.

jgyates, 2019-07-23:
Are you using the same cable that worked before? 
Have you validated the serial port is setup correctly with serialconfig.py as stated in the serial troubleshooting guide?

This page has the steps to troubleshoot this issue:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

jimland58, 2019-07-24:
Ran serialconfig.py and everything showed okay but still not receiving data and getting packet timeouts.

Checking : BT service disabled: OK
Checking : Serial console command line removed: OK
Checking : Is enable UART in boot config: OK
Checking : Is BT overlay disabled: OK
Checking : Serial console service disabled: OK

Serial port settings are OK. A reboot is needed if changes were made.

Checking mymodbus.log, I am getting messages like this for all the different registers:

2019-07-23 20:20:22,298 : Error: timeout receiving slave packet for register 0036 Buffer:0
2019-07-23 20:20:25,322 : Error: timeout receiving slave packet for register 0237 Buffer:0
2019-07-23 20:20:28,350 : Error: timeout receiving slave packet for register 0033 Buffer:0

The file myserial.log is empty.

Monitor page looks like this:

Monitor :

    Generator Monitor Stats :
        Monitor Health : 
System Initializing. Not receiving data. 

        Controller : 
Nexus, Air Cooled

        Run time : 
Generator Monitor running for 0:10:36.

        Generator Monitor Version : 
V1.13.27

    Serial Stats :
        Packet Count : 
M: 209, S: 0

        CRC Errors : 
0 

        CRC Percent Errors : 
0.00%

        Packet Timeouts : 
208

        Packet Timeouts Percent Errors : 
99.52%

        Modbus Exceptions : 
0

        Validation Errors : 
0

        Invalid Data : 
0

        Discarded Bytes : 
0

        Comm Restarts : 
0

        Packets Per Second : 
0.33

    Platform Stats :
        CPU Temperature : 
126.68 F

        Pi Model : 
Raspberry Pi 3 Model B Rev 1.2

        Pi CPU Frequency Throttling : 
OK

        Pi ARM Frequency Cap : 
OK

        Pi Undervoltage : 
OK

        CPU Utilization : 
2.91%

        OS Name : 
Raspbian GNU/Linux

        OS Version : 
10 (buster)

        System Uptime : 
0:11:06

        Network Interface Used : 
wlan0

        WLAN Signal Level : 
-71 dBm

        WLAN Signal Quality : 
39/70

        WLAN Signal Noise : 
-256 dBm

From: jgyates 
Sent: Tuesday, July 23, 2019 4:37 PM
To: jgyates/genmon 
Cc: jimland58 ; Mention 
Subject: Re: [jgyates/genmon] GenMon stopped working several weeks ago - Evol 2.0 (#284)

Are you using the same cable that worked before?
Have you validated the serial port is setup correctly with serialconfig.py as stated in the serial troubleshooting guide?

This page has the steps to troubleshoot this issue:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

—

---
This email has been checked for viruses by AVG.
https://www.avg.com

jimland58, 2019-07-29:
Reset the Evol 2 controller and now working!

jimland58, 2019-07-29:
Technically, I guess I "Power Cycled" the controller and that fixed the problem with loss of communication

jgyates, 2019-07-30:
Outstanding! Thanks for the feedback on how this was resolved.