jgyates/genmon #285: ATS didn't switch over-feature request

jcherney, 2019-07-06:
Tonight I lost power when I was away from home. Got the emails right on schedule from Genmon, and figured everything was just fine. Except, the ATS never switched to Gen power. I didn't know this until I got home a couple hours later. Generator was running. The switch was still in utility mode. This isn't a Genmon issue.

But, I was wondering if there is a way to monitor the switch status in the Genmon program. I know the program will allow me to start and transfer, and I was hoping it could give a bit more information.

jgyates, 2019-07-06:
@jcherney 

The controller activates a relay that signals the transfer switch to start. The state of this relay is readable over mobus on evolution liquid cooled models. It is not readable (or so it appears, I have tried looking on a few occasions) on evolution air cooled and nexus models. Even on the Evo liquid cooled model I can only read the state of the relay, if the ATS fails and the relay is still active the controller thinks it is working as expected. This has more to do with the smart controller / dumb switch architecture on residential models than anything else. When an outage occurs the controller detects this and signals (via relay) to the ATS. The ATS will set a line active back to the generator, but this status, to my knowledge, is not readable since the controller would not act on this return signal in any way.

On the evo liquid cooled models, when the relay to the ATS changes state an email is set as a separate notification. Also, all of the add-on programs that use the mynotify library are also notified when the ATS relay changes state (gensms, genlog, genslack, etc).

jcherney, 2019-07-06:
Thanks for the explanation. I had a feeling that this might be the case, but was hoping!

closed 2019-07-06 by jgyates