jgyates/genmon #287: [Cosmetic] Web Interface green/yellow/red gauges indicator not correctly set for my country

JMVS, 2019-07-06:
### Expected Behavior

Have gauges at the web interface show the correct limits for voltage for my country (Argentina). Our standard supply voltage is 220VAC with a range of 200-230 for yellow.

### Actual Behavior

Gauges show miplaced limits (green/yellow/red) for my country.

### Steps to Reproduce (including precondition)

Set the voltage output of the generator to 220VAC

### Screenshot or Pictures relating to the problem (if possible)

[image]

### Your Environment

- Generator Model: 0062710 (Generac Evo 13KVA Air Cooled)
- Generator Registers: Sent via the About page
- Genmon Version: 1.13.21

### Suggestion

Maybe have an option in the advanced settings to set the limits for each gauge and band limit.

jgyates, 2019-07-07:
Hi @JMVS 

I checked in v1.13.22 that has two new options on the advanced page.

Use Nominal Volts Override - Set to true to enable this feature
Override nominal line voltage in UI - set to 220 in your case 

If you make these two settings and save them then the gauges will use 220 as the nominal. The green will span +/- 5% of the nominal and the yellow will span +/- 10% of the nominal.

Let me know if you have any issues or questions.

JMVS, 2019-07-07:
Working perfectly @jgyates. Thank you.

closed 2019-07-07 by JMVS