jgyates/genmon #288: [Bug?] Can't remote stop generator from web interface when in service alarm

JMVS, 2019-07-07:
### Expected Behavior

Being able to remotely stop generator via web interface after initiating start or start and transfer from the same interface, even when there's a service alarm. Buttons status should change to reflect current status.

### Actual Behavior

After initiating the remote start (or remote start & transfer) the web interface does not update it's status and the Stop button remains green and the other two grey. It can be stopped it by sending <ip address>/cmd/setremote?setremote=stop
The web interface "links" color shows up in red.

### Steps to Reproduce (including precondition)

Having the generator in a "non-critical" alarm (like "service due" in my case, could happen with "inspect battery" too) go to GenMon web interface > Maintenance > Remote Commands and select "Start Generator" or "Start Generator and Transfer". After confirming, you wouldn't be able to send the "Stop Generator" command (nothing happens when you click on it)

_I'm guessing here because I actually realized this behavior after servicing the generator and clearing the alam (whcih would be the only difference between the expected behavior and actual behavior)._

### Screenshot or Pictures relating to the problem (if possible)

Not available

### Your Environment

- Generator Model: 0062710 (13 KVA Evo controller)
- Generator Registers: Sent via the About page
- Genmon Version: 1.13.21

jgyates, 2019-07-07:
@JMVS 

I updated version 1.13.22 to include this fix. If you already updated just update again.

The issue is that the web ui javascript code disables the start, stop, start and transfer buttons based on the state of the generator. I changed the behavior so all of the buttons are active if an alarm is present or if service is due. Note that this gives the maximum flexibility but for some critical alarms the remote commands may not work due to limitations in the controller firmware. As you probably know, some alarms require a the operator to physically reset the alarm at the panel before the generator can be started. Let me know if you have any questions or issues.

JMVS, 2019-07-07:
Thank you for you prompt reply & fix @jgyates. Cannot really test this as I don't have any alarms now, but I suppose this can be reopened if someone runs into the same problem after 1.13.22 update.

closed 2019-07-07 by JMVS