jgyates/genmon #290: Journal not saved on web interface

JMVS, 2019-07-07:
### Expected Behavior

Add a new journal entry on the web interface and have it stored to retrieve and view at a later stage.

### Actual Behavior

After a journal entry is saved, moving away from the Journal Entry page and back to it makes it dissapear. It doesn't appear again after reboot either. I have not tried to Clear logs. I was only able to save two entries as shown in screenshots. The entries were not saved in chonological order if that helps. In the screenshots, after saving, moving away and back, the Service Journal looks like the first one again.

### Steps to Reproduce (including precondition)

Try to save more than two entries in the Service Journal.

### Screenshot or Pictures relating to the problem (if possible)

[image]
[image]
[image]
[image]
[image]

### Your Environment

- Generator Model: G0062710
- Generator Registers: Sent via the About page on July 6th '19
- Genmon Version: 1.13.22

jgyates, 2019-07-07:
Hi @JMVS 

I can not duplicate this. My first guess is that this could be an issue with the OS or browser settings and the interaction with the javascript, more specifically your system is likely using spanish as the default language and the javascript may not be handling that properly. In your logs from the 6th I noticed these entries in /etc/genmon/genmon.log:

    2019-07-06 14:26:13,046 : Error in ValidateMaintLogEntry: expecting MM/DD/YYYY : unconverted data remains:  09:00 : controller.py:1307
    2019-07-06 14:29:09,662 : Error in AddEntryToMaintLog: No JSON object could be decoded : controller.py:1280
    2019-07-06 14:38:48,020 : Error in ValidateMaintLogEntry: expecting MM/DD/YYYY : unconverted data remains:  15:00 : controller.py:1307
    2019-07-06 14:40:42,727 : Error in AddEntryToMaintLog: No JSON object could be decoded : controller.py:1280
    2019-07-06 17:46:16,573 : Error in AddEntryToMaintLog: No JSON object could be decoded : controller.py:1280

Let me look at the code and see if I can find anything that may be suspect, but if you could do the following this may help me figure out specifically where the problem lies.

1) Open the browser developer console 
2) go to the Network section in the developer console 
3) try to make an entry in the service journal
4) once you click save you should see a call from the browser to genmon named add_maint_log. If you select that entry it will show details on call. What is need is the full URL of that call. It should look something like this:

    http://192.168.1.16:8000/cmd/add_maint_log?add_maint_log=%7B%22date%22%3A%2207%2F07%2F2019%2016%3A26%22%2C%22type%22%3A%22repair%22%2C%22hours%22%3A611%2C%22comment%22%3A%22test2%22%7D

Let me know if you have any questions.

jgyates, 2019-07-07:
Also, while you have the developer console open,  check to see that you don't have any errors on the console.

jgyates, 2019-07-07:
also, what browser / os are you using

JMVS, 2019-07-07:
Please confirm this is what you're asking for:
Requested URL:http://<redacted>/cmd/add_maint_log?add_maint_log={"date":"07/06/2019 18:00","type":"maintenance","hours":611,"comment":"Reemplazo de filtro de aceite (Japanparts F0-898S) y aceite (Gulf 15W40 Mineral) ($800 Lubricentro MS). Falta realizar verificación de luz de válvulas y cambio de filtro de aire."}
Response: "Invalid input for Maintenance Log entry (2)."

No errors on Console.

OS: Windows 10 64-bit (10.0.17763.557)
Browser: Firefox 67.0.4 (64-bit)

jgyates, 2019-07-07:
I found the problem. For some reason it is choking on the accented "o" in verificación. If you change or take out that letter it will save the entry. Let me look into why this is happening and see if I can fix this.

JMVS, 2019-07-07:
I can confirm that accentuated words (maybe all extended characters?) are not handled gracefully. By adding an entry without accents the same was saved with no problems. I suppose this should be still an open issue? I can do further tests if needed.

jgyates, 2019-07-07:
yes, let's leave this open. I am hoping to find a fix for this soon.

jgyates, 2019-07-08:
I checked in V1.13.23 which should fix the problem. Let me know if you have any issues. I am closing this issue for now but if you find anything else on this topic feel free to post in this thread.

closed 2019-07-08 by jgyates