jgyates/genmon #291: [Feature request] Accesing web interface from WAN without opening ports on router or VPN

JMVS, 2019-07-07:
@jgyates 
Would it be possible to add a service like serveo.net that would allow to create a reverse SSH tunnel between the RPi and the web server to be able to access over the Internet without opening ports or knowing the RPi's address (nor having to create a DDNS)?

Right now I'm doing this with a systemctl service like below, adapted from this [guide](https://techrad.io/how-to/1231/access-your-raspberry-pi-from-internet/)

[Unit]
Description=Revesre SSH Tunnel

[Service]
Type=idle
ExecStart=/usr/bin/ssh -R _anyname_:80:localhost:8000 serveo.net &
ExecReload=/bin/kill -HUP $MAINPID
KillMode=process
Restart=on-failure
RestartSec=42s

[Install]
WantedBy=multi-user.target

My system is configured like [this](https://github.com/jgyates/genmon/issues/277)
I wasn't able to make it work with HTTPS enabled on GenMon settings.

I think it would be great to have it implemented as part of GenMon, allowing the user to access the web interface from the Internet without needing to deal with opening ports or knowing the IP address.

JMVS, 2019-07-07:
I just found this that might be useful as well: https://github.com/equals215/autossh-tunnel-systemd
BTW I see serveo.net is down now 👎 (cached version [here](https://webcache.googleusercontent.com/search?q=cache:pFz1x7yuR7oJ:https://serveo.net/))

jgyates, 2019-07-08:
Let me look into this. It sounds nice, but I want to make sure there are not some security downsides.

JMVS, 2019-07-08:
Great. Thank you @jgyates.
Here are a few other free alternatives:
http://localhost.run/
https://localtunnel.github.io/www/

RoyRomeo, 2019-07-15:
This is a service that most GenMon owners would love to see added.

jgyates, 2019-07-17:
@JMVS 

I have tried both localhost.run and serveo.net and have been unable to get them working if the genmon web app is running in HTTPS mode. Using reverse ssh without the username and password is problematic as this would allow anyone to access your generator. They both work with HTTP however genmon is currently designed to use username / pw with HTTPS and no username/pw with HTTP mode. At this point I am not really sure about this approach. I will continue to look a this however I did want to give you an update.

RoyRomeo, 2019-07-18:
Thanks for the update.  Hopefully there is a work around.

JMVS, 2019-07-18:
@jgyates Thanks for the follow up. Yes, I struggled with that too and I thought you may have better luck. Anyway, as I mentioned in my OP, I used the workaround suggested by you in issue #277 to get around that. I still have HTTPS/SSH access (provided by serveo.net) and authentication by GenMon. That's my current setup and so far it's working great.

RoyRomeo, 2019-07-18:
Is there any way that you could post a detailed step-by-step instructional guide for this work around, for us novices?

USAFPride, 2019-07-28:
Best bet is to get a router that supports OpenVPN and set that up for secure access. No extra service provider required.

scyto, 2019-11-11:
I use an nginx reverse proxy for this scenario. If you are using docker and want a GUI driven version you can use proxy manager.

klinquist, 2019-11-21:
I do the same as @scyto.  You can easy set HTTP Basic Auth on a reverse proxy.  Use ssl cert from letsencrypt, dyndns provider of your choice.   No tunnels necessary.

JMVS, 2019-11-21:
@scyto @klinquist It would be great if you could post a guide for this. Thanks.

klinquist, 2019-11-21:
I'll provide some details here, google should hopefully be able to fill in the blanks.

1) Get a dynamic DNS provider setup & working. I actually use a bash script & amazon route53 for this, but dyndns.org is a popular choice as well.  This will give you your own host such as https://jmvs.dyndns.org.

Now, you can install nginx directly on the raspberry pi.  You'll have to port forward port 443 through your home firewall & set up a letsencrypt SSL cert. Instructions for the latter can be found here:
https://www.digitalocean.com/community/tutorials/how-to-set-up-let-s-encrypt-with-nginx-server-blocks-on-ubuntu-16-04

Now for the reverse proxy bit.  Here is a "location" server block you can add to your nginx config (/etc/nginx/sites-enabled/default):

`        location /genmon {
                proxy_set_header X-Real-Ip $remote_addr;
                proxy_set_header X-Forwarded-For $proxy_add_x_forwarded_for;
                proxy_set_header Host $http_host;
                proxy_http_version 1.1;
                proxy_set_header Upgrade $http_upgrade;
                proxy_set_header Connection "Upgrade";
                proxy_pass http://localhost:8000;      #IP ADDRESS OF SERVER RUNNING GENMON
                subs_filter \"http\",\s\"ws\" \"https\",\"wss\" gr;
                subs_filter http://YOUR_PUBLIC_HOSTNAME/(.*) https://YOUR_PUBLIC_HOSTNAME/genmon/$1 gir;
                rewrite ^/genmon/(.*) /$1 break;
                proxy_ignore_client_abort on;
        }`

Restart nginx and now you should be able to access https://your_public_hostname/genmon

Now that it's wide open, you'll want to protect it with a login/password.  You do this by adding basic auth to the above block:

https://docs.nginx.com/nginx/admin-guide/security-controls/configuring-http-basic-authentication/

zekyl314, 2019-11-22:
I think using cloudflare Argo tunnels would be the easiest option and pretty sweet setup. I’ll see if I can document a setup using this method and post it here. I may take me a while with other hints going on, but I’ll see what I can do.

JMVS, 2020-02-24:
Serveo is down for several months now. There's a similar service called [sysb.AI](url:https://www.sysb.ai/):
`ssh -R 0:localhost:8000 demo@sysb.ai`

You can also get a static [subdomain](url:https://www.sysb.ai/blog/how-to-get-static-subdomain.html)

RoyRomeo, 2020-02-25:
Is there any way that you can list the step-by-step process for setting this up.  Some of us have no experience.

jgyates, 2020-02-25:
I would be extremely cautious with these reverse ssh servers. this new url uses the (.ai) top level domain for Anguilla, a small island nation in the Caribbean Sea. All of your traffic from your remote location to the pi would go through this server. Sounds like a potential way to get hacked. No offence to anyone from there :)

FYI, the main reason that I have not integrated this as an add on to genmon is that I have concerns that this would open folks up to have their generator hijacked or have their home LAN compromised. Also, the reverse SSH service providers are not extremely reliable.

Feel free to post to this thread if you want to share info on this topic, however I am going to close this thread since, at least for now, I am not going to implement this as an add on due to security issues.

closed 2020-02-25 by jgyates

liltux, 2020-02-25:
My opinion is use a VPN server.  That you own/pay for.

klinquist, 2020-02-25:
What liltux said. I use Ubiquiti firewall/router(s) in two houses.  I can VPN into either house remotely... and I even have a permanent VPN between them!

scyto, 2020-02-25:
I agree for SSH (and to be clear i expose ~15 internal web sites to the outside world using an nginx reverse proxy).  The OP could consider one of the many jumpbox style containers or something like https://royalapps.com/server/main/features