jgyates/genmon #292: [Feature request] Scheduled times for start & transfer when utility fails

JMVS, 2019-07-09:
Hello @jgyates. Following my failed #267 feature request, I would like to know if it is possible to add a schedule function to GenMon in which you select the days and times in which you want the generator to start and transfer when utility is lost. Would be kind of expanding on what the "Enhanced Exercise" add-on already does. I understand that once you start and transfer the generator, you'll lose control to remotely stop it, but it's better than nothing.
I'm assuming that the remote controls work even when the generator controller is in the STOP mode (which would be needed for GenMon to be able to control the start & transfer at will).

jgyates, 2019-07-09:
@JMVS 

Unfortunately the remote commands only work if the switch is in the AUTO position. If the switch is in the OFF position remote commands do not work. Liquid cooled Evolution models have the ability to set the switch position remotely but the air cooled units do not.

JMVS, 2019-07-09:
Mmmm...It seems that I'm out of luck....Would 2-wire start help in any way?

jgyates, 2019-07-09:
I am not sure, @liltux may have an answer to that one. From what I read if the controller is in two wire mode it will no longer start / stop when the utility service is not present / present. In two wire mode you would be responsible for starting and stopping and removing the load before stopping.

liltux, 2019-07-09:
@JMVS if the controller is configured for 2-wire stop it assumes the transfer switch is in full control.  So under 2-wire start AND in Auto mode the generator will only start when the 2-wire start contact is closed.  The unit will start and run but will not control the transfer switch.  The assumption is if you are using a 2-wire start setup, then you have a "smart" switch that handles all if the transfer and retransfer.  Then generator then becomes a dumb device and just runs when is told to and stop when it's not.  What you are trying to achieve is not the proper way to operate an emergency power generation system. The proper way to start and stop a generator during a utility loss is to have an operator/owner present that can turn off/on the correct circuit breakers, allow proper warm up and cooldown times, and transfer safely to protect from equipment and personal damage.

JMVS, 2019-07-09:
@liltux Thanks for your input. Now I understand it not really achievable. I ran out of ideas.
Anyway...may a nice feauture to implement for controllers that can support it? I leave it up to you @jgyates.

closed 2019-07-12 by jgyates