jgyates/genmon #293: [Feature request] NG Fuel Consumption calculation

JMVS, 2019-07-09:
Would it be possible to add fuel consumption calculation based on load?
The spec sheet for my gen shows the consumption for 1/2 Load & Full Load. With linear interpolation we can get the _estimated_ consumption for any given load. 
[image]

From here you can expand to have a graphic for estimated fuel consumption (like the kW output) and even have the possibility to add the fuel costs to also have the estimated costs.
I've created a worksheet with the calculations for my gen as an example (screenshot attached).

[image]

I don't know how much information is provided in other gens for consumption, but a table could be created with the real (manual stated values) and interpolate from there the rest of the values. This can be included on GenMon for every supported generator and/or have the user imput the real values in the setup.

jgyates, 2019-07-10:
It is possible to add this feature. The method you describe is similar to how it is being handled with propane and diesel fuel consumption. The reason I initially only supported propane and diesel is two fold. 1) Propane and diesel require a tank to be filled periodically and the main goal was to monitor the tank level so the user could fill the tank when needed and 2) Since gaseous models typically support either NG or LP it simplified the design to only have one polynomial per generator model instead of a unique polynomial for both NG and LP for each model. I can put this on the list of things to add, but it is fairly time consuming to gather the data for each model generator so it not something that I could add in the short term.

jgyates, 2019-07-12:
@JMVS 

I have added support for NG fuel consumption for Evolution Air Cooled units as of V1.13.25. I still need to gather info for liquid cooled units. for Propane and Diesel Evolution controllers, fuel consumption gets you a fuel gauge with an estimate of how much is left in the tank, a 30 day estimate of fuel used and and entry in the outage log for estimated fuel used for the outage. For NG you don't get the fuel tank level estimate as this would not  be applicable. 

For fuel consumption estimates for outages, this applies to any outages after you install the upgrade to 1.13.25. NG reports in cubic feet for units, unless the "Use Metric" is enabled in settings and then it reports in cubic meters (Propane and Diesel us gallons and liters as units).

Let me know if you have any questions or problems.

USAFPride, 2019-07-13:
I've wished for this, but never voiced it.  Thanks for doing this.  Is there any chance to get us old Nexus people the same?

jgyates, 2019-07-13:
Unfortunately the to implement this you need to be able to read the current, then convert it to power and get a percentage of load of the generator. The Nexus lines does not appear to export the current output over modbus. This is the reason that Nexus is not supported.

JMVS, 2019-07-13:
Thank you @jgyates. I've just updated.  As I've never seen this feature and from what you're saying I undestand that it will only show when there's an outage and as a record of estimated usage, right? There should be any new gauge showing during an outage? I know nothing changes when is excersing as that was today and nothing new appeared/chaged.

jgyates, 2019-07-13:
For Natural Gas, since there is no tank involved, there is no gauge to show tank level.

On the outage page, the outage log will show the amount of fuel used for an outage once the outage is over as an entry in the outage log. For example and outage for a diesel would have an entry in the outage log like this:

       2019-07-02 11:46:16, Duration: 2:55:39, Estimated Fuel: 2.18 gal

The above value is calculated based on the entries in the power log for the duration of the outage.

NG would show cubic feet or cubic meters, depending on the "Use Metric Units" setting.

The maintenance page will have entries like this (diesel example shown):

    kW Hours in last 30 days : 26.03 kWh
    Fuel Consumption in last 30 days : 2.39 gal
    Run Hours in last 30 days : 3.40 h

The above 30 day fuel consumption is calculated based on the power log entries for the last 30 days. If you do not have any power log entries this value will be zero.

jgyates, 2020-09-02:
Closing this issue as this has been implemented.

closed 2020-09-02 by jgyates