jgyates/genmon #294: GenMon Did not Restart after Yearly Maintenance

shpegel, 2019-07-10:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below.

***NOTE:*** If you are having issues with your serial connection, please read this page before posting:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

If you are having other issues, please see the following page:

https://github.com/jgyates/genmon/wiki/3.5---General-Troubleshooting

If you need to send you logs and registers to the developer, if you email is setup and working properly you can click send your logs on the About page in the web interface.

### Expected Behavior
Not an Issue. Just FYI  GenMon is working fine now

GenMon did not run after Service call by generator company. After yearly service GenMon did not restart although Pi was running and I could log into it. I tried a soft  reboot but it did not help. I went to Pi and interrupted power to it and all was well again. 

I did not watch service tech but noted he had a load type battery tester and suspect he may have done a battery load test without disconnecting positive lead from battery. Just a guess. If true then then perhaps this put serial converter in odd state and it took a power cycle to clear it.  

### Actual Behavior

 

### Steps to Reproduce (including precondition)

 

### Screenshot or Pictures relating to the problem (if possible)

 

### Your Environment

- Generator Model: Generac 7907 (17KW) Evo controller
- Generator Registers: Submit via the About page
- Genmon Version:1.13.23

liltux, 2019-07-10:
I see that as an excellent possibility as the battery test would drop battery voltage below 12VDC.  Depending on the tolerance of your DC to DC power supply, could have caused enough dip to cause issues.

jgyates, 2019-07-10:
@shpegel, Thanks for the info

closed 2019-07-12 by jgyates