jgyates/genmon #295: USB -> TTL possible?

bdwilson, 2019-07-14:
Sorry for opening up an issue for a question; I have parts at home and wanted to see if they would work:

For serial, would a USB to TTL adapter work connected to the Pi via usb vs the GPIO -> serial -> TTL? Then you could splice the wires from the 8 pin connector direct to this, and also grab also 5v and ground from 8 pin connector splices to tie to a micro USB to power the device. 

https://www.amazon.com/dp/B00QT7LQ88/

Curious if this has been done or why it would/wouldn't work. Looking forward to trying out once I get the rest of the parts in. Thanks for your work on this!

jgyates, 2019-07-14:
@bdwilson 

No problems with questions in the issue forum.

This will not work if interfacing the pi to the generator. The adapter you reference has TTL on one side and USB on the other. If you connect the USB side of this adapter to the pi you get a TTL output, which is the same voltage level (TTL is 0V to 5V) as the serial GPIO port. If you connect the wires from the 9 pin port to TTL level signals it will not communicate and possibly harm your TTL inputs since RS-232 has voltages that range from minus 12V to positive 12V. To make this USB adapter work would would still need a TTL to RS-232 converter board to interface to the generator 8 pin port.

bdwilson, 2019-07-14:
Got it. Makes complete sense. Also, I checked the for the ports on my Generac/Eaton with Evolution LCD; there are 2 ports with stickers under the LCD - accessory 1 and accessory 2. I assume either will work? Both are 8 pin. Thank you for the quick response!

jgyates, 2019-07-14:
I am not sure about the 2nd port. Post a picture and your model number and maybe someone with a similar generator can answer that question. If you have a multimeter you could measure the voltages on each connector for power and ground listed on this page:

https://github.com/jgyates/genmon/wiki/3.1--Making-a-Cable

jgyates, 2019-07-14:
Closing this issue, feel free to post to this thread if you have questions on this topic.

closed 2019-07-14 by jgyates

JMVS, 2019-07-16:
Just chiming in to post a picture of the **Evolution** ports. In this case, you have to hook up the cable to **PORT 1**
[image]

lwbeam, 2020-01-04:
With regard to the original question, as has already been stated, the suggested cable will not work. However, a suitable console cable (USB to RS-232), such as this one, will.

https://www.amazon.ca/gp/product/B0769J9QXP/

I’m currently in the process of setting up my monitor and have successfully tested this cable.

bdwilson, 2020-01-04:
@lwbeam how did you wire this? USB port on the Pi side with what connections on the other end?

lwbeam, 2020-01-05:
@bdwilson Correct, the USB end goes into the RPi. No additional drivers are required, so it's automatically recognized as **/dev/ttyUSB0**. Simply substitute this port designation in the config files or add to commands for executing related apps.

As for the other end, I'm currently using a short piece of cat5e cable with an 8-pin Molex connector on one end and a [Leviton QuickPort Cat 5 RJ45 connector](https://www.amazon.ca/Leviton-5G108-AW5-QuickPort-Adaptor-10-Pack/dp/B000U3BTSE) on the other. I think it's wired as follows, but I may have had to switch (crossover) the RxD and TxD wires at one end:

RxD - Molex pin 7 - RJ45 pin 3 (white/green)
TxD - Molex pin 8 - RJ45 pin 6 (green)
Gnd - Molex pin 2 - RJ45 pin 4 (blue)

My generator is located next to my heated garage. I have an ethernet connection for my network in the garage. I was initially going to run the serial cable into the garage, and locate the RPi there. However, I've since decided to power the RPi from the generator's battery in order to eliminate the need for a UPS. This will require the RPi to be located in the generator cabinet, which already has a  trickle charger installed just above the battery. As a result, I'm using a very small weather-proof box to house the RPi, which won't allow me to use to above console cable. To save space, I'm now going to use a tiny [MAX3232 board](https://www.amazon.ca/gp/product/B07QKC4S8R) for the TTL to RS232 convertion. I've tested this board with the RPi, but not with the generator. As I said earlier, I'm still in the process of getting things setup.

JMVS, 2020-01-05:
> @bdwilson Correct, the USB end goes into the RPi. No additional drivers are required, so it's automatically recognized as **/dev/ttyUSB0**. Simply substitute this port designation in the config [...]

I've installed this very same tiny board. Make sure you have a good soldering. I've killed my first MAX3232 board by a combination of bad solder and bad cables (nothing happened to the RPi nor the generator, just lost communication). The first one ran good for a couple of weeks and then died (I assume the vibration from the generator made the solder came loose). I tried re-soldering but then applied to much heat and ended up losing the pads on the board. On the second one I was more careful and has been running fine since then.

buzzshot, 2020-01-12:
I have tried the same MAX3232 as well and have the same problem. For my self I was thinking it may have been a bit of flux causing it to slowly crash. Mine is soldered to a little proto board along will my RJ-45 dongle. It’s just a touch wider then the pi zero. I was trying to keep a small form factor.