jgyates/genmon #296: mqtt setup

farberm, 2019-07-16:
MQTT integration

I am trying to integrate MQTT with my Indigo home automation system.

I have turned on the MQTT integration on the genmon and set the server address to 192.168.2.200 and port 1883. I set certificate to optional and left the username and password blank

I set up a MQTT broker on my Synology NAS (ip address 192.168.2.200). I started it.

I can connect from my mac computer with an app  MQTT.fx by creating a profile of 192.168.2.200 and port 1883
I cannot see the topics but I does appear that the MQTT server is receiving msg from the generator mqtt integration based upon seeing mosquitto version 1.4.14 Subscriptions listed as 30 and info in the log

2019-07-15 20:25:22,392  INFO --- MqttFX ClientModel             : messageArrived() with topic: $SYS/broker/uptime
2019-07-15 20:25:22,392  INFO --- MqttFX ClientModel             : messageArrived() with topic: $SYS/broker/load/messages/received/1min
2019-07-15 20:25:22,393  INFO --- MqttFX ClientModel             : messageArrived() with topic: $SYS/broker/load/messages/sent/1min
2019-07-15 20:25:22,393  INFO --- MqttFX ClientModel             : messageArrived() with topic: $SYS/broker/load/bytes/received/1min
2019-07-15 20:25:22,393  INFO --- MqttFX ClientModel             : messageArrived() with topic: $SYS/broker/load/bytes/sent/1min

I must be doing something basically wrong as I am a NOOB at MQTT....

Any help would be greatly appreciated...

jgyates, 2019-07-16:
The log file on the pi /var/log/genmqtt.log may have some details that are helpful.

The messages you listed in your post do not appear to be from genmon.

messages would look like 

/generator/Status/Engine/Switch State/Auto
/generator/Status/Engine/Engine State/Off

farberm, 2019-07-16:
No route to host error?
2019-07-14 12:39:11,259 : Error in MyMQTT init: [Errno 113] No route to host : genmqtt.py:337
2019-07-14 21:43:46,060 : Error in MyMQTT init: [Errno 113] No route to host : genmqtt.py:337
2019-07-14 21:45:11,900 : Error in MyMQTT init: [Errno 113] No route to host : genmqtt.py:337
2019-07-15 17:45:25,340 : Error in MyMQTT init: [Errno 113] No route to host : genmqtt.py:337
2019-07-15 17:50:18,699 : Error in MyMQTT init: [Errno 113] No route to host : genmqtt.py:337

I get email message and other notifications?

farberm, 2019-07-16:
Ok had a typo on port number

jgyates, 2019-07-16:
OK. I will close this thread. If you have any other issues on this topic you can post them in this thread.

closed 2019-07-16 by jgyates

farberm, 2019-07-16:
Are there a list of the available subscriptions somewhere, I currently can see 8 but I guess there may be more?

Utility Voltage, Max, Min - 3
Ambient Temp Senspor - 1
Battery Voltage - 1
Line Utility Voltage, Min, Max - 3

jgyates, 2019-07-16:
They follow the format of the text on the web interface.

/generator/Status/Engine/Switch State/Auto
/generator/Status/Engine/Engine State/Off
...
/generator/Monitor/Serial Stats/CRC Errors/0

etc

All subscriptions are active unless you specify key words in the blacklist on the genmon mqtt add on settings. For example if you use the word "Status" in the blacklist, it will not include any subscription that has Status in the path.

the flush interval parameter in the genmon mqtt setup controls how often the values are sent if then do not change. If flush interval is zero then the are all sent once, then only values that are sent are ones that change. If flush interval is 30 then all entries are sent every 30 seconds plus entries are sent if they change.

jgyates, 2019-07-16:
Also, I am not sure what the negative values you are seeing. The values on the status page can be sent as string or integer depending on the "JSON for Numerics" setting

farberm, 2019-07-16:
Those were not negative values, just the number of subscriptions, sorry about that. I can derive from the web page thanks for the input. Only other issue is the data location (path) which I am working on figuring out. The server is now communicating, I just need to determine where the MQTT server stores the data?

lakee911, 2020-10-28:
I am having trouble getting any data published from genmon to my broker. I tried monitoring with mosquitto_sub and nothing comes in. Flush is set to 10 seconds to force data. Looked at netstat and there is a connection made from genmon to the broker. I've checked and rechecked IPs, ports, and user/pass. Had one message in /var/log/genmqtt.log about a connection time out that I fixed ... nothing since. 

What's left to check?

Thanks.

jgyates, 2020-10-28:
You can add this line to /etc/genmon/genmqtt.conf

    debug = True

this will output additional data to /var/log/genmqtt.log that may be useful. This is disabled by default to conserve the life of your SD card so be sure to disable it when you have it working.

lakee911, 2020-10-29:
Nice.

Unfortunately, I see /var/log looking fine. You?

ettings/Rated Max Power: 14 kW: <type 'unicode'>
2020-10-28 21:14:17,735 : Confirmed:
generator/generator/Monitor/Communication Stats/Timeout Errors: 0
2020-10-28 21:14:17,741 : Publish:
 generator/generator/Maintenance/Exercise/Exercise Time: Weekly Monday
17:30 Quiet Mode Off: <type 'unicode'>
2020-10-28 21:14:17,751 : Publish:
 generator/generator/Maintenance/Service/Service A Due: 189 hrs or
12/28/2021 : <type 'unicode'>
2020-10-28 21:14:17,762 : Publish:
 generator/generator/Maintenance/Service/Service B Due: 348 hrs or
11/14/2021 : <type 'unicode'>
2020-10-28 21:14:17,768 : Publish:
 generator/generator/Maintenance/Service/Battery Check Due: 12/28/2020 :
<type 'unicode'>
2020-10-28 21:14:17,766 : Confirmed:
generator/generator/Monitor/Communication Stats/Timeout Percent Errors:
0.00%
2020-10-28 21:14:17,781 : Publish:
 generator/generator/Maintenance/Service/Total Run Hours: 132.00 h: <type
'unicode'>
2020-10-28 21:14:17,793 : Confirmed:
generator/generator/Monitor/Communication Stats/Modbus Exceptions: 0
2020-10-28 21:14:17,796 : Publish:
 generator/generator/Maintenance/Service/Hardware Version: V1.05: <type
'unicode'>
2020-10-28 21:14:17,808 : Publish:
 generator/generator/Maintenance/Service/Firmware Version: V1.07: <type
'unicode'>
2020-10-28 21:14:17,812 : Confirmed:
generator/generator/Monitor/Communication Stats/Validation Errors: 0

Still nothing coming in on the broker, that I can pick up

pi@mosquitto:/var/log $ mosquitto_sub -u iot -P ******** -d -t
generator/generator/
Client mosqsub|20689-mosquitto sending CONNECT
Client mosqsub|20689-mosquitto received CONNACK (0)
Client mosqsub|20689-mosquitto sending SUBSCRIBE (Mid: 1, Topic:
generator/generator/, QoS: 0)
Client mosqsub|20689-mosquitto received SUBACK
Subscribed (mid: 1): 0
Client mosqsub|20689-mosquitto sending PINGREQ
Client mosqsub|20689-mosquitto received PINGRESP
Client mosqsub|20689-mosquitto sending PINGREQ
Client mosqsub|20689-mosquitto received PINGRESP
Client mosqsub|20689-mosquitto sending PINGREQ
Client mosqsub|20689-mosquitto received PINGRESP
Client mosqsub|20689-mosquitto sending PINGREQ
Client mosqsub|20689-mosquitto received PINGRESP

jgyates, 2020-10-29:
The "publish" log entries show when the message was sent to the server.

The "confirm" log entries are an acknowledge that the server received the message. 

Not sure about the generator/generator unless you added that additional "generator" 

I can validate the default settings with this (root topic parameter is blank on the Add On page:

     mosquitto_sub -h localhost -t generator/# -u username -P password

lakee911, 2020-10-29:
You're the man! The issue was the omission of the # on my topic. Data was
coming through all along.

And, the generator/generator thing was my doing ... I did add that in my
effort to troubleshoot.

Thanks for the help again. Great product and great support!