jgyates/genmon #297: Feature Request for .jpg

swvalenti194, 2019-07-16:
Is it possible to make a .jpg of the status page that I can implement on my home automation software?  For example I have a radar loop on my home automation page that I refresh every 1 minute but it needs to be a .jgp in order to work.  Below is the radar loop image that I use in hopes that GenMon can do the same. 

http://dsx.weather.com/util/image/map/us_ne_9regradar_plus_usen.jpg

jgyates, 2019-07-16:
@swvalenti194 

I will think on this request, but my initial thoughts are this is a bit out of the scope of the project. There are several other ways to achieve this functionality using other utilities. You can also export the status data (in non image format) and convert it into a format of your choice in a few ways:

Assuming a pi address of 192.168.1.10:

    http://192.168.1.10:8000/cmd/status

this will give you a text output of the status page (no graphs or gauges)

     http://192.168.1.10:8000/cmd/status_json

This will give you a JSON formatted output of the status page (no graphs or gauges)

    http://192.168.1.10:8000/cmd/getbase

this will give you a one word status (READY, ALARM, EXERCISING, OFF, SERVICEDUE, RUNNING, RUNNING-MANUAL, or MANUAL

bdwilson, 2019-07-16:
I do something similar with data from Weewx (which is in json format).  I pull data from my Weewx json file, take a snapshot image from one of my cameras, then add a header to the top with the Weewx data.  Here is info on how to do that for @swvalenti194 assuming he has a server that has PHP and Curl. [snapshot.php](https://gist.github.com/bdwilson/abbcb028236cf97430970d0bc5e8572e)

swvalenti194, 2019-07-16:
Didn't realize it would be outside the scope my bad.

jgyates, 2019-07-16:
No worries, never hurts to ask.

closed 2019-07-17 by jgyates