jgyates/genmon #298: Platform Stats to MQTT broker

ronmillsap, 2019-07-16:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below.

***NOTE:*** If you are having issues with your serial connection, please read this page before posting:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

If you are having other issues, please see the following page:

https://github.com/jgyates/genmon/wiki/3.5---General-Troubleshooting

If you need to send you logs and registers to the developer, if you email is setup and working properly you can click send your logs on the About page in the web interface.

### Expected Behavior

I currently have Genmon running without issues. I have also set it up to pass all data to my Home Assistant MQTT broker and then to the influxdb instance of Home Assistant. everything is working great without any issues. 
I would find it very useful to incorporate the Genmon Platform Stats with the Home Assistant mqtt data but it doesn't appear to be included in the MQTT addon.
Is it possible to include some of the Platform Stats or is that unrealistic? 
I'm too dumb to write python code.

### Actual Behavior

Do not see platform stats as mqtt topic

### Steps to Reproduce (including precondition)

{Please write here}

### Screenshot or Pictures relating to the problem (if possible)

{Please write here}

### Your Environment

- Generator Model: Generac
- Generator Registers: Submit via the About page
- Genmon Version: current (25)

jgyates, 2019-07-16:
all of the text data on the status, maintenance, outage and monitor pages are included in the MQTT data. I would look at the "Blacklist Filter" setting on the maintenance page. The default settings block any data that has the following words in the full path:  

    Monitor,Run Time,Monitor Time,Generator Time,External Data.

If you edit this setting and remove the Monitor word from the list you will see info from the monitor page. You can remover everything to get all data.

ronmillsap, 2019-07-16:
Ok, got it.
Sorry that didn't click with me. That proves that I'm not too good at writing code.
Thanks for such a quick reply.

jgyates, 2019-07-16:
No problem. I will close this thread. If you have any other questions on this topic feel free to post them here.

closed 2019-07-16 by jgyates