jgyates/genmon #299: How to tell if Evolution 1 or 2

bdwilson, 2019-07-18:
I've read the docs a few times but haven't seen it discussed how to tell if you have an Evolution 1 or 2. Is it different hardware version or the firmware on the unit? I tried looking for pics of one vs. the other on google images, but couldn't really tell. Mine has "accessory" ports, installed in 2014, so I'm assuming it's a v2. How can I find out for sure. Thanks!

jgyates, 2019-07-18:
If it was 2014 then I would assume it is an Evolution 1. Evo2 did not come to the market until 2018 I believe. Genmon will identify the controller type and display it on the Maintenance page:  "Evolution" for Evo1 and "Evolution 2.0" for Evo2.

Also, the Evolution 2.0 has the option for having wifi. Evo1 does not.

bdwilson, 2019-07-18:
Ahh misread the part on making the cable. Evolution 'adds' these pins. The Evo1 pins are the same though. Thanks!

closed 2019-07-19 by jgyates