jgyates/genmon #300: Clarify serial testing language

bdwilson, 2019-07-19:
Was hoping you could help me clarify this wording, if my assumption is correct, maybe update the wording. Maybe it's a left brain/right brain thing :) For the first serial test this is the language:

> Loopback at the Pi Expansion Connector This test assumes you are using the internal Pi serial port. Loopback pin 8 (transmit) and pin 10 (receive) at the pi expansion connector. This test requires no cables and will validate that your serial port is setup for use by the program. The program can be executed with no additional parameters if you are using the build in serial port on the pi and it has been assigned to /dev/serial0. From the OtherApps directory type:
> 
>  python serialtest.py
> If this test passes then you have your pi setup correctly.
> 
> If it fails or passes intermittently, check your connections, power and ground for the pi and validate you have the settings in /boot/cmdline.txt and /boot/config.txt correct if you are using the build in serial port. A common issue if this test fails is skipping a step on the setup. Use serialconfig.py to validate this.

The statement "this test requires no cables" which I assumed meant I didn't have to do anything.  I assume you still have to short/connect pins 8 and 10 in order for this test to work, and one of the ways to short them is with a dupont cable.  Perhaps this wording would work?

> This test assumes you are using the internal Pi serial port.  With your RS232 board disconnected from the PI GPIO pins, connect both pin 8 and pin 10 on the PI GPIO board together using a wire/dupont cable.  Connecting these pins and running serialtest.py will validate that the serial port has been correctly configured on your Pi and that you are ready to connect your RS232 board.  Then proceed to the next test. 

Thanks for helping me clarify.

bdwilson, 2019-07-19:
[image]
For the next testing section, if you want to show how to connect the pins to test, here is a pic I took.

jgyates, 2019-07-19:
@bdwilson 

Thanks! I will update the wiki with this info.

jgyates, 2019-07-19:
https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

closed 2019-07-19 by jgyates

waynewarburton, 2019-12-06:
I'm a novice having troubles checking the serial port...more-so using the Thonny software.  When I run "serialconfig.py", text reads "option not selected".  I know it's looking for me to input "e, r, or c" to enable, restore, or check but can't figure out how to input this information in. 

Hardware I am good at, software...very new to it.  The site is very geared towards experienced programmers and this I am not.  I have everything wired on my desk and running Thonny (haven't worked with basic level software since the late 80's).  I have read many links today and none touched on this (almost all I did today was read links other than blow out the driveway).  Any help would be greatly appreciated!

jgyates, 2019-12-06:
I am not familiar with Thonny. I use a console to run python programs. Once you open a console, from a console you type:

       sudo python serialconfig.py -c

the "sudo" part runs the program with administrator (administrator is named "root" on linux systems), the "python" is running the python program which interprets the program file. The python source file is "serialconfig.py" and the "-c" is the options passed to serialconfig.py.

Also, instead of posting questions to a random thread, please either post to a new thread or to a thread that is specific to the topic in question. Adhering to this helps keep the questions organized so others can benefit. From this link you should be able to create a new issue or search for old issues (closed issues will not show up unless you search for them). Issues are tagged by topic so you can search them easier.

waynewarburton, 2019-12-06:
Ok Thanks for the advice on a new link.  I'll copy/paste this over.  
I have tried "sudo python serialconfig.py -c" in the console and I get a response back "python can't open file 'serialconfig.py': [Err02] No such file or directory' .  No doubt something simple I'm missing.

bdwilson, 2021-12-04:
the other thing that needs to be noted many times, serialtest can't be run if genmon is running. I don't know how many times I've run into this but it should be updated in the docs somewhere. @jgyates Doing an install on RPI Zero 2 tonight.

jgyates, 2021-12-04:
@bdwilson 

It is listed here:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting#run-a-loopback-test

and here:

https://github.com/jgyates/genmon/wiki/1----Software-Overview#otherappsserialtestpy-optional

Let me know if you think this needs additional clarification.

bdwilson, 2021-12-04:
I was thinking maybe an error message from serialtest.py if wrong output is seen, perhaps even check for genmon running when running the test. I’ll see if I can work on a PR to do this.

jgyates, 2021-12-04:
I just checked in an update that will fail reliability if genmon is loaded and print and error from serialtest.py stating you should stop genmon. I had to add an exclusive flag to the serial port open on both genmon and serialtest.