jgyates/genmon #301: Notification error

farberm, 2019-07-19:
I am getting the following email notifications from genmon about twice a day

Outrage Recovery Notice

Time: 01:17:01 PM
Date: Fri 19-Jul-2019

Utility Power Restored. Duration of outage 0:00:00

There is no outage at the house. Is this an error with the system or is there something else going on?

I have restarted genmon several times and also has mqtt turned off. For the first week I had genmon running this was not he-penning

Any guidance or help would be appreciated

### Your Environment

- Generator Model: Evolution 2, Lia propane 20 kw
- Generator RegisteR - submitted
- Genmon Version: 1.13.23 and also issue with 1.13.25

jgyates, 2019-07-19:
These messages are sent when the software detects and outage. Evolution controllers have three values that are relevant to what is defined as an outage. 

     1) The utility voltage. For the US this value is nominally 240V, +/- 10%
     2) The threshold voltage - This is the voltage level that if the utility voltage goes below it is considered an outage
     3) the pickup voltage - This is the voltage level that the voltage must rise above to be considered out of an outage (typically a little higher than threshold voltage)

The generator has a startup delay parameter (see the outage page for your value) that you can set in the controller menu. If the utility voltage drops below the threshold voltage for the number of seconds in the startup delay (default is 5 seconds) then the generator attempts to start.

While the controller will not start the generator unless . the utility voltage is below the threshold voltage value for 5 seconds, the genmon software will send a notice if it detects the utility voltage is below the threshold for any amount of time. This is to log brown outs and other utility anomalies. 

You also have the option to  disable outage monitoring by enabling the Do Not Check for Outages setting on the settings page.

My guess is that you have brief fluctuations in your utility voltage, which is not that uncommon in itself. Also, sometimes the can come in clusters, I am guessing due to either the power company not producing enough power, lightening, or working being done on power lines, etc. To put this into perspective, this is my outage log for the last year:

Outage Log : 
        2019-07-02 11:46:16, Duration: 2:55:39, Estimated Fuel: 2.18 gal
        2019-06-10 18:25:42, Duration: 3:54:38, Estimated Fuel: 3.16 gal
        2019-06-10 18:10:19, Duration: 0:00:03, Estimated Fuel: 0.00 gal
        2019-04-26 14:42:34, Duration: 0:00:19, Estimated Fuel: 0.00 gal
        2018-12-09 06:03:35, Duration: 13:37:28
        2018-12-09 06:03:04, Duration: 0:00:19
        2018-12-09 04:43:28, Duration: 0:00:03
        2018-12-09 04:43:22, Duration: 0:00:02
        2018-12-09 03:50:46, Duration: 0:00:00
        2018-12-09 03:50:42, Duration: 0:00:00
        2018-11-19 23:15:23, Duration: 0:00:31
        2018-11-19 23:15:19, Duration: 0:00:02
        2018-10-31 10:07:32, Duration: 4:17:53
        2018-09-24 07:41:40, Duration: 0:00:06
        2018-09-24 07:32:29, Duration: 0:00:02
        2018-09-24 07:32:18, Duration: 0:00:02
        2018-09-24 07:32:06, Duration: 0:00:00
        2018-09-24 05:37:32, Duration: 0:00:01
        2018-09-24 05:37:29, Duration: 0:00:01
        2018-09-24 05:37:26, Duration: 0:00:00
        2018-09-24 05:37:24, Duration: 0:00:01
        2018-09-24 05:37:20, Duration: 0:00:01
        2018-09-24 05:37:18, Duration: 0:00:00
        2018-09-24 05:37:17, Duration: 0:00:00
        2018-09-24 05:37:16, Duration: 0:00:00
        2018-09-24 01:58:10, Duration: 0:00:01
        2018-09-24 01:58:07, Duration: 0:00:01
        2018-09-24 01:58:04, Duration: 0:00:01
        2018-09-24 01:58:02, Duration: 0:00:00
        2018-09-24 01:58:00, Duration: 0:00:01
        2018-09-24 01:57:57, Duration: 0:00:01
        2018-09-24 01:57:53, Duration: 0:00:01
        2018-09-24 01:57:52, Duration: 0:00:00

Any duration of less than 5 seconds, the generator never attempts to crank.

Of course the other problem could be if you are experiencing an serial errors that correspond to your outage times, the serial errors could be causing invalid data to be received. You can see the times of any serial errors in the file /var/log/mymodbus.log and /var/log/myserial.log. 

You do have serial errors in your log, the last day with serial errors was on July 17th. Have you had outage notices since then?

jgyates, 2019-07-19:
Sorry, I got two sets of logs at the same time, your last serial errors were on 7/15

farberm, 2019-07-19:
Yes I have notifications yesterday and today of outages. If I turn off outage Genmon will still notify me if the generator starts correct?

jgyates, 2019-07-19:
Correct, you will just not get outage emails and the outage log will not be updated. You will still get notices when the state of the generator changes.

closed 2019-07-22 by jgyates