jgyates/genmon #302: Consumption question

bdwilson, 2019-07-20:
I’m up and running! Thank you for the helpful answers, all the code and all the docs. Much appreciated!  One thing I wanted to check on was fuel consumption. I have a propane tank that supplies other things in our house - not just the generator, so I’m trying to see how the fuel gauge would work for me. It’s at 60% now, so how does it calculate usage?  What would be nice would be instead of a gauge to show how much fuel is left, have an option to show fuel used in current running session (or last session if not running). I saw some other issue posts around consumption and the 30 day number but I’m unsure if this is calculated after a running session or if during it’s also calculated, and it sounds like some of this is on the logs now, so not a biggie. I can see the power going out this winter and us wanting to know how much fuel has been used -perhaps it’s already there but doesn’t show until the generator has run (which I haven’t done yet). Thanks again for your work on this. My donation is coming!

bdwilson, 2019-07-20:
[image]

Here are the gauges I see now.

bdwilson, 2019-07-20:
FYI, your donate button goes to eBay.

jgyates, 2019-07-20:
@bdwilson 

Thanks, I updated the paypal link.

This page has a description of how fuel monitoring works:

https://github.com/jgyates/genmon/wiki/Appendix-H-Monitoring-Fuel-and-Power-Usage

In short,  the software can estimate the current and know the exact voltage so power can be calculated and from there a percentage of the load of the maximum the generator can produce.  This percentage is applied to the fuel consumption algorithm for your specific generator engine to produce a volume of fuel over a time period. This volume is subtracted from the size of your tank (assuming a full tank). There is a parameter on the advanced page (click the gear icon in the upper right twice) that will allow you to subtract fuel form the estimate. This approach only is accurate if the generator is the only appliance using the tank. The maintenance page shows fuel used over 30 days and the amount of fuel in the tank (estimated). The add for a third party tank meter from www.tankutility.com is supported on the add on page (see External Tank Fuel Monitor on the add on page). this may be a better solution to monitor your tank.

For fuel monitoring the gauge does show how much is left, but again it assumes that you start with a full tank or have subtracted some of the volume off the full tank.

Let me know if you have any other questions.

bdwilson, 2019-07-20:
Does the maintenance page update fuel used while the unit is on generator or does it update after it’s stopped? How hard would it be to have a toggle in the settings to show fuel used instead of fuel left? Could also have a setting in settings to reset it. I haven’t dove into the code to see how hard it would be to add but will do that.

jgyates, 2019-07-20:
I will add an entry to the maintenance page that will show the total fuel used. Note that this value is the total calculated fuel used for the readings reported in the power log. To reset this value you can reset the power log on the Maintenance page.

To show fuel for a give time period you can enter this URL:

http://192.168.11.10:8000/cmd/power_log_json?power_log_json=0,fuel

the zero in the command specifies all entries in the power log.

http://192.168.11.10:8000/cmd/power_log_json?power_log_json=43200,fuel

Fuel used in the past 30 days, 43200 minutes in 30 days

You can change the last two values like this:

The second to the last value is the number of minutes, zero minutes is for all entries in the log.
The last parameter is either "kw", "fuel" or "time". kw returns kWHours, fuel returns the volume of fuel, time returns run hours.

I should have a check in with the update to the maintenance page later today or tomorrow. I will let you know once it is there,

bdwilson, 2019-07-20:
Thank you. I will try to get the json data into Hubitat so I can display it in my dashboard. Also, as an FYI, I’m running newest buster release with no (apparent) issues.

jgyates, 2019-07-20:
Also, this command:

http://192.168.11.10:8000/cmd/power_log_json?power_log_json=0

will import the power log reduced to 500 entries. No last parameter gives the raw power log entries for the time period given.

jgyates, 2019-07-20:
The kw, fuel and time do not return json, they just return a string of the value. If you omit the last parameter then json is returned for the power log entries.

jgyates, 2019-07-21:
V1.13.27 is checked in with the update mentioned above. Let me know if you have any questions. I am closing this for now but feel free to post any related questions to this thread.

closed 2019-07-21 by jgyates