jgyates/genmon #303: Genalexa question

RoyRomeo, 2019-07-28:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below.

***NOTE:*** If you are having issues with your serial connection, please read this page before posting:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

If you are having other issues, please see the following page:

https://github.com/jgyates/genmon/wiki/3.5---General-Troubleshooting

If you need to send you logs and registers to the developer, if you email is setup and working properly you can click send your logs on the About page in the web interface.

### Expected Behavior

Ability to connect to Alexa

### Actual Behavior

Amazon Alexa voice command app enabled, UPnP on router enabled, Alexa cannot discover generator.

### Steps to Reproduce (including precondition)

{Please write here}

### Screenshot or Pictures relating to the problem (if possible)

{Please write here}

### Your Environment

- Generator Model:  0062440
- Generator Registers: Submit via the About page
- Genmon Version: V1.13.28

jgyates, 2019-07-29:
@RoyRomeo 

One thing that you can do to help determine the source of the problem is to edit the file /etc/genmon/genalexa.conf and enable additional debug info to be collected.

     sudo nano /etc/genmon/genalexa.conf

Then add the line:

    debug = True

Save the file by typing Ctrl+X, then type the letter y for yes, then enter.

Then restart genmon

    cd /home/pi/genmon
    ./startgenmon.sh restart 

Let genmon run and try to discover the generator. Then submit your logs on the about page.

@buschauer will be helping me debug this as he coded the genalexa app.

MichaelB2018, 2019-07-29:
Also, did you try discovery twice. I did see it before that the first time it timed out too quickly after 20 secs. So try a second time and enable debug so we see what's going on.

jgyates, 2019-07-30:
@RoyRomeo 

I checked in an update to genalexa.py (i did not update the version number) that adds some additional debug output. Can you give it another shot and send the log again?

Also, what version Echo do you have?

To clarify, when you discover devices, do you get any response at all?

RoyRomeo, 2019-07-30:
Okay, logs sent.I believe I have the 1st generation Echo.I get "No new devices found" when I run discover devices.

jgyates, 2019-07-30:
@RoyRomeo 

I checked in one more update with some more debug info. Repeat the same process except do the discovery twice.

RoyRomeo, 2019-07-30:
Repeated the process and did the discovery several times, no success.

jgyates, 2019-07-30:
Can you send the logs? The update contained additional debug info.

RoyRomeo, 2019-07-30:
Just did.

jgyates, 2019-07-30:
I got your logs. Did you update to the latest version of genalexa.py? I do not see the additional debug info in the log file.

RoyRomeo, 2019-07-30:
Where do I find the latest version of genalexa.py?

jgyates, 2019-07-31:
Just go to the About page and click "Update to the latest". It will download the latest and restart the software.

jgyates, 2019-07-31:
Also, I did not update the version info since this update only provides more debug info so it will tell you you are running the latest but update anyway.

RoyRomeo, 2019-07-31:
Updated and resent logs.

RoyRomeo, 2019-08-01:
Did the update.  Not sure if I submitted the logs last night, so I submitted again.

jgyates, 2019-08-01:
Got them. thanks! I will let you know what we find.

jgyates, 2019-08-04:
@RoyRomeo 

Michael suggested trying this:

With a valid Alexa speaker in the house, use http://alexa.amazon.com. Select "Smart Home". Select "Devices". Scroll all the way down. Note down the number of devices ("Showing XX). Then select "Discover". I sometime have to do this twice myself. Not quite sure why. For debugging purposes try this 3 or 4 times, unless he sees the number of devices increasing. *IF* it increases (which it does for me), the new one will be listed at the top. Then send the log again.

RoyRomeo, 2019-08-04:
Followed these steps, 1 device listed.  Selected "Discover" at least 6 times, but no new devices discovered.

MichaelB2018, 2019-08-04:
can you submit the logs again. Thanks.

RoyRomeo, 2019-08-05:
Okay, just submitted them.

jgyates, 2019-08-07:
@RoyRomeo, below are some instructions from Michael for running a test using the library that genalexa is based on. This should tell is if the problem lies in the library or this projects implementation:

The log file is not providing any useful info. Not sure you has anything special in his network, maybe a internal firewall or similar? But let's take a step back and try the original FauxMo Library. 

    Launch Pi ssh session with putty or locally through VNC and type following two commands Pi command prompt to update it. "sudo apt-get update" and "sudo apt-get upgrade" (This will take a while)

    Download this github project as zip file with following command "wget 
     https://github.com/nassiramalik/IOT-Pi3-Alexa-Automation/archive/master.zip"

    Unzip downloaded zip file with "unzip master.zip" command and type "cd IOT-Pi3-Alexa-Automation-master" command after unzip completes

    Enter "sudo python3 RPi_name_port_gpio.py" command to run Pi IOT program

    Give voice command to Alex to discover devices "Alexa discover devices" it will search your network and discover your Raspberry Pi 3 as an IOT device.

And let's see if this discovers any devices. There is also a nice Video to show Fauxmo in action

https://www.youtube.com/watch?v=uS5dTx8vjq4

and documentation is here

https://github.com/nassir-malik/IOT-Pi3-Alexa-Automation

RoyRomeo, 2019-08-08:
Hi,
Followed this process.  After the upgrade, there was a "node-red" error message stating that not all files were upgraded, some were left unchanged.  After running the last command, "sudo python3 RPi_name_port_gpio.py", there was an error that RPi did not exist.  Anyway, tried "Alexa discover devices", no new devices found.

 https://github.com/nassiramalik/IOT-Pi3-Alexa-Automation/archive/master.zip"

And let's see if this discovers any devices. There is also a nice Video to show Fauxmo in actionhttps://www.youtube.com/watch?v=uS5dTx8vjq4and documentation is herehttps://github.com/nassir-malik/IOT-Pi3-Alexa-Automation—

liltux, 2019-08-08:
@RoyRomeo , you have one of my pre-built devices, correct?  If so, I am using node-red on the devices for wifi configuration.  If you are not using a pre-built device, if you are using the version of raspbian that installs "Everything", I believe node-red is installed.  During and apt-get update and apt-get upgrade, any additions to the main work flow to your node-red install will not be updated.  Does not mean there is a problem, just if you wish the extra node-red modules to be updated you will have to perform manually through npm or node-red scripts.

jgyates, 2019-08-08:
Just curious if anyone else has tried genalexa besides @RoyRomeo or @MichaelB2018? It looks like there may be some limitations with the FauxMo code that was used to built this. I don't have an Amazon dot so I am unable to test this.

RoyRomeo, 2019-12-01:
Just got two new Amazon EchoShow 5 units and installed them.  Went through the discovery process and they found my generator right away.  I can now "turn on" and "turn off" the generator via Alexa.  I guess my older Amazon Echo does not have the technology that is needed to discover the generator.

jgyates, 2019-12-01:
That is a good data point. I will update the document to note this. Thanks for the follow up.

closed 2019-12-01 by jgyates

rwskinner, 2022-07-05:
I have 2 Show 5's and several Echo Dots, and none of them will discover the generator using GenAlexa.  I also turned on debug and submitted the logs but they keep failing to send.

The genalexa.log just shows the same thing over and over no matter how many times I try to discover.

2022-07-05 08:49:23,133 : Responding to search for generator
2022-07-05 08:49:23,134 : HTTP/1.1 200 OK
CACHE-CONTROL: max-age=86400
DATE: Tue, 05 Jul 2022 13:49:23 GMT
EXT:
LOCATION: http://XXX.XXX.XXX.XXX:52004/setup.xml  #XX blanked out for privacy
OPT: "http://schemas.upnp.org/upnp/1/0/"; ns=01
01-NLS: 73e844fd-1c59-442c-bdd4-90a98150c32b
SERVER: Unspecified, UPnP/1.0, Unspecified
ST: urn:Belkin:device:**
USN: uuid:Socket-1_0-3c767656e65726::urn:Belkin:device:**
X-User-Agent: redsonic

aaciii, 2024-07-21:
I have a Gen 3 Amazon Echo Dot, and I cannot get the generator to be discovered either.