jgyates/genmon #304: Serial Comm Loss after Firmware Update - restarting controller doesn't fix

jwtaylor310, 2019-07-29:
### Your Environment

- Generator Model: Generac 22kW Evolution II
- Genmon Version: V1.13.26/28

Genmon is reporting that it has lost communication with the generator.  The problem started several hours after Genmon reported a controller firmware change from 1.07 to 0.98 and then a few seconds later it reported a change from 0.98 to 1.07.  I have removed power from the controller and restarted it with no improvement.   I have followed the troubleshooting steps for serial communications and "serialtest.py" reports no problem with the loopback installed at the generator.

The sequence of events:

24-Jul 03:34:49 AM alert emailed stating that the firmware changed from V1.07 to V0.98
24-Jul 03:34:54 AM alert emailed stating that the firmware change from V0.98 to V1.07
24-Jul 12:35:58 PM alert emailed stating that Genmon is not communicating with the controller

No updates have been received since then.  Today I have:

1. Reset generator controller 
2. Updated Genmon from V1.13.26 to V1.13.28
3. Reset generator controller
4. Power cycle Raspberry running Genmon
5. Stopped Genmon, installed loopback at generator, serialtest.py runs without error

Any suggestions?  Thank you!

jgyates, 2019-07-29:
Is this the procedure you used to reset the controller?

https://github.com/jgyates/genmon/wiki/Appendix-I-Power-Cycle-the-Evolution-or-Nexus-Controller

jwtaylor310, 2019-07-29:
I used that procedure the first time.  The second time, I removed the battery connector and disconnected the incoming AC power.  In both cases, the controller display went blank and upon restoral went through the setup procedure.  Neither procedure helped.  --John

 

https://github.com/jgyates/genmon/wiki/Appendix-I-Power-Cycle-the-Evolution-or-Nexus-Controller

jgyates, 2019-07-30:
@jwtaylor310 

If you have validated your cabling and your ground connection to the pi (both would cause serial to not work). If you ground wire to the pi (either from the battery or the molex connector) is not making a good connection to the pi then serial will not work, even if a loopback does. Vibrations from the generator or corrosion on the battery terminals could case issues with grounding.

The firmware update notice is sent when the contents of the firmware version register changes. Other than that, there is nothing special about this email notice.  I don't know what is going on in the Evolution 2 firmware but I would guess that one of the following is occurring:

1) A serial timeout occurred and the controller responded after 3 seconds and genmon had already sent another request. This could cause the register requests and responses to get out of sync which could cause the software to record the wrong value for the register. If this occurs this is likely due to an underlying serial integrity problem.
2) The Evolution 2 firmware has a bug that is reporting the incorrect value of the firmware version register.

After validating the cable and ground, when you power cycle the controller, how long are you letting the controller sit unpowered? You may try it for a longer period of time.

jwtaylor310, 2019-07-30:
This one has me stumped….

 

I have a serial cable running from the generator into my garage (a few feet away) and the Raspberry and RS-232 adapter is in the garage and powered by a “wall wart” plugged into a local UPS.  The serial cable runs from the Raspberry’s adapter to the molex connector at the generator.  At the generator it is connected to pins 7 & 8 (RXD/TXD) and 4 (Ground) on the molex connector.  I checked continuity from pin 4 on the molex connector to the ground connection on the Raspberry’s RS-232 adapter and it tests less than 1 ohm.  I tried moving the ground connection from pin 4 on the molex connector to the generator’s battery negative post and also to sheet metal on the chassis with no packets received.  However, my meter shows an open from pin 4 on the generator side of the molex connector to the battery negative post  and to the generator sheet metal so it appears that the RS-232 ground is isolated from the chassis – does that make sense?

 

The system was installed in May of this year and communication was flawless until this problem.  At one point the system showed over a million packets received with no timeouts or other errors.  There were no hardware or software changes made prior to the failure.

 

This generator also has Mobile Link connected via WiFi and RS-485 through the same molex connector and Mobile Link updates normally.

 

I have reset the controller repeatedly including several times with “off” periods in excess of 15 minutes.  In every case the controller restarted by going through the initial setup as expected.

 

I appreciate how difficult it is to troubleshoot this type of problem remotely.  Any suggestions will be most welcome as I haven’t a clue as to how to fix this.

liltux, 2019-07-30:
Try turning off the generac wifi and unplug the generac wifi.  Then power up the controller and see if comms returns?

jwtaylor310, 2019-07-30:
No joy.  WiFi off, WiFi cable disconnected, controller restarted.  Genmon still doesn’t see any packets.

jgyates, 2019-07-30:
Is there a ground connection from the RS-232 board to the pi ground? If not that could be your problem. The controller, the RS-232 adapter and the pi must share a ground. A picture of your cabling may help to validate it is setup properly. If your ground is not correct, it *might* work, for a while at least.

jwtaylor310, 2019-07-30:
Yes, there is a ground from the RS-232 board to the Pi.  The first photo shows the adapter with the cable to the generator on the right and the jumpers to the RPi on the left.  The second photo shows the jumper connections at the RPi.  The green jumper is the ground.  

 

The third photo shows the molex plug.  The yellow wire on the left is RXD (Pin 7), the one on the right is TXD (Pin 8).  The labelled black wire is ground (Pin 4).  All other wires pass through to another molex connector that attaches to the WiFi connector.  The yellow and black wires are soldered to the cable running to the RPi with color code Black-RXD, Red-TXD and White-GND.   The RPi end of this cable is on the right in the first photo.  The solder connections are encased in shrink-wrap tubing so I can’t provide a photo of them, but I have verified end-to-end connectivity and that there are no shorts between any combination of the three.

 

The only shared ground between the RPi and the generator is from the RS-232 adapter at the RPi to Pin 4 of the molex connector.  The RPi gets its power from a 120VAC “wall wart” so there is no ground connection to the generator through the RPi’s power supply.

 

Is Pin 4 the correct ground point for the RS-232?  I see that Pin 2 is also labelled “ground”.

liltux, 2019-07-30:
Check pin 2 to frame ground.   I believe I always used pin 2 when bring in ground from controller.  Worth trying.

jgyates, 2019-07-30:
@jwtaylor310 

You must attached photos from the github web interface. Replying to an email thread will post your comments but attachments are stripped so we did not recieve your pictures.

jwtaylor310, 2019-07-30:
I’ll give pin 2 a try.  Do you know if the RS-232 ground should show continuity to the frame and/or battery negative?

 

There’s a lightning storm nearby so I may not be able to check this until tomorrow.  Wouldn’t want to have the generator enclosure open and control disabled in the event of rain and/or a power outage!

jgyates, 2019-07-30:
Since you are running with a wall wart, I would make sure that your ground on the molex and the ground on the pi GPIO header are connected. If you powered a pi with a wall wart and the pi ground is not the same RS-232 will not work reliably. Zero to Five volts out of the wall wart would be interpreted as some random voltage  with a difference of 5 volts (for example 50V - 55V) instead of 0V to 5V.

If the ground pin  on both sides (the 9 pin side an the 4 pin side) of the RS-232 connector have  continuity then you are probably OK (assuming all connections are good). If they do not then your pi is not grounded with with the controller ground and this is likely the source of your problem.

jwtaylor310, 2019-07-30:
Photos are attached...
[image]
[image]
[image]

jwtaylor310, 2019-07-30:
I switched the ground from pin 4 to pin 2 on the molex plug and the system is now communicating normally.  Thank you for your help and your patience!

closed 2019-07-30 by jwtaylor310

jgyates, 2019-07-30:
OK. That is good to know. I am not sure where the info for pin 4 came from. I don't have an Evolution 2 but this was added to the wiki. You could have a bad pin on pin 4 on your molex or there could be something about pin 4 we don't understand yet.

jwtaylor310, 2019-07-30:
I originally picked pin 4 as the ground since it is called out as the "RS-232 port ground" under the "Evolution 2.0 Controller Aux Port" section here: https://github.com/jgyates/genmon/wiki/3.1--Making-a-Cable.  Since that pin wasn't connected to anything else in the generator, it was easy to just connect the pin to the ground connection to Genmon.

It looks like pin 2 is normally used as the ground for the WiFi module so it requires a "Y" connection to allow both Genmon and the WiFi module to connect to that pin.  That makes the cabling a bit messier, but if that is what it takes to get the system to communicate, so be it.

I don't understand why the original wiring worked flawlessly for over 2 months.  As you have suggested, it could be a flaky molex connector but that seems unlikely.  In any event, I'm happy the change has fixed the problem and appreciate your continued support!

liltux, 2019-07-30:
It could also be something Generac is controlling in the firmware.  Also those grounds through the circuit board are not all common.  As in some travel through other components to get to ground on the PCB, possibly.

jgyates, 2019-07-30:
I will make a note of this in the wiki.