jgyates/genmon #305: Genmon as a Wireless Modbus Ethernet/Serial Gateway to HA

wackyhobo, 2019-07-30:
Great work on Genmon. I set it up in no time and it works great. I also use Home Assistant which has a Modbus component. Instead of using MQTT for Genmon to talk to Home Assistant, is it possible for Genon to act as an Wireless Modbus TCP server for Home Assistant to utilize?

Is it also possible to interface with my Generac to Home Assistant with a  Modbus Ethernet/Serial Gateway such as B+B MERS901?

jgyates, 2019-07-30:
@wackyhobo 

Genmon can use serial over IP devices that transparently pass all serial data over a TCP socket.

https://github.com/jgyates/genmon/wiki/Appendix-F----Serial-over-IP

Genmon does not act as a modbus TCP server for other applications. I assume you could attach a serial gateway to your generator and genmon and HA could access the data.

jgyates, 2019-07-30:
Closing this issue for now. Feel free to post any follow up questions to this thread.

closed 2019-07-30 by jgyates

wackyhobo, 2019-08-11:
Now that I have successful connected Genmon (V1.13.2) to my Generac Guardian 22kW Generator (G0065521) using Serial RS232 to Ethernet Converter (USR-TCP232-410S), I would like to attempt the modbus integration into Home Assistant. Would you be so kind to email me the complete modbus mapping? Looking at you work some of the resisters are obvious but most is not clear to me. Any other related modbus documents that you want to send me I would most appreciative. Thank you.

jgyates, 2019-08-12:
Unfortunately, I don't have any documentation. I determined the functionality by observing the modbus registers change over time or when the generator is in use. Other people helped with testing since I only have a liquid cooled evolution. The best place to start is here in the code:

https://github.com/jgyates/genmon/blob/master/genmonlib/generac_evolution.py#L109

This contains a data structure with modbus register values and the comments for each entry in the data structure are the closest I have to what you are asking for. If you have specific questions I can answer them. 

All of the evolution and nexus specific code is in /genmon/genmonlib/generac_evolution.py. In general you can get the registers of interest and brief description from the data structure in the link above and the search the code for the register to see the code interpreting the value. For example, register 0019 is model ID register for Evolution Air Cooled models. If you search the code for . "0019" (include the quotations in the search) you will find the code that parses this and the associated values.

There are many items that are the same for all generac residential variants (Evolution Liquid Cooled, Evolution Air Cooled, Nexus Liquid Cooled , Nexus Air Cooled, Evolution Power Pact, Evolution Air Cooled 2.0), however there are many differences between these models. The code uses a set of Flags which are set early in the load process when the controller type is identified. The controller identification takes place here by reading register "0000":

https://github.com/jgyates/genmon/blob/master/genmonlib/generac_evolution.py#L824

The modbus spec is available here:

http://www.modbus.org/docs/Modbus_Application_Protocol_V1_1b.pdf

I never have taken the time to document the registers since this project has grown in an organic way and the preference has been to document the softwares use over the generators behavior.

wackyhobo, 2019-08-12:
How did you decode/ interpret /define states of the following Modbus Addresses:

1.	Switch, Engine and Alarm Status (0001)
2.	Last Alarm Code (05f1)

 

3.	What is the Modbus addresses for the logs and how do you interpret the modbus value?

 

https://github.com/jgyates/genmon/blob/master/genmonlib/generac_evolution.py#L109

https://github.com/jgyates/genmon/blob/master/genmonlib/generac_evolution.py#L824

http://www.modbus.org/docs/Modbus_Application_Protocol_V1_1b.pdf

jgyates, 2019-08-12:
If you are asking how I figured this stuff out, the first attempt at decoding was done initially by intercepting traffic in both directions while running Mobile Link. The first version I had running (before I released it on github) ran alongside mobile link. this only got me so far and then I transitioned to a version that ran without mobile link and I instrumented the code to look for bits that changed and this allowed me to find other registers an many of the bit values were obvious once you get into the mindset of what you think generac is doing. Some of the values took a while to figure out an others came pretty quickly. There are still quite a few registers that change that I do not know exactly what they do.

There were also some values that I had to test by removing the connections to some sensors on my generator (low coolant alarm, some of the status bits). THere were also some info that was helpful in the generac service manual for the bits in the dealer menu that I had to piece together from the instrumented software an the info in the service manual.

Register 0001 is one of the registers that the software treats as a 32 bit value, but it is actually just reading register 0001 at 0002 at the same time via a single modbus command. when read as a single 32 bit value the format is this:

xxxxyyyy

where xxxx is the Engine state:
The function GetEngineState decodes this an is located here:

https://github.com/jgyates/genmon/blob/master/genmonlib/generac_evolution.py#L2681
and yyyy is the switch state or alarm value if an alarm is active. You can not read the switch state when an alarm is active.
0000   - auto
0007 - off
0006 - Manual
00017 - two wire start active

The rest of the decode of the yyyy component comes is decodec in the table in the function GetAlarmState here 

https://github.com/jgyates/genmon/blob/master/genmonlib/generac_evolution.py#L2497

If the generator is in auto and off and not alarm, this value will read 00000000. 

The alarm code in register 0001 is an internal value . to the controller an not the same thing as the Generac published E-Codes.

Register 5f1 returns the last E-Code (in hex) since the controller booted. This register is for Evolution only since Nexus controllers do not use the concept of E-Codes. genmon uses the file /genmon/data/ALARMS.txt to cross reference the value read to an documented E-Code value (decimal). 

The logs are a bit detailed. Evolution has three (Alarm, Maintenance, and Start/ Stop) separate logs, Nexus only has two (no maintenance log). All 5 of these logs are at different addresses. Each log hold 50 entries, each entry is either 4 or 5 consecutive registers. Log entries look like this:

Log Entries are in one of two formats, 16 (On off Log, Service Log) or 20 chars (Alarm Log)
         AABBCCDDEEFFGGHHIIJJ
           AA = Log Code - Unique Value for displayable string
           BB = log entry number
           CC = minutes
           DD = hours
           EE = Month
           FF = Date
           GG = year
           HH = seconds
           IIJJ = Alarm Code for Alarm Log only

Note that the log alarm code is a separate decode than E-Codes or register 0001. These decodes are located in the function ParseLogEntry located here:

https://github.com/jgyates/genmon/blob/master/genmonlib/generac_evolution.py#L2146

wackyhobo, 2019-10-24:
Still plugging away at the HA Modbus interface to my Generac. Thanks for welcoming my questions.

For the Generac Evolution, can the following be written via Modbus and if so what in the Modbus HEX address?

Evo AC Exercise Time 

Evo AC Exercise Freq (Weekly, Biweekly, Monthly)

Evo AC Exercise Day of Week

Evo AC Quiet Mode

 

https://github.com/jgyates/genmon/blob/master/genmonlib/generac_evolution.py#L2681

https://github.com/jgyates/genmon/blob/master/genmonlib/generac_evolution.py#L2497

https://github.com/jgyates/genmon/blob/master/genmonlib/generac_evolution.py#L2146

jgyates, 2019-10-24:
These  are for Evolution Air or Liquid Cooled only:
002c   - Exercise Time HH:MM
002d - Exercise Frequency (Weekly 0000, Biweekly 0001, Monthly 0002)
002e - Exercise Time Exercise Day Sunday =0, Monday=1

NOTE: Writing quiet mode over modbus  only works on Evolutio Liquid Cooled models
002f - Quiet Mode  - 1 = quiet, 0 = disable quiet 

I am not aware of a way to write exerciser time on any model.

jgyates, 2019-10-26:
clarification: I am not aware of a way to write exercise duration on any model.

wackyhobo, 2019-11-19:
I’m still chipping away at interfacing my Generac 22kw (AirCooled) Evolution controller to my Home Assistant via modbus and ran into some more conundrums that I could your assistance getting past.

 

1.	On the aircooled, is there a modbus register that represents the transfer switch status or is that inferred? If it is represented by a modbus register which one and what do the values represent?
2.	What modbus addresses are written to and the required values to do the following:

a.	Remote Stop Generator
b.	Remote Start Generator
c.	Remote Start Generator and Transfer

 

Thank you for your help.

jgyates, 2019-11-19:
I don't know if a way to read the transfer switch relay status on an Air Cooled unit, only on Evo Liquid Cooled.

The remote commands may require extra measures than other reads and writes. To execute remote commands you have to write two commands back to back without writing to any other registers in between.

This function handles these commands:

https://github.com/jgyates/genmon/blob/master/genmonlib/generac_evolution.py#L1044

Start write 0000 to 0004 and 0001 to 0003
Stop write 0000 to 0004 and 0000 to 0003
Start Transfer write 0000 to 0004 and 0002 to 0003

You could probably do this with implement 32 bit modbus writes, but I have not tested these commands with 32 bit writes.

Start write 00010000 to 0003
Stop write 00000000 to 0003
Start/ Transfer write 00020000 to 0003

liltux, 2019-11-19:
@jgyates I just want to clarify.  There is no feedback on transfer switch state.  The controller energizes a relay on the controller pcb that takes wire 23 to ground.  Once wire 23 has ground potential there is now a path for current to flow and the transfer relay will energize and the transfer switch should move to standby position.  The controller only knows that wire 23 was grounded at the controller.  Sorry it’s my technician side.

jgyates, 2019-11-19:
Agreed. Thanks for the clarification. No apology needed, more details are always better in my view.

wackyhobo, 2019-11-20:
For a simple test using ModScan32, I was able to write  of integer value of “1” to 40004 (0003) and the generator started and of integer value of “0” to 40004 (0003) and the generator stopped. So cool. Thanks!

 

https://github.com/jgyates/genmon/blob/master/genmonlib/generac_evolution.py#L1044

wackyhobo, 2019-11-20:
I got so excited I forgot to ask about the auto-manual-off switch. Can that also be manipulated remotely?

 

I see the following in the generac_evolution.py#L1044 but I wasn’t sure which register got written to. So for clarity, can the auto-manual-modes be change remotely and if so what modbus addresses are written to and the required values to change the switch states?

 

                if Command == "auto":

                    Register = 0x000f   # set button to auto

                elif Command == "manual":

                    Register = 0x000e

                elif Command == "off":

                    Register = 0x0010

 

 

 

 

https://github.com/jgyates/genmon/blob/master/genmonlib/generac_evolution.py#L1044

jgyates, 2019-11-20:
the remote control of the auto/manual/off switch appears to be only work on  Evolution Liquid Cooled controllers.

liltux, 2019-11-20:
@jgyates you can place your generator in off position remotely?  Then back in auto?

jgyates, 2019-11-20:
yes

liltux, 2019-11-20:
Are you capable of doing this during an outage?  If so that is bad.  We don't want anyone turning a generator off remotely during an outage.  It could cause equipment and property damage.

jgyates, 2019-11-20:
I have not tried  to do this during an outage.   It does work when the generator is running in manual, for example I can start the generator remotely and then stop it by setting the switch to off remotely.

liltux, 2019-11-20:
Well if you test under load or "in auto", remove the utility voltage From the sensing circuit by either turning off the main utility breaker or pulling the sensing fuses (that way the generator will think there is an outage), then open (turn off) the generator breaker.  No power going into the home and gen in auto, then you can safely test the feature.  In reality if in auto under load the ability to turn off remotely should not be available or possible.

jgyates, 2019-11-20:
I will try to schedule a test on this, but my guess is that this will work just like the off / auto /manual switch in that you can turn it off, even during an outage. When I remotely change the switch state, the lights on the front panel button also change.

liltux, 2019-11-20:
If you could verify that would be great.  Performing the steps above should allow you a safe way of testing.  I know several people that want this ability but I always strongly advise against it, unless they have a thorough understanding of the implications of shutting down a generator under a load.  Doing this does not initiate a cool down and also creates a brown out on the electrical system.

jgyates, 2019-11-25:
Can you give more details? Are you referring to the current output from the web interface? or are you measuring this value with a meter? If you are referring to the web interface value then this page will give you more info:

https://github.com/jgyates/genmon/wiki/Appendix-D-Known-Issues

See number 5 .  Since the current output from genmon is actually an estimate based  sensor data read from the controller, I would not expect this to work as well as a meter. The internal sensors typically work off induction so they will produce some value even without a load.