jgyates/genmon #306: Question about alarm "Alarm Log : 08/07/19 10:12:09 Inspect Air Filter"

mdetweiler, 2019-08-07:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below.

***NOTE:*** If you are having issues with your serial connection, please read this page before posting:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

If you are having other issues, please see the following page:

https://github.com/jgyates/genmon/wiki/3.5---General-Troubleshooting

If you need to send you logs and registers to the developer, if you email is setup and working properly you can click send your logs on the About page in the web interface.

### Expected Behavior

Not sure

### Actual Behavior

I'm wondering if anyone knows how the alarm for an air filter is generated?  Is it a real sensor detected issue or a timed reminder.  If it is a timed reminder, then should it say CRITICAL in the alarm message?
Subject: Generator Alert at Woodland: CRITICAL
Body excerpt: Alarm Log : 08/07/19 10:12:09 Inspect Air Filter

### Steps to Reproduce (including precondition)

{Please write here}

### Screenshot or Pictures relating to the problem (if possible)

{Please write here}

### Your Environment

- Generator Model: {Please write here}
- Generator Registers: Submit via the About page
- Genmon Version: {Please write here}

liltux, 2019-08-07:
@mdetweiler all of the service timers are hrs based or timer.  You have annual inspect battery as well as level A and level B run hr timers.  I believe the Air Filter is a 2 yr reminder but not sure.

jgyates, 2019-08-07:
@mdetweiler 

The logic behind the emails is this:

If any alarm is present the subject includes the "CRITICAL" message since the generator typically will not start until the alarm is recognized. Some alarms are considered a warning by the controller firmware,  depending on the alarm condition and controller version, but genmon labels all alarms as "critical". Since I don't have a definitive list of what alarms prevent the controller from starting and which ones do not I opted to just make the email subject read the same for all alarms.

mdetweiler, 2019-08-07:
Excellent explanation!  Thanks!

closed 2019-08-07 by jgyates

grsthegreat, 2020-07-26:
Yes, thanks. I just recently had an alert about CRITICAL battery issue. Was startled until i realized it was only the annual battery inspection reminder......i was going to ask about that here until i read old posts. Theres such a large lists of alarm (gen wont run) codes and warning (gen will run) that i can see why it would be a hard task to write this code. There are 24 error codes and 19 warning codes at minimum

skipfire, 2020-07-26:
If there is a list of "always" warning codes it should not be too hard to create an exclusion list of codes.  The problem would be if any of the codes are warnings are one controller and errors on another.  It could potentially be a list of warning codes entered into the UI if it might be per model.

jgyates, 2020-07-26:
The file  [ALARMS.txt](https://github.com/jgyates/genmon/blob/master/data/ALARMS.txt) contains the known (at least to me) list of Evolution alarm codes. Nexus does not use alarm codes that are displayed. Which of these keeps the generator from starting vs which ones allow the generator to run, I do not know.

Another issues is that alarms are reported two ways 1) the engine status register (Evo ande Nexus) and 2) the alarm log. The controller enumerates the alarms differently for these two ways as there are fewer values of the alarm codes for the engine status register vs the alarm codes for the alarm log. Evolution stores the alarm code listed in ALARMS.txt in the alarm log, Nexus does not have alarm code in the alarm log, but does appear to share the alarm values of the engine status register. The values of alarm codes in the engine status register were unknown and I have had to figure out these values with the help of people submitting registers via the "Automated Feedback" setting (genmon sending me emails when register values do not make sense). I believe I have *most* of the alarm codes included (at least the more common alarms), but I feel sure that some are missing.

The way it works now is this:

Try to see if there is an alarm from the engine status register, if it can not then fall back to the Alarms.txt lookup method. If an alarm is detected, then put the words "CRITICAL" in the email. All alarms are treated equally.

jgyates, 2020-07-26:
Also, service due alarms do not generate an alarm log 4 digit alarm code, but they do generator a unique two digit alarm in the engine status register