jgyates/genmon #308: Can Not Connect Serial over TCP/IP

wackyhobo, 2019-08-11:
I am trying to connect Genmon (V1.13.2) to my Generac Guardian 22kW Generator (G0065521) using Serial RS232 to Ethernet Converter (USR-TCP232-410S).

I was successful connecting Genmon serially to the generator.

I have the Serial RS232 to Ethernet Converter setup and I can connect with Modscan32 and see live values in the registers.

I enabled Serial over TCP/IP, set Serial Server TCP/IP Address and port. I then save changes. When that was unsuccessful, I reboot the Raspberry PI.

[image]
[image]
[image]
[image]
[image]
[image]
[image]

- Generator Model: {Generac Guardian 22kW Generator (G0065521e}
- Generator Registers: Submit via the About page
- Genmon Version: {Version V1.13.28}

wackyhobo, 2019-08-11:
[image]
[image]
[image]

jgyates, 2019-08-11:
@wackyhobo 

It looks like you have the USR device setup to be a modbus TCP server. Genmon expects the USR device to act as a simple serial TCP server (i.e. genmon handles the modbus protocol and the serial over IP device is just a pass through of the data).  Try changing the "Work Mode" parameter to TCP Server / None instead of TCP Server Modbus TCP.

This page has the requirements for serial over IP functionality.

https://github.com/jgyates/genmon/wiki/Appendix-F----Serial-over-IP

Let me know if this works for you.

wackyhobo, 2019-08-11:
Perfect. It works now! Thanks!

closed 2019-08-11 by wackyhobo