jgyates/genmon #309: Evol 2 Firmware Updated from V1.07 to V1.08

jimland58, 2019-08-13:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below.

***NOTE:*** If you are having issues with your serial connection, please read this page before posting:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

If you are having other issues, please see the following page:

https://github.com/jgyates/genmon/wiki/3.5---General-Troubleshooting

If you need to send you logs and registers to the developer, if you email is setup and working properly you can click send your logs on the About page in the web interface.

### Expected Behavior

{Please write here}

### Actual Behavior

{Please write here}

### Steps to Reproduce (including precondition)

{Please write here}

### Screenshot or Pictures relating to the problem (if possible)

{Please write here}

### Your Environment

- Generator Model: {Please write here}
- Generator Registers: Submit via the About page
- Genmon Version: {Please write here}

jimland58, 2019-08-13:
Got this notification from GenMon yesterday afternoon "NOTE: This message is a notice that the version of the generator controller firmware has changed from V1.07 to V1.08"

GenMon is still working so that's good.  Anyone know what changes were made with V1.08?

liltux, 2019-08-13:
AC Firmware Rev 1.08 For Evo 2.0 and Sync 3.0 

Improved Mobile Link Cellular 4G LTE

Improved Wi-Fi

closed 2019-08-16 by jgyates