jgyates/genmon #310: Model info wrong and inconsistent.

mdetweiler, 2019-08-18:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below.

***NOTE:*** If you are having issues with your serial connection, please read this page before posting:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

If you are having other issues, please see the following page:

https://github.com/jgyates/genmon/wiki/3.5---General-Troubleshooting

If you need to send you logs and registers to the developer, if you email is setup and working properly you can click send your logs on the About page in the web interface.

### Expected Behavior

The model information is not correct in most places.

### Actual Behavior

Model information should be correct if possible and consistent.
Actual Model Number: QT02224ANA

GenMon shows:
Maintenance :
    Model : Generic Air Cooled
    Generator Serial Number : 
    Controller : Nexus, Liquid Cooled
    Nominal RPM : 3600
    Rated kW : 20

Monitor :

    Generator Monitor Stats :
        Monitor Health : OK
        Controller : Nexus, Air Cooled
        Run time : Generator Monitor running for 42 days, 8:50:38.
        Generator Monitor Version : V1.12.6

It also says 20kw even though this is a 22kw model.

### Steps to Reproduce (including precondition)

{Please write here}

### Screenshot or Pictures relating to the problem (if possible)

{Please write here}

### Your Environment

- Generator Model:  QT02224ANA
- Generator Registers: Submit via the About page Done!
- Genmon Version: V1.12.6

jgyates, 2019-08-18:
The model and kw rating detection works different depending on the model (EvoAC, EvoLC, NexusAC, NexusLC)
I received your registers but to give you more info I would need to see your logs (the other button on the about page).

When I run a detection based on the registers you submitted I get the following:

Maintenance : 
    Model : Unknown
    Generator Serial Number : 
    Controller : Nexus, Liquid Cooled
    Nominal RPM : 1800
    Rated kW : 22 kW
    Nominal Frequency : 60 Hz
    Fuel Type : Propane
    Engine Displacement : 2.4 L

Note that an "Unknown" model will get replaced in the web interface by "Generic Air Cooled" or "Generic Liquid Cooled" depending on how the registers are read and if you have overridden the autodetect in /etc/genmon/genmon.conf file.

If you are reading a different controller type on the Maintenance page vs the Monitor page (and based on your post it appears you are), then this is due to a setting in the /etc/genmon/genmon.conf file.

in the default version of the /etc/genmon/genmon.conf file there are two lines that look like this:

#liquidcooled = True

#evolutioncontroller = True

The # at the beginning of the line is a comment which make the software ignores this line. If the #  is removed then the autodetection of the controller is overridden. Based on the data I have I would guess that one or both of these settings has been changed from the default. The default is the recommended setting.

Also, your registers you submitted do not appear to  have a register for the serial number (Register 01f4). The serial number is used to get a model number by looking up the serial number on the internet. The logs may have more info about why you are not reporting a value of this register, however if you have replaced your controller then the serial number will not be present, but you should still have a reading in the log file, but you do not. 

You can override the serial number from the advanced page if this is the case (click the gear icon in the upper right corner twice to get the advanced page). Add your serial number to the setting named "Serial Number if Missing". The serial number must be 10 digits, pad with leading zeros to make it 10 digits.

Let me know if this resolves your issue.

mdetweiler, 2019-08-18:
I sent the logs to help you debug.  I will check the config asap tomorrow.  I have not changed anything but this was a fully configured system that I bought on eBay so it might have some changes made by the builder.

Sent from Bloomberg Professional for Android

----- Original Message -----
From: jgyates <reply@reply.github.com>
At: 17-Aug-2019 21:47:16

The model and kw rating detection works different depending on the model (EvoAC, EvoLC, NexusAC, NexusLC)
I received your registers but to give you more info I would need to see your logs (the other button on the about page). 
When I run a detection based on the registers you submitted I get the following: 
Maintenance :
Model : Unknown
Generator Serial Number :
Controller : Nexus, Liquid Cooled
Nominal RPM : 1800
Rated kW : 22 kW
Nominal Frequency : 60 Hz
Fuel Type : Propane
Engine Displacement : 2.4 L 
Note that an "Unknown" model will get replaced in the web interface by "Generic Air Cooled" or "Generic Liquid Cooled" depending on how the registers are read and if you have overridden the autodetect in /etc/genmon/genmon.conf file. 
If you are reading a different controller type on the Maintenance page vs the Monitor page (and based on your post it appears you are), then this is due to a setting in the /etc/genmon/genmon.conf file. 
in the default version of the /etc/genmon/genmon.conf file there are two lines that look like this: 
#liquidcooled = True 
#evolutioncontroller = True 
The # at the beginning of the line is a comment which make the software ignores this line. If the #  is removed then the autodetection of the controller is overridden. Based on the data I have I would guess that one or both of these settings has been changed from the default. The default is the recommended setting. 
Also, your registers you submitted do not appear to  have a register for the serial number (Register 01f4). The serial number is used to get a model number by looking up the serial number on the internet. The logs may have more info about why you are not reporting a value of this register, however if you have replaced your controller then the serial number will not be present, but you should still have a reading in the log file, but you do not. 
You can override the serial number from the advanced page if this is the case (click the gear icon in the upper right corner twice to get the advanced page). Add your serial number to the setting named "Serial Number if Missing". The serial number must be 10 digits, pad with leading zeros to make it 10 digits. 
Let me know if this resolves your issue.

jgyates, 2019-08-18:
I don't see anything significant in the logs. Is this something that started happening recently?  You may try rebooting your pi and see if that makes a difference. 

Are you seeing any communication errors? You log does not show any recently but just making sure there are no strange communication issues.

Also you could try to update the software on the about page. You are running version V1.12.6 from 2018-12-31, the latest is V1.13.28 from 2019-07-20

mdetweiler, 2019-08-19:
This has been an issue from day one for me.  Other that this, it works great, but I was just hoping to get it 100% perfect.  Looks like both values in the config are commented out so that isn't it.  I will try an update when I can and see how that goes.

mdetweiler, 2019-08-19:
Ok..  the upgrade seems to have helped.  I see that the serial number is still off, but it tells me that the controller was replaced.  Is there a way to tel the controller about the serial number of should I just override it in genmon?

jgyates, 2019-08-19:
I would recommend override it in genmon. This will give you all the functionality you would have if there was a serial number present. Any time a controller is replaced it has an empty serial number in the controller. Empty serial numbers tell the firmware that a registration is not required. If we wrote out the serial number via modbus then you would have to go through the registration process again with generac or enter the code from your initial registration.

jgyates, 2019-08-19:
Closing this issue for now. Feel free to post to this thread if you have any issues.

closed 2019-08-20 by jgyates