jgyates/genmon #312: Zero utility voltage values being reported to MQTT

mattster98, 2019-09-05:
I have been monitoring my utility input voltage by charting the value being sent to MQTT via the plugin.  My charts had occasional single points and periods where the voltage reported was zero.  My utility was not interrupted during this time, so I began tracking the problem down.  The MQTT plugin is in fact sending 0V over mqtt, which is the source of my problem.  I'm not sure why it's getting zero, but the periods when it happens have the log snippet attached below aligned with the time period that the zeros show up in MQTT.  

I updated to 1.13.29 today to see if that helped, and it started sending these "firmware updated" emails where curiously enough, the firmware version is also reported as zero.

### Expected Behavior

Voltage is reported as measured and updated when the voltage changes.  Zeros would only be present when a utility outage occurs.

### Actual Behavior

Zero volts is being reported sporadically, without utility issues.

### Steps to Reproduce (including precondition)

Working genmon setup
mqtt plugin enabled and configured, connected to mosquitto
"JSON for numerics" set to ON to clean up the voltage value (is text with " V" tacked on the end otherwise)

### Screenshot or Pictures relating to the problem (if possible)

influxdb data from mqtt:
================

1567690776723408366 r710-reborn devices/generator/Status/Line/Utility Voltage/value 244
1567690778777651429 r710-reborn devices/generator/Status/Line/Utility Voltage/value 245
1567690784914723465 r710-reborn devices/generator/Status/Line/Utility Voltage/value 244
1567690789010188728 r710-reborn devices/generator/Status/Line/Utility Voltage/value 245
1567690795168168210 r710-reborn devices/generator/Status/Line/Utility Voltage/value 244
1567690797223901517 r710-reborn devices/generator/Status/Line/Utility Voltage/value 245
1567690801326505679 r710-reborn devices/generator/Status/Line/Utility Voltage/value 244
1567690805430830656 r710-reborn devices/generator/Status/Line/Utility Voltage/value 245
1567690809516510813 r710-reborn devices/generator/Status/Line/Utility Voltage/value 244
1567690813621983589 r710-reborn devices/generator/Status/Line/Utility Voltage/value 245
1567690819771320204 r710-reborn devices/generator/Status/Line/Utility Voltage/value 0
1567690825918760723 r710-reborn devices/generator/Status/Line/Utility Voltage/value 0
1567690832072373404 r710-reborn devices/generator/Status/Line/Utility Voltage/value 245
1567690834166753242 r710-reborn devices/generator/Status/Line/Utility Voltage/value 244
1567690838226365447 r710-reborn devices/generator/Status/Line/Utility Voltage/value 245
1567690840259973685 r710-reborn devices/generator/Status/Line/Utility Voltage/value 244
1567690844362219227 r710-reborn devices/generator/Status/Line/Utility Voltage/value 245
1567690848464899897 r710-reborn devices/generator/Status/Line/Utility Voltage/value 244
1567690854595316702 r710-reborn devices/generator/Status/Line/Utility Voltage/value 245
1567690856645001908 r710-reborn devices/generator/Status/Line/Utility Voltage/value 244
1567690862768368722 r710-reborn devices/generator/Status/Line/Utility Voltage/value 244
1567690868921882040 r710-reborn devices/generator/Status/Line/Utility Voltage/value 244

genmon.log from a period where zeros were reported and the firmware update email went out (not the same event as above): 
===
2019-09-05 14:24:05,946 : Validation Error: Invalid register length (base) 0001:0000 4 2
2019-09-05 14:24:05,987 : Validation Error: Invalid register length (base) 05f1:00000000 2 4
2019-09-05 14:24:06,397 : Validation Error: Invalid register length (base) 0001:0000 4 2
2019-09-05 14:24:06,438 : Validation Error: Invalid register length (base) 05f1:00000000 2 4
2019-09-05 14:24:06,836 : Validation Error: Invalid register length (base) 0001:0000 4 2
2019-09-05 14:24:06,888 : Validation Error: Invalid register length (base) 05f1:00000000 2 4
2019-09-05 14:24:07,175 : Frimware Changed from V1.06 to V0.00
2019-09-05 14:24:07,306 : Validation Error: Invalid register length (base) 0001:0000 4 2
2019-09-05 14:24:07,347 : Validation Error: Invalid register length (base) 05f1:00000000 2 4
2019-09-05 14:24:07,735 : Validation Error: Invalid register length (base) 0001:0000 4 2
2019-09-05 14:24:07,776 : Validation Error: Invalid register length (base) 05f1:00000000 2 4
2019-09-05 14:24:08,165 : Validation Error: Invalid register length (base) 0001:0000 4 2
2019-09-05 14:24:08,207 : Validation Error: Invalid register length (base) 05f1:00000000 2 4
2019-09-05 14:24:08,595 : Validation Error: Invalid register length (base) 0001:0000 4 2
2019-09-05 14:24:08,636 : Validation Error: Invalid register length (base) 05f1:00000000 2 4
2019-09-05 14:24:08,912 : Validation Error: Invalid register length (base) 0001:0000 4 2
2019-09-05 14:24:08,953 : Validation Error: Invalid register length (base) 05f1:00000000 2 4
2019-09-05 14:24:09,352 : Validation Error: Invalid register length (base) 0001:0000 4 2
2019-09-05 14:24:09,403 : Validation Error: Invalid register length (base) 05f1:00000000 2 4
2019-09-05 14:24:09,811 : Validation Error: Invalid register length (base) 0001:0000 4 2
2019-09-05 14:24:09,852 : Validation Error: Invalid register length (base) 05f1:00000000 2 4
2019-09-05 14:24:10,275 : Validation Error: Invalid register length (base) 0001:0000 4 2
2019-09-05 14:24:10,326 : Validation Error: Invalid register length (base) 05f1:00000000 2 4
2019-09-05 14:24:10,561 : Validation Error: Invalid register length (base) 003a:2020 4 2
2019-09-05 14:24:10,602 : Validation Error: Invalid register length (base) 003c:00000000 2 4
2019-09-05 14:24:10,724 : Validation Error: Invalid register length (base) 0001:0000 4 2
2019-09-05 14:24:10,765 : Validation Error: Invalid register length (base) 05f1:00000000 2 4
2019-09-05 14:24:11,173 : Validation Error: Invalid register length (base) 0001:0000 4 2
2019-09-05 14:24:11,214 : Validation Error: Invalid register length (base) 05f1:00000000 2 4
2019-09-05 14:24:11,612 : Validation Error: Invalid register length (base) 0001:0000 4 2
2019-09-05 14:24:11,654 : Validation Error: Invalid register length (base) 05f1:00000000 2 4
2019-09-05 14:24:12,052 : Validation Error: Invalid register length (base) 0001:0000 4 2
2019-09-05 14:24:12,093 : Validation Error: Invalid register length (base) 05f1:00000000 2 4
2019-09-05 14:24:12,797 : Validation Error: Invalid register length (base) 0001:0000 4 2
2019-09-05 14:24:12,869 : Validation Error: Invalid register length (base) 05f1:00000000 2 4
2019-09-05 14:24:13,572 : Validation Error: Invalid register length (base) 0001:0000 4 2
2019-09-05 14:24:13,653 : Validation Error: Invalid register length (base) 05f1:00000000 2 4
2019-09-05 14:24:14,387 : Validation Error: Invalid register length (base) 0001:0000 4 2
2019-09-05 14:24:14,458 : Validation Error: Invalid register length (base) 05f1:00000000 2 4
2019-09-05 14:24:15,188 : Validation Error: Invalid register length (base) 0001:0000 4 2
2019-09-05 14:24:15,259 : Validation Error: Invalid register length (base) 05f1:00000000 2 4
2019-09-05 14:24:15,950 : Validation Error: Invalid register length (base) 0001:0000 4 2
2019-09-05 14:24:16,021 : Validation Error: Invalid register length (base) 05f1:00000000 2 4
2019-09-05 14:24:16,724 : Validation Error: Invalid register length (base) 0001:0000 4 2
2019-09-05 14:24:16,795 : Validation Error: Invalid register length (base) 05f1:00000000 2 4
2019-09-05 14:24:28,948 : Frimware Changed from V0.00 to V1.06

### Your Environment
Serial to WiFi interface on the generator, running Genmon in a lxc container getting data over the network.

- Generator Model: Generac Guardian 22KW Air Cooled, Evolution 2.0 controller
- Generator Registers: Submitted via the About page
- Genmon Version: 1.13.29

jgyates, 2019-09-05:
@mattster98 

I have a couple of questions.

Is this new behavior, i.e. has this been working for you but stopped, or have you seen this since installing the software?

What model serial to TCP device are you using?

Based on your logs it looks like you are seeing a communication issue. My guess is that the serial device is timing out and the software moves on to the next packet, then the serial to tcp device responds with the first packet, which causes genmon to get invalid data for the registers. This would explain all of the data you are seeing.

If this is something that was working but then stopped, you can try power cycling your serial to tcp converter. This may fix the problem.

You can also try adding more time to the modbus timeout on the advanced page. Double click the gear icon in the top right twice, then look for the setting named "Additional Modbus Timeout (sec)" make this 1 or 2, then save the setting on the advanced page
 
The base issue is that while the modbus spec does provide for a mechanism to cancel a modbus request, generac does not appear to implement this on the residential controllers. So a time out that on the modbus request could still receive a modbus response after the timeout period. the time outvalue is also not defined by the modbus spec so this is also a grey area.

this is typically not an issue for direct serial, however since network times can vary I added the ability to add to the modbus delay with the "Additional Modbus Timeout" setting. 

Let me know how it goes.

mattster98, 2019-09-05:
Thanks for the quick response!

I have had genmon running for a while but only recently started paying attention to the MQTT values, so I don't think anything changed to make it start happening.

The serial to TCP device is: USR-WIFI232

I suspect the wifi signal to the serial to TCP device is sub-par, and can see timeouts being the problem.  However, I would expect timeouts to yield no values (null) rather than zeros.  Nulls I can handle with my charting since they aren't valid voltage values.

I will start by rebooting the converter.  Then if that doesn't help, I'll try adjusting the modbus timeouts.  I also have some ideas on how to improve the wifi signal which I'll do if the timeout has no effect.  This has been on my list, but was "working" so it hadn't become a priority. :)

jgyates, 2019-09-05:
If a time occurs the modbus will not assign a value (null) the the request. The problem is that the next request is initiated and the remote end (the controller) responds back with the value for the first request. Delays in the network make this issue more likely to occur. The default timeout is about 3 seconds. If genmon does not get a packet back it moves on to the next request, but if the far end responds but the response reaches genmon after the next modbus request is started then you will see the issue you and you logs are describing. Since this occurs at the modbus level, genmon thinks it is receiving data, which it is, it is just the wrong data. 

If it is related to your wifi signal then the longer timeout would likely make this issue better, but possibly not solve it 100%

mattster98, 2019-09-06:
Rebooting the converter didn't solve the problem, so I set the additional modbus timeout to 2.

It's been almost 24 hours without a 0 V report, so I think that helped.  I did see a few errors like before in the log, but I guess none long enough to trigger the zeros from being reported.  

I messed around with the wifi antenna also, but made it slightly worse.  I think I'm going to have to drill a hole in the enclosure to get the signal out.  It's 100% metal, with multiple layers in many places!

closed 2019-09-09 by jgyates