jgyates/genmon #314: Tankutility

shpegel, 2019-09-23:
### Expected Behavior
Fuel gauge on Status page reflects Tankutility data. Tankutility add on shows a save button after opening the first time. Information entered in Tankutility add on should not show password after saving. Function of Tankutility add on should give some idea of fuel tank level. Low fuel emails arrive if fuel is low.

### Actual Behavior

Fuel gauge on Status page was showing previous to activating the add on then disappeared after activating add on.  Has not come back yet. After turning on Tankutility add on and filling information no Save button appeared until add on was turned off, a save button appeared, I hit the save button and turned it back on and then Save button was present. I am getting random low fuel emails. This started previous to starting add on.

### Steps to Reproduce (including precondition)

as above

### Screenshot or Pictures relating to the problem (if possible)

on request

### Your Environment

- Generator Model: Generac 22KW Air-Cooled
- Generator Registers:  Done
- Genmon Version: 1.13.29

jgyates, 2019-09-23:
@shpegel 

The fuel gauge labeled "Estimated Fuel" is for the estimated fuel calculation that does not use the TankUtiliy meter. If the Tankuntility add on is enabled this gauge is replaced with a gauge labeled "External Tank"

Regarding this statement "After turning on Tankutility add on and filling information no Save button appeared until add on was turned off, a save button appeared"

What is your browser  and operating system and can you send a picture of this? Are you running any browser add-ons that would stop javascript from running in your browser? The save button code runs in the browser so this is likely a browser related issue.

I got your register data, but I  would need to see your logs to see if the tankutility add on is running properly.

shpegel, 2019-09-23:
Hi Jay,
I was using Chrome on a Windows 10 64bit system but just tried the same
thing on MS Edge and got same behavior. Below are screen shots of add on
before and after.  Java installed and up to date. Logs submitted.

Thanks,

Howard

[image: image.png]
[image: image.png]

jgyates, 2019-09-23:
actually number 3 above should be:

     sudo cp ./genmon/conf/genmon.conf /etc/genmon/genmon.conf
     sudo bash ./startgenmon.sh restart

shpegel, 2019-09-23:
[image]
[image]

shpegel, 2019-09-23:
Lines entered as instructed. BTW must be in genmon directory for second line to work. Status screen looks normal and I see no fuel gage. Tankutility add on came back looking like second shot above. No save button.

jgyates, 2019-09-23:
Looks like you closed this inadvertently. I re-opened it.

I got your logs, looks like the login for tank utility is failing

You gentankutil.log file has this error:

Error logging in, error code: 401 

401 is unauthorized access. Can  you validate your username and password by logging in here:

https://www.tankutility.com/

Then validate that the UI and the file /etc/genmon/gentankutil.conf have the same credentials.

Regarding the UI save button, do you see this if you enable any other add ons or is this just the tankutil? You can enable / disable without clicking save or click About (or some other menu item) and it will not save .

jgyates, 2019-09-23:
I tried to duplicate the save button issue on Windows 10 with both Chrome and Edge and I am not able to duplicate this.  Can you try it on another computer or device?

shpegel, 2019-09-23:
Sorry, had put hash marks instead of password and forgot to change back. Now tankutility is working and I see the gauge on status screen.  Entry screen is working but save button only shows up when a change has been made. Tried on another windows 10 and an android device, all working the same way. Not sure what happened the first time.  Thanks.
[image]

jgyates, 2019-09-24:
I looked at the code and talked with MichaelB (he wrote the javascript for the web interface). It does appear that the Save button is not suppose to appear unless a change is made. For what ever reason this is  not the  behavior I am seeing on any platform I test with. I will look into this further, but for now I don't see the save button issue as a problem (rather an artifact) as it does not impact the functionality of the software.

closed 2019-09-24 by jgyates

shpegel, 2019-09-24:
I agree.

Many thanks for the support yesterday.

Best regards,

Howard