jgyates/genmon #315: Exercise email giving incorrect information

jcherney, 2019-09-23:
[image]

### Expected Behavior
Proper RPM, frequency, Output voltage to be correct or close.
### Actual Behavior
See attached pics. Way off.
### Steps to Reproduce (including precondition)
Unknown because its inconsistent. 
### Screenshot or Pictures relating to the problem (if possible)
attached.
### Your Environment
- Generator Model: QT04842ANAX
- Generator Registers: Submit via the About page
- Genmon Version:1.13.28

Any idea what might be causing this? When I go to the actual web page data, it's correct. It's the email received thats wrong. I've seen it before and dismissed it.

jcherney, 2019-09-23:
Normal info is 1800 RPM, 60hz, and 240 v

jgyates, 2019-09-23:
If the email is the only thing  wrong, the question I have is which email is wrong (Look at the engine state in each email). It is likely that you get a few emails when starting the exercise cycle. Each email should have a different engine state listed in the body of the  email. If the values of RPM, Frequency and output voltage are not in the nominal range when starting or cranking, then this is normal since these values ramp up from zero. Emails are sent when the status register changes and the contents  of the registers are taken from that moment the status register changed.

jcherney, 2019-09-24:
Its the exercising state email (see the screenshot). I expect this behavior on cranking and 2 wire stopping,but not on exercising.

jgyates, 2019-09-24:
I  would guess the cause of this is one of the following:

a) The generator is not fully started or the  controller is not fully reporting the output 
or
b) The controller is reporting the  exercise state before the generator is fully ramped up to the expected output.

From the software perspective, the registers are read in a continuous loop from the controller, wth the status register being read more often in the loop to improve the responsiveness of the software reporting a change in the generator state. Once the status register is read it is compared to the last status value read. If the value changed an email is sent. When an email is sent it uses the last read values of the other registers to populate the values in the email (RPM, voltage, etc). Since the frequency, voltage output and RPM all ramp up from zero to their respective nominal value when the generator starts and this ramp up process takes a few seconds at a minimum I would guess that what you are seeing is not a cause for concern but rather an anomaly of the controllers reporting responsiveness and the speed of the serial link.

jcherney, 2019-09-24:
Thanks. I wasn't really concerned because when I looked at the data live it was correct. Just wanted to point it out. Thanks for the input.

closed 2019-09-24 by jcherney