jgyates/genmon #316: 2008 residential serial interface

nrcantin, 2019-10-02:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below.

***NOTE:*** If you are having issues with your serial connection, please read this page before posting:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

If you are having other issues, please see the following page:

https://github.com/jgyates/genmon/wiki/3.5---General-Troubleshooting

If you need to send you logs and registers to the developer, if you email is setup and working properly you can click send your logs on the About page in the web interface.

### Expected Behavior

{Please write here}

### Actual Behavior

Hi, I'm really impressed with your project and software. however, I am having (I think a minor) problem.

I have installed Genmon on a Raspberry 2B pi.  The software seems to run fine.  I can view the web interface on my PC.  I have interfaced it to my 14KW generac generator.  I have a "pre-nexus" circa 2008 model with the controller you note in the "known issues" item 6.   I have scoped the RS-232 interface to/from the generator.  I can see Genmon polling the generator controller.  The generator controller is responding.  However it looks like the protocol is non Modbus.  I have looked for the advanced settings on the web interface and cannot find it.  Also I have reviewed the Genmon configuration file and cannot find the "response_address" variable to configure.  I have run the serial config, it reports all is fine.  The loop back test works fine.  I suspect its the protocol conflict you note in the know issues.  Can you point me to where I can set the Modbus Slave transmit address / response_address.  
Thank you in advance for any help.

### Steps to Reproduce (including precondition)

{Please write here}

### Screenshot or Pictures relating to the problem (if possible)

{Please write here}

### Your Environment

- Generator Model: {Please write here}
- Generator Registers: Submit via the About page
- Genmon Version: {Please write here}

jgyates, 2019-10-02:
To reach the advanced page, click the gear icon in the top right of the web interface twice. The Modbus Slave transmit address parameter is the third parameter on the advanced page.  It  should be set to 41 for a pre-nexus model.

nrcantin, 2019-10-02:
Thanks so much for your quick response!  Unfortunetley its raining now...so testing will have to wait until the rain stops.  Thanks again!

jgyates, 2019-10-05:
I am closing this issue for now  but  feel  free to post to  this thread if you have questions related to this topic.

closed 2019-10-05 by jgyates

nrcantin, 2019-10-05:
Thanks much. Will do.

nrcantin, 2019-10-05:
Should have mentioned, I had to travel to LA. Will test with the update wen I get back home

jgyates, 2024-10-04:
@nrcantin 

do you see anything like the behavior described in this thread: https://github.com/jgyates/genmon/issues/1191

nrcantin, 2024-10-05:
Hi JayNo I haven’t seen this behavior.Norm