jgyates/genmon #317: Uptime is incorrect.

steve-bailey1, 2019-10-05:
Under the monitor section the run time for the generator monitor is showing 18,000+ days however the pi has only been up for 9 days.  I have attached a screenshot.

[image]

steve-bailey1, 2019-10-05:
Sorry forgot to mention ... latest code and everything else seems fine.

jgyates, 2019-10-05:
Looks like  you  are  running CentOS, what is the output of this command?

  cat /proc/uptime  
 

on raspbian it looks like this:

     cat  /proc/uptime 
     705944.45 2812974.41

and the corresponding uptime  command:

       uptime
      13:48:18 up 8 days,  4:05,  1 user,  load average: 0.00, 0.00, 0.00

The first digits in the /proc/uptime file should be the uptime in seconds.

This is the same behavior on debian  and ubuntu, but RedHat / CentOS may be different.

jgyates, 2019-10-05:
scratch that. I read your post wrong, let me look at the code and see  if I see anything odd.

jgyates, 2019-10-05:
That is odd. I am not  sure what is happening.  The program calculates this with the standard python time libraries. It read the system time when the program starts and then when the page is displayed it reads the current time again and subtracts the two and the python libraries will output a string of this time difference. 

I would guess that  either your system time or time zone has changed recently, a software update has occurred and a restart is needed.

Is your time set via NTP? Is your system time accurate? 

You could try restarting the software or restarting the system. After looking at the code I am guessing that this is something unique to your system in some way.  The code for this is known to work and has not change for quite  a while.

lmamakos, 2019-10-05:
Possibly the clock wasn't set when genmon started, and the got set to the present/correct time  (and jumped forward) subsequently.  The system-maintained uptime number (at least as far as the Linux kernel is concerned) isn't based on timestamps, but elapsed time (e.g., ticks/clock interrupts/etc) since boot.  This is why these two things could be far out of whack with each other.

Probably the system time was the UNIX epoch, since the time right now is about 18174 days since the UNIX epoch (Jan 1, 1970).

steve-bailey1, 2019-10-05:
Didn’t think about that. That’s most likely it. I have a cron job on boot to get time from ntp and then another to start genmon. I can adjust the timing and see what happens. 

Thanks guys

jgyates, 2019-10-06:
Makes sense to me. Thanks for the input @lmamakos

closed 2019-10-06 by jgyates