jgyates/genmon #319: Edit Journal Entries

lmackley, 2019-10-21:
A nice feature would be the ability to edit or delete individual journal entries.  Currently the only option is  to clear the entire journal and recreate if there is an error or a misspelling.

jgyates, 2019-10-21:
@lmackley 

I  have this on the list for new features. I agree that this is a needed feature. Hopefully I will be able to get to this soon.

lakee911, 2020-05-09:
I think this would be fantastic. An added benefit would be to provide a setting to disable clearing the journal entries. While not foolproof, it could simply hide the button and provide an extra level of safety (multiple-step process).

Does the backup feature also store the journal?

jgyates, 2020-05-09:
Yes, the journal is backed up.

jgyates, 2020-09-02:
This has been implemented with V1.14.28.

closed 2020-09-02 by jgyates