jgyates/genmon #320: Number of outages listed from log

dennis48755, 2019-10-31:
Hi,

Are the quantity of outages listed from the Outage Log restricted? I would like to be able to view all of them.

In GenMon under Outage, I have 50 log entries listed (with a scroll bar). However my outage.txt file contains 68 entries.

I did manually edit my outage.txt to add outages that occurred before I installed GenMon, but believe I did it correctly.

Thanks,
Dennis

[outage.txt](https://github.com/jgyates/genmon/files/3795633/outage.txt)

jgyates, 2019-10-31:
outage.txt will hold more than 50 outages and does not have a limit. The web interface will only display the last 50 entries for optimization reasons.

dennis48755, 2019-11-03:
Jason, is there any way to make the truncated log data available from within the web interface. Maybe a drop-down selection to optional display 100 or all log entries?

Dennis

jgyates, 2019-11-04:
Hi Dennis,

I checked in V1.13.30 that should make the display limit 100 instead of 50. Let me know if you have any issues.

Jason

dennis48755, 2019-11-04:
Thanks Jason.

Dennis

closed 2019-11-04 by jgyates