jgyates/genmon #321: Genmon can't communicate with gen after an outage

SethL, 2019-11-06:
So genmon works awesome and will not lose a single serial packet for months.  Whenever I have an outage that lasts for 5 or so hours (or more) genmon stops communicating. During the outage genmon works great and never misses a serial packet.  As soon as the utility power comes back up and my transfer switch flips over, genmon stops communicating with the generator.  All serial packets stop transferring. 

My Pi is connected via Ethernet and I have all of the cables shielded.  When genmon stops communicating, I can still VNC into the Pi.  I initiate a reboot and genmon still can't talk to the generator.  I have to physically open up the generator and disconnect the USB cable powering the Pi. After powering it back on, it works great.  It will also work fine for short outages and exercising. Its something about the transfer back to street power.

Any guidance would be appreciated.

Seth

jgyates, 2019-11-06:
Seth,

I am not sure what is happening. One thing to check would be the ground connection. Is the pi grounded to the generator via the serial cable? If it is you may want to validate this with a multimeter if you have one to make sure the connection is not faulty or intermittent.  How is your pi powered? Are you using the battery to USB adapter or are you using a wall wart? 

Another thing to check the next time this happens is the look on the monitor page and see if there are any undervoltage, frequency cap or frequency  throttling events. This would indicate a problem with the power being supplied to the pi.

Jason

SethL, 2019-11-07:
hmmm I'm not sure how its grounded honestly. Is there a good method to ground the pi directly to the generator?  I have the 12 usb motorcycle adaptor running off of the generator battery.  I would use a 12V AC adapter but there isn't an outlet in the generator.   I didnt see any errors on the page before restarting it.

Thanks for your help!

jgyates, 2019-11-07:
Since this occurs when power is restored, the thing that comes to mind is that the phase of the 60Hz power to  your house would change with the transfer switch moving back to utility. This can cause some problems with sensitive electronics, but I have never heard of it causing a problem with a pi. This however does not explain your problem since your pi is running on battery. But the battery is charged from the AC from your house. The problem could lie in your motorcycle adapter or the pi could be corrupt or damaged, possibly even the RS232 level converter.

If this occurred once, I would wait and see if it happens again, if has occurred multiple times then you may have to make a guess at replacing one part at a time.

Sorry I don't have a better idea. I have never heard of this specific issue .

SethL, 2019-11-07:
So I found this on your site:

Unless you Pi power supply ground is common to the generator battery ground you will need to connect ground to your serial ground on your Pi.

 

So mine is grounded on the cable as well as the USB on the gen battery.  I work with audio and you can have problems with different equipment grounded to 2 different sources.  You think that could be an issue here?  I could lift the ground on the serial cable maybe?

jgyates, 2019-11-08:
The controller ground and the batter ground should be common (assuming the generator was installed correctly. This reason you need a ground connection separate from the motorcycle USB power supply is that the USB power supply, like a wall wart power supply, uses a transformer to convert  12  volts on the battery to a 5  volts which is suitable for USB. Transformers are not required to provide a  common ground between the 12V side and the 5V side. The ground cable to the controller ensures that the pi voltage is using the same ground reference as the controller. If this were not the case you would have extremely unreliable communications.

jgyates, 2019-12-04:
@SethL ,

I am going to close this issue for now but feel free to comment in this thread if you have more data or additional questions.

closed 2019-12-04 by jgyates